bb-qq/r8152 #1: DS1815+

Youpimatin, 2019-09-17:
I would like to use an ASUSTOR AS-U2.5G on a DS1815+ DSM 6.2.2

I've begun to try to compile this myself but I've not succeeded for the moment.
I tried both to cross-compile on a VM and to direct compile on the DS1815+ and got stuck, honestly my knowledge of linux is too weak and I'm not sure where to start and what to use/install.

(I'd like to see how you made it, I'm eager to learn how to do it myself.)

Youpimatin, 2019-09-17:
Linux Ds1815 3.10.105 #24922 SMP Wed Jul 3 16:37:24 CST 2019 x86_64 GNU/Linux synology_avoton_1815+

bb-qq, 2019-09-18:
I added a driver for avoton platforms. Please check the Release page.

But it is safer if you can build the driver yourself because I cannot guarantee to continue providing the driver in the future. These articles should help you.
https://originhelp.synology.com/developer-guide/create_package/compile_kernel_module.html
https://github.com/SynologyOpenSource/pkgscripts-ng

Youpimatin, 2019-09-19:
Thanks a lot !

Here are the filtered logfiles of the Syno booting with TWO of these Asustor adapters connected.

they appears as eth4 and eth5
then eth5 disappears and the integrated UMS/virtualCDROM appears

`
dmesg | grep -i usb

[   12.902659] ACPI: bus type USB registered
[   12.907476] usbcore: registered new interface driver usbfs
[   12.913781] usbcore: registered new interface driver hub
[   12.919912] usbcore: registered new interface driver ethub
[   12.926269] usbcore: registered new device driver usb
[   12.948691] ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
[   12.975528] ehci-pci 0000:00:16.0: new USB bus registered, assigned bus number 1
[   13.005061] ehci-pci 0000:00:16.0: USB 2.0 started, EHCI 1.00
[   13.011913] hub 1-0:1.0: USB hub found
[   13.034821] uhci_hcd: USB Universal Host Controller Interface driver
[   13.085880] etxhci_hcd-170202 0000:04:00.0: new USB bus registered, assigned bus number 2
[   13.096497] ethub 2-0:1.0: USB hub found
[   13.113015] etxhci_hcd-170202 0000:04:00.0: new USB bus registered, assigned bus number 3
[   13.123126] ethub 3-0:1.0: USB hub found
[   13.345341] usb 1-1: new high-speed USB device number 2 using ehci-pci
[   13.495994] hub 1-1:1.0: USB hub found
[   13.802824] usb 1-1.1: new high-speed USB device number 3 using ehci-pci
[   14.460216] usb 2-2: new low-speed USB device number 2 using etxhci_hcd-170202
[   15.668547] usb 2-2: ep 0x81 - rounding interval to 128 microframes, ep desc says 160 microframes
[   16.789241] usb 3-3: new SuperSpeed USB device number 2 using etxhci_hcd-170202
[   18.025261] usb 3-4: new SuperSpeed USB device number 3 using etxhci_hcd-170202
[   18.051517] drivers/usb/core/ethub.c (2055) Same device found. Change serial to 000000000
[   30.981386] usb 2-2: ep 0x81 - rounding interval to 128 microframes, ep desc says 160 microframes
[   42.418374] usb-storage 1-1.1:1.0: USB Mass Storage device detected
[   42.425616] scsi10 : usb-storage 1-1.1:1.0
[   42.430357] usbcore: registered new interface driver usb-storage
[   42.494052] usbcore: registered new interface driver usblp
[   42.619201] usbcore: registered new interface driver usbhid
[   42.625443] usbhid: USB HID core driver
[   61.301318] usb 2-2: ep 0x81 - rounding interval to 128 microframes, ep desc says 160 microframes
[   69.874389] usb 3-3: reset SuperSpeed USB device number 2 using etxhci_hcd-170202
[   71.000591] usb 3-4: reset SuperSpeed USB device number 3 using etxhci_hcd-170202
[   71.023369] usb 3-4: device firmware changed
[   71.086060] usbcore: registered new interface driver r8152
[   71.092804] usb 3-4: USB disconnect, device number 3
[   77.198607] usbcore: registered new interface driver snd-usb-audio
[   77.427728] usbcore: registered new interface driver snd-usb-hiface
[   83.085961] usb 3-4: new SuperSpeed USB device number 4 using etxhci_hcd-170202
[   83.110226] drivers/usb/core/ethub.c (2055) Same device found. Change serial to 000000000
[   83.120541] usb-storage 3-4:1.0: USB Mass Storage device detected
[   83.127514] scsi11 : usb-storage 3-4:1.0
[   84.135479] scsi 11:0:0:0: CD-ROM            Realtek  USB CD-ROM               2.00 PQ: 0 ANSI: 0

dmesg | grep -i 8152

[   39.981527] igb 0000:00:14.0 eth0: igb: eth0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX/TX
[   68.725520] r8152: module verification failed: signature and/or required key missing - tainting kernel
[   70.489943] r8152 3-3:1.0 eth4: v2.12.0 (2019/04/29)
[   70.495895] r8152 3-3:1.0 eth4: This product is covered by one or more of the following patents:
[   71.053409] r8152 3-4:1.0 (unregistered net_device): Get ether addr fail
[   71.061187] r8152 3-4:1.0 eth5: v2.12.0 (2019/04/29)
[   71.066758] r8152 3-4:1.0 eth5: This product is covered by one or more of the following patents:
[   71.086060] usbcore: registered new interface driver r8152

lsusb

|__usb1          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (ehci_hcd 0000:00:16.0) hub
  |__1-1         8087:07db:0002 09  2.00  480MBit/s 0mA 1IF  ( ffffffd1ffffffb2ffffffdbffffffad) hub
    |__1-1.1     f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation ----------------)
|__usb2          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (Linux 3.10.105 etxhci_hcd-170202 Etron xHCI Host Controller 0000:04:00.0) hub
[... 19 lines cut ...]
7: bond0: <BROADCAST,MULTICAST,MASTER,UP,LOWER_UP> mtu 9000 qdisc noqueue state UP mode DEFAULT
    link/ether --:--:--:--:--:-- brd ff:ff:ff:ff:ff:ff
8: eth4: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN mode DEFAULT qlen 1000
    link/ether --:--:--:--:--:-- brd ff:ff:ff:ff:ff:ff

synonet --show

System network interface list:

Host Name: Ds1815
Network interface: bond0
Manual IP
IP: 192.168.1.181
Mask: 255.255.255.0
Gateway: 192.168.1.1
DNS: 1.0.0.1
MTU Setting: 9000
1000, Full duplex, active mtu 9000
RX bytes: 2108783
TX bytes: 1169819
`

Yesyerday i've managed to test iperf3 between my laptop and the syno, each with a 2.5Gbps adapter,
2.3GBps rock solid for half an hour.
I've also managed to use SAMBA a little, but it works only for a minute and then it bugs and resets the driver... i'll retry tonight and sen you the logfiles if you want (but I imagine it's not a driver problem)

Youpimatin, 2019-09-19:
(Oh, and I used "ip" command to assign an IP / MTU to eth4, as synonet reports an error when used to configure an IP)

bb-qq, 2019-09-20:
You mean:
- When attached one device, it works but unstable
- When attached two devices, second one is disappear

right? Assuming these are correct, I try to answer.

Regarding first issue, it is essential to isolate the problem whether it caused by laptop or NAS. In case of Windows, I heard that disabling "power save feature" in the device manager is needed for stability. Is your laptop Windows?

Regarding second issue, I have no idea. If it is caused by power supply, using a self powered hub(with external power supply) might help.

Youpimatin, 2019-09-23:
You're righ on the two points, you understood well.
My laptop is a Windows one, so I'll try to disable power saving, thank you.

I think I have a log file when the transfer was unstable, I'll try to give it to you.

I said that synonet reports an error when used to configure an IP, it's true, however the interface appears in the synology control panel !

For the moment I have a new problem, not driver-related, I think it's because I tried to make a bond with the 2.5Gbit/s adapter and the four 1Gbit/s network cards of the synology.

After many tests I don't know why but I can't synchronise the two adapters to more than 1 Gbit/s ! (I tried with different 1m/10m CAT6A cables between them, tried on different machines, windows and linux)

It seems than :
- at least one of the adapter is broken and don't work anymore.
- or at least one adapter is locked to a maximum speed of 1 Gbit/s ? (and it keeps settings even when powered off ? maybe kept in an eeprom or a flash memory embedded in the adapter, maybe than I can reset with a windows/linux command or with an utility ? I don't know)
And I don't have any other multi-gigabit hardware to test adapters separately.

bb-qq, 2019-09-25:
> For the moment I have a new problem, not driver-related, I think it's because I tried to make a bond with the 2.5Gbit/s adapter and the four 1Gbit/s network cards of the synology.

I think bonding cannot improve throughput when using one connection. Does the test you conducted use multiple connections and measure ”total” performance correctly?

> After many tests I don't know why but I can't synchronise the two adapters to more than 1 Gbit/s ! (I tried with different 1m/10m CAT6A cables between them, tried on different machines, windows and linux)

I also heard that this chip slows down link speed to 1Gbps when the device is overheating and this behavior can be controlled by the "power save " setting I mentioned previously. I don't know whether the information is correct, but you might want to cool the device and measure again.

I have not encountered problems like this, because my adapter has some slits for cooling. But the ASUSTOR's seems not equip such mechanism.

Youpimatin, 2019-09-30:
> I think bonding cannot improve throughput when using one connection. Does the test you conducted use multiple connections and measure ”total” performance correctly?

It was just a test, the goal was to set the same IP to multiple ethernet adapters, not to improve bandwith on one connection (but it can work by enabling experimental "SMB multi channel")

I've got my hands on a multi-gigabit switch and a CLUB 3D ethernet adapter and I've isolated the ASUSTOR adapter that's broken (unable to connect at more than 1Gbit/s and -even in that case- not working)

So I confirm that CLUB 3D CAC-1420 (Type-A) is working with the driver.

> I also heard that this chip slows down link speed to 1Gbps when the device is overheating and this behavior can be controlled by the "power save " setting I mentioned previously. I don't know whether the 

information is correct, but you might want to cool the device and measure again.
> 
> I have not encountered problems like this, because my adapter has some slits for cooling. But the ASUSTOR's seems not equip such mechanism.

I still have stability problems when using the ASUSTOR adapter intensely for long period of time, it does some kind of "thermal throttling" (with periods of total inactivity) and sometimes completely disconnects while in SAMBA 

transfer, and the NAS is lost, rendering this solution unusable... at least for the moment, and I'm really disapointed.

One more thing : if you want a readme.fr.md I can do the translation.

bb-qq, 2019-10-01:
I understand your situation.

> So I confirm that CLUB 3D CAC-1420 (Type-A) is working with the driver.

> I still have stability problems when using the ASUSTOR adapter intensely for long period of time, it does some kind of "thermal throttling" (with periods of total inactivity) and sometimes [...]

And thank you for reporting behavior of both devices. This information of stability issue is very helpful for other users. I will update the readme.

> One more thing : if you want a readme.fr.md I can do the translation.

Thank you for proposing this, but I would like to keep the documents up to date, so I'm thinking of preparing only two languages.

closed 2019-10-11 by bb-qq