bb-qq/r8152 #2: DS216+ Celeron N3050

JohnsonWang1222, 2019-09-19:
I have installed the r8152-braswell-2.12.0-1.spk into my DS216+ which is using Celeron N3050 (braswell) 。

Then I plug ASUSTOR AS-U2.5G  into the front of DS216+ with a Type-C to Type-A adapter, no other response on DSM 6.2.2.

synonet or ifconfig didn't show the additional network interface.

bb-qq, 2019-09-19:
Please let me know outputs of:

- dmesg
- lsusb

Probably current driver does not have a entry of vid/pid for the device.

JohnsonWang1222, 2019-09-20:
after I use:

synonet --set_ip -4 eth1 add 192.168.1.1/24

The ASUSTOR 2.5G USB network adapter is working, and present in DSM control panel -> network

I have test DS216+ and DS218+, they are all working now.

Can anybody tell my how to boot the adapter in DHCP mode not static IP mode?

bb-qq, 2019-09-20:
As you mentioned in issue #3, currently we need to configure IP configuration after boot. This seems caused by initialization order of DSM. (boot -> configure network -> start services installed by packages) We need to find the way to load driver before "configure network".

Youpimatin, 2019-10-04:
I've found an possibly useful article, see sections "Auto load modules on boot" and "Auto configure eth1 on boot"
[http://www.nzederr.net/synology/2013/08/27/USB-eth1-Syn.html](url)

bb-qq, 2019-10-08:
Thank you for the information, but just using this method can not keep the configuration edited by Web GUI. We need to load the driver before "rc.network" is executed.  Of Course, editing the system script is one solution, but it will lost after upgrading of DSM.

So I am investigating suitable extension point of DSM to do that.

bb-qq, 2019-10-11:
Settings issue is merged to #3 .

closed 2019-10-11 by bb-qq