bb-qq/r8152 #4: Support DS1513+ with ASUSTOR AS-U2.5G

oldtwegg, 2019-09-26:
Hi~~
Could you please support Support DS1513+ with ASUSTOR AS-U2.5G?
Due to I would like to use the ASUSTOR AS-U2.5G on a DS1513+ with DSM 6.2.2, thanks a lot.

bb-qq, 2019-10-01:
I created the package. https://github.com/bb-qq/r8152/releases/download/v2.12.0-1/r8152-cedarview-2.12.0-1.spk
Please try this.

oldtwegg, 2019-10-03:
Thank you very much, it works!
But my DS1513+ is not powerful enough, so the transfer speed is up to 150 MB/s only. (Read & Write)

bb-qq, 2019-10-03:
Using jumbo-frame might help improve throughput because it reduce number of transactions generated by USB stack which consume processing power.

oldtwegg, 2019-10-04:
Thanks for your suggestion, I will try it later! 
And then report the performance asap after tested.

oldtwegg, 2019-10-04:
And I also tried the QNAP TS-877, and the performance is good enough.
But the transfer path is not native by kernel module, and it depends on the performance of native VM.
https://www.mobile01.com/topicdetail.php?f=494&t=5886024&p=2#74832653

bb-qq, 2019-10-11:
I close this issue since the driver itself works correctly.
FYI: You can try iperf if performance bottle neck is disk I/O.

closed 2019-10-11 by bb-qq