bb-qq/r8152 #5: DS218+ ？？

aetaf, 2019-10-08:
root@DS218:~# dmesg | grep -i 8152
[   90.520673] r8152: module verification failed: signature and/or required key missing - tainting kernel
[   90.650959] usbcore: registered new interface driver r8152
[   90.925962] r8152 2-2:1.0 eth1: v2.12.0 (2019/04/29)
[   90.931546] r8152 2-2:1.0 eth1: This product is covered by one or more of the following patents:
[ 1708.401014] r8152 2-3:1.0 eth1: v2.12.0 (2019/04/29)
[ 1708.406659] r8152 2-3:1.0 eth1: This product is covered by one or more of the following patents:
[ 4283.871200] r8152 2-3:1.0 eth1: v2.12.0 (2019/04/29)
[ 4283.876916] r8152 2-3:1.0 eth1: This product is covered by one or more of the following patents:

root@DS218:~# synonet
Copyright (c) 2003-2019 Synology Inc. All rights reserved.

Usage: synonet (Version 24922)
        --help
        --show
        --dhcp iface
        --manual iface ip mask
        --check ip {mask}
        --la_new bondX {1|2} ethX ethX
        --la_del bondX
        --la_enum
        --la_get bondX
        --la_exist
        --set_gateway gateway
        --set_dns dns
        --dns_update
        --set_mtu iface MTU
        --set_hostname new_hostname
        --get_hostname
        --vlan_enum
        --vlan_create device vlan_id {dhcp| ip mask}
        --vlan_remove vlan_interface
        --vlan_exist
        --set_pppoe iface username password
        --get_if pppoe
          (only support pppoe now)
        --wake xx:xx:xx:xx:xx:xx ethX
        --get_block ethX
        --set_block ethX {ha|aha|virtualization|mailplus} {true|false}
        --dhcp_server_change_interface old_interface new_interface
        --get_mac_addr iface {original | ha}
        --set_ip -4/-6 ethx {flush | add | del} {empty | ip/prefix_len}
                ex1     : --set_ip -4 eth0 add 192.168.55.66/16
                ex2     : --set_ip -6 eth0 del 2001:b011:4002:f0:211:32ff:fe22:efcb/64
                ex3     : --set_ip -4 eth0 flush
                ex4     : --set_ip -6 eth0 flush
                note    : it's different from --manual, ifcfg doesn't modified
        --ipv6-init (Do NOT use this after system booting)

bb-qq, 2019-10-08:
I found out that synnonet is unnecessary, so you can configure IP settings from WebGUI after up the interfaces by "ifconfig ethX up" command.

If you have an issue of link-speed, please try to disable auto-negotiation on PC side. (in case of direct connection)

aetaf, 2019-10-08:
Thank you, how to boot recognition?

aetaf, 2019-10-08:
root@DS218:~# dmesg | grep -i 8152
[   90.520673] r8152: module verification failed: signature and/or required key missing - tainting kernel
[   90.650959] usbcore: registered new interface driver r8152
[   90.925962] r8152 2-2:1.0 eth1: v2.12.0 (2019/04/29)
[   90.931546] r8152 2-2:1.0 eth1: This product is covered by one or more of the following patents:
[ 1708.401014] r8152 2-3:1.0 eth1: v2.12.0 (2019/04/29)
[ 1708.406659] r8152 2-3:1.0 eth1: This product is covered by one or more of the following patents:
[ 4283.871200] r8152 2-3:1.0 eth1: v2.12.0 (2019/04/29)
[ 4283.876916] r8152 2-3:1.0 eth1: This product is covered by one or more of the following patents:
[ 4918.300791] usbcore: deregistering interface driver r8152
[ 4929.515955] r8152 2-3:1.0 eth1: v2.12.0 (2019/04/29)
[ 4929.522280] r8152 2-3:1.0 eth1: This product is covered by one or more of the following patents:
[ 4929.544160] usbcore: registered new interface driver r8152
[86770.461007] r8152 2-3:1.0 eth1: v2.12.0 (2019/04/29)
[86770.466581] r8152 2-3:1.0 eth1: This product is covered by one or more of the following patents:

bb-qq, 2019-10-10:
After running ``ifconfig eth1 up``, you can configure IP address by Web browser. Currently we need this procedure every time when the system boot up.

bb-qq, 2020-01-26:
This issue is closed due to timeout. Please feel free to reopen the issue if you have additional questions.

closed 2020-01-26 by bb-qq