bb-qq/r8152 #6: DS3617xs  Unable to install

LunaticTian, 2019-10-10:
The notebook cpu I am using is I54200. With DS3617, I can't install it in Package Center. I don't support this file.

LunaticTian, 2019-10-10:
Thank you very much, I have successfully installed, I have correctly installed the 8153 device and the system has successfully read it,but I can't use ip normally.

root@DiskStation:~# ifconfig eth1 up
eth1: ERROR while getting interface flags: No such device

bb-qq, 2019-10-10:
What is DSM version? This module is built for version 6.2.

> [ 49.302979] r8152: version magic '3.10.105 SMP mod_unload ' should be '3.10.102 SMP mod_unload '

It seems that kernel version is older than expected.

LunaticTian, 2019-10-11:
Ok,this is my problem, thank you for your answer.

closed 2019-10-11 by LunaticTian