bb-qq/r8152 #7: DS412+ (Atom D2700)

Siress, 2020-01-02:
### Description of the problem

I am not adept in the wizardly ways of creating drivers, and would very much like to have a 2.5 or 5Gbps network connection with my NAS. 

### Description of your products

DS412+
Linux COMARC 3.10.105 #24922 SMP Wed Jul 3 16:35:48 CST 2019 x86_64 GNU/Linux synology_cedarview_412+
DSM 6.2.2
Planned: CableCreation PN B07VNFLTLD.

### Description of your environment

Planned: Mikrotik RS305-1G-4S+IN switch with an ipolex 10G-SFP-T transceiver.
Planned: Mellanox Connectx-2 
CAT6

### Output of ``dmesg`` command

````
Far too much data for me to parse through for any potential doxing info... If necessary, I can email it directly to you.
````

### Output of ``lsusb`` command

````
|__usb1          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (ehci_hcd 0000:00:1a.7) hub
|__usb2          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (ehci_hcd 0000:00:1d.7) hub
  |__2-1         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 650043CB10B76D76)
|__usb3          1d6b:0001:0310 09  1.10   12MBit/s 0mA 1IF  (uhci_hcd 0000:00:1a.0) hub
|__usb4          1d6b:0001:0310 09  1.10   12MBit/s 0mA 1IF  (uhci_hcd 0000:00:1d.0) hub
|__usb5          1d6b:0001:0310 09  1.10   12MBit/s 0mA 1IF  (uhci_hcd 0000:00:1d.1) hub
|__usb6          1d6b:0001:0310 09  1.10   12MBit/s 0mA 1IF  (uhci_hcd 0000:00:1d.2) hub
|__usb7          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (Linux 3.10.105 etxhci_hcd-170202 Etron xHCI Host Controller 0000:04:00.0) hub
  |__7-1         051d:0002:0090 00  2.00   12MBit/s 2mA 1IF  (American Power Conversion Back-UPS RS 1300G FW:863.L5 .D USB FW:L5  4B1410P56556  )
|__usb8          1d6b:0003:0310 09  3.00 5000MBit/s 0mA 1IF  (Linux 3.10.105 etxhci_hcd-170202 Etron xHCI Host Controller 0000:04:00.0) hub

````

bb-qq, 2020-01-03:
Since the controller chip of B07VNFLTLD is RTL8156, the dongle should work with my driver.
However please note that NBaseT(2.5G/5G) needs dedicated transceivers for it. Probably 10G-SFP-T does not support 2.5/5G.

Siress, 2020-01-04:
Thank you for that information! I will give it a shot and report back.

Siress, 2020-01-17:
### **Description of the problem**
Confused about how to configure the IP. Should it be listed as a DSM Network Interface? r8152-cedarview-2.12.0-1.spk is running, B07VNFLTLD is recognized per lsusb, but I have no idea where/how to access the Web UI for step 3.

### **Description of your products**
DS412+
Linux 3.10.105 #24922 SMP Wed Jul 3 16:35:48 CST 2019 x86_64 GNU/Linux synology_cedarview_412+
DSM 6.2.2
CableCreation PN B07VNFLTLD.

### **Description of your environment**

Planned: Mikrotik RS305-1G-4S+IN switch with an ipolex 10G-SFP-T transceiver.
Planned: Mellanox Connectx-2
CAT6

**### Output of lsusb command**

|__usb1          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (ehci_hcd 0000:00:1                                                 a.7) hub
|__usb2          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (ehci_hcd 0000:00:1                                                 d.7) hub
  |__2-1         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskSta                                                 tion 650043CB10B76D76)
|__usb3          1d6b:0001:0310 09  1.10   12MBit/s 0mA 1IF  (uhci_hcd 0000:00:1                                                 a.0) hub
|__usb4          1d6b:0001:0310 09  1.10   12MBit/s 0mA 1IF  (uhci_hcd 0000:00:1                                                 d.0) hub
|__usb5          1d6b:0001:0310 09  1.10   12MBit/s 0mA 1IF  (uhci_hcd 0000:00:1                                                 d.1) hub
|__usb6          1d6b:0001:0310 09  1.10   12MBit/s 0mA 1IF  (uhci_hcd 0000:00:1                                                 d.2) hub
|__usb7          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (Linux 3.10.105 etx                                                 hci_hcd-170202 Etron xHCI Host Controller 0000:04:00.0) hub
  |__7-1         051d:0002:0090 00  2.00   12MBit/s 2mA 1IF  (American Power Con                                                 version Back-UPS RS 1300G FW:863.L5 .D USB FW:L5  4B1410P56556  )
**|__usb8          1d6b:0003:0310 09  3.00 5000MBit/s 0mA 1IF  (Linux 3.10.105 etx                                                 hci_hcd-170202 Etron xHCI Host Controller 0000:04:00.0) hub
  |__8-2         0bda:8156:3000 00  3.20 5000MBit/s 512mA 1IF  (Realtek USB 10/1                                                 00/1G/2.5G LAN 000000001)**

bb-qq, 2020-01-17:
Did you run ``ifconfig eth2 up`` and is it succeeded?
Please confirm the result of ``ifconfig -a``.

Siress, 2020-01-17:
I did run the command, but it did not list eth2 after "ifconfig -a".

bb-qq, 2020-01-17:
Hmm, it's weird. VID/PID shown in the result of ``lsusb`` is certainly supported by the driver.

So could you let me know the result of ``dmesg``? The required part is only about detection of the dongle.

That can be obtained by the following steps:
1. detach the ethernet dongle
2. run ``dmesg``
3. attach the ethernet dongle
4. run ``dmesg`` and check increased part from previous output

Siress, 2020-01-18:
[77414.866474] usb 8-2: new SuperSpeed USB device number 3 using
etxhci_hcd-170202
[77414.894713] etxhci_hcd-170202 0000:04:00.0: WARN Event TRB for slot 2 ep
2 with no TDs queued?
[77414.903531] etxhci_hcd-170202 0000:04:00.0: WARN Event TRB for slot 2 ep
3 with no TDs queued?
[77414.912282] etxhci_hcd-170202 0000:04:00.0: WARN Event TRB for slot 2 ep
6 with no TDs queued?
[77416.023156] usb 8-2: reset SuperSpeed USB device number 3 using
etxhci_hcd-170202
[77416.580569] r8152 8-2:1.0 eth2: v2.12.0 (2019/04/29)
[77416.585547] r8152 8-2:1.0 eth2: This product is covered by one or more
of the following patents:
                        US6,570,884, US6,115,776, and US6,327,625.

Siress, 2020-01-18:
Also, I just ran "ifconfig eth2 up" again and received this message:
"SIOCSIFFLAGS: Operation not permitted". I then elevated to root and ran it
again. As root, no error message was generated. Logging out as root and
repeating the command as admin once again led to the SIOCSIFFLAGS message.
I originally ran the command as root, so perhaps this has happened both
times.  Risky I guess, but I *just* archived my entire NAS anyway.

bb-qq, 2020-01-19:
Can you assign an IP address after that?

Siress, 2020-01-19:
YES! I don't know what changed, but it now shows up as LAN3 and I can
assign the IP manually. Thank you so much for your work and for helping me
get this to work for my system. I've connected it to the Mikrotik switch
via IPOLEX 10G-SFP-T to CAT6 and DSM reports it as "2500 Mbps full-duplex,
MTU 1500". I'm still waiting on an SFP+ NIC for my computer, tho.

This is a huge accomplishment on your part. I hope people recognize the
work you've done. Do you have a donation page anywhere?

bb-qq, 2020-01-22:
It's my pleasure. 

> Do you have a donation page anywhere?

I have created a donation page for the experiment. Please take a look if you feel like it.

Siress, 2020-01-24:
Excellent, thank you. I have one more issue that I resolved myself. This is more of an FYI. 

Here's how I left my network yesterday. For brevity, all transceivers (Tx) are IOPLEX 10G-SFP-T models.

- Router (Asus RT-AX55U) connected to Syno_LAN1 and Mikrotik console port.
- USB RTL8156 device (LAN3) from Syno to Tx, to Mikrotik.
- PC Mellanox Connectx-2 card, Tx to Tx, to Mikrotik. 

I had just connected the system together and configured the IP for LAN3 in DSM, verified that the IP was reporting, and then tried to get iperf3 running. I failed to get iperf3 to do anything except hang in terminal, and then left the system for ~24hours. Upon return, LAN3 had disappeared. The lights on the RTL8156 device were still flashing (white and orange, indicating 2.5GbE connection?). My system did not experience a power failure. It's connected to a UPS and neither system reported an outage. I was able to restore the device to active state by doing the following steps, though I do not know if all steps were required. 1) I stopped and started the RTL driver through Package Center. 2) I ran `ifconfig eth2 up` as root. Again, the SIOCSIFFLAGS error was thrown when executing the command as admin, and did not appear to have any effect. As root, LAN3 immediately came back to life. Interestingly, perhaps, the network settings for LAN3 were retained.

Could we perhaps have a user-script in Task Scheduler that checks that eth2 us active and, if not, run the command?

bb-qq, 2020-01-24:
You may want to try the standard Gigabit Ethernet connection first. If it works, there seems to be a problem with the NBASE-T environment or thermal issue.

If it doesn't work, try removing other USB devices.

If that doesn't work too, reloading the driver may be the last solution:
`grep down /sys/class/net/eth2/operstate && synoservice pkgctl-r8152 --restart` should invoke restarting the driver when the link state is down.

Also, thank you for your donation!

Siress, 2020-01-25:
After my last post, I started using the RTL as the only gateway. I changed it's IP to my default IP for the NAS and unplugged LAN1. The connection has been reliable for the past couple of days. It sustained itself even through a scheduled sleep cycle. 

Thank you for the script!

I don't believe the rest of this has anything to do with you work. I'm really just sharing it because it confirms your suspicion, and explains why I haven't verified that the setup hit the intended performance goal.

I finally had time to get iperf3 working and it's confirming your suspicion about the Tx. After you brought up your concern with 10GBase-T transceivers operating as 2.5/5 GBase-T speeds, I was sure to purchase a model of Tx that supposedly supports 2.5/5/10 GBase-T. Considering I'm not even hitting 100 Mbps speeds when it's connected to anything other than a matching Tx, I think it's fairly conclusive that it's "support" means that there is a connection...but only just barely.

Here's the summary. When I say "no Tx", that means I'm using the built-in RJ45 console port:
PC to Mikrotik (10GbE, same Tx on both ends), then Mikrotik to...
...Router (GbE, no Tx), Router to LAN1 (GbE): Saturated, ~950 Mbps
...Router (GbE, no Tx), Router to RTL (presumably GbE): Saturated, ~915 Mbps
...RTL (presumably 2.5 GbE, includes Tx): Choked, ~87.5 Mbps
...LAN1 (GbE, includes Tx): Choked ~32.5 Mbps
...LAN1 (GbE, no Tx): ~945 Mbps

Guess I'll add a 2.5GBase-T Tx and hope for the best.

bb-qq, 2020-01-26:
Hmm, it is a difficult situation...

What is your MTU value? Because of USB based ethernet drivers consume more CPU and the USB protocol has a large amount of overhead, enabling Jumbo-frame is required to achieve wire-speed in some cases. 

> Further testing the Tx, I connected the PC (Tx) to the RTL directly. Windows reported 10 Gbe...so that's not to be trusted... and iperf resulted in 73 to 400 Mbps (avg. of 264).

Can you fix link-speed to 1G form the Device Manager on the PC side?

Siress, 2020-01-26:
I connected the PC (Tx) to RTL again for these values:
DSM for LAN3 (RTL): 2500 Mbps, Full duplex, MTU 1500
Win10 for Mellanox: 10 Gbps, Jumbo Packet size of _1514_  

I haven't found a way to change the link-speed of the Mellanox card. 

I varied the paired jumbo packet size:
- 1500 => 384 Mbps (182 to 515).
- 2000 => 288 Mbps (148 - 470).
- 3000 =>  7 Mbps (1 - 29).
- 9000 => 2 Mbps (1 to 4).
- 1500 => 174 Mbps (69 - 338); 2nd run => 325 Mbps (195 - 520).

I then varied the Mellanox Rx/Tx Buffers (256 to 4096 range per). Starting point: Rx/Tx buffer 4096/2048 
- 4096/4096 => 233 Mbps (89 - 325)
- 1024/1024 =>  48 Mbps (29 - 117)

- 9k MTU and 4096/4096 => 2 Mbps (1-4)
- 1.5k MTU on RTL and 9k MTU on Mellanox, 4096/4096 buffers => 233 Mbps (80-392)
- 2k MTU on RTL ... => 312 Mbps (176-504)
- 3k MTU on RTL ... => 4 Mbps (1-14); 2nd run => 4 (1-15)

Returned all to 1500 MTU, buffers to default 4096/2048
=> 131 Mbps (35 - 241); 2nd run => 269 Mbps (52 - 380).

Siress, 2020-01-26:
I found a specification sheet for the Tx I've been using. Despite what I had previously read, it is not compatible with 2.5 / 5 GBase-T. I've ordered a FS(.com) 10G-T-SFP-PLUS that explicitly states compatibility with 1/2.5/5/10 GBase-T in the specification sheet.

bb-qq, 2020-01-27:
Do you have another PC? Confirming PC to PC communication should help to isolate the problem.

Also, I noticed that Connectx-2 does not support NBASE-T because the device was released before publishing IEEE  802.3bz. We can only use 1Gigabit link under direct connection.

legion7298, 2020-01-29:
> Can you assign an IP address after that?

yeah~ after "ifconfig eth3 up", goto your dsm webpage and go control panel -> network panel.
Then you can see new Lan number what you added. And right click on Lan3 and go IPV4.
Then check the get auto dhcp address.

Siress, 2020-01-30:
I received the new Tx ([10G-T-SFP-PLUS](https://img-en.fs.com/file/datasheet/10g-t-sfp-plus.pdf)) today. I installed it into the Mikrotik, then connected it to the RTL (Syno) and ran iperf. No dice. ~88 Mbps to the Mellanox/desktop.

> Do you have another PC? Confirming PC to PC communication should help to isolate the problem.

I have a laptop with GbE and USB 3.0 ports, but nothing I can plug the Mellanox card into. I connected the laptops GbE port into the RTL (Syno) and received GbE performance as confirmed by NIC status and iperf3 to my desktop. I then moved the RTL to a USB 3 port on my laptop and attached it to the new Tx in the Mikrotik. Iperf3 from my desktop... 69 Mbps. Sigh.

> Also, I noticed that Connectx-2 does not support NBASE-T because the device was released before publishing IEEE 802.3bz. We can only use 1Gigabit link under direct connection.

That is a good catch. I haven't established a connection with the CRS305 interface yet, but I hope to find a way to run speed tests directly to/from it when I have time.

bb-qq, 2020-01-31:
> I have a laptop with GbE and USB 3.0 ports, but nothing I can plug the Mellanox card into. I connected the laptops GbE port into the RTL (Syno) and received GbE performance as confirmed by NIC [...]

This result seems to indicate that CRS305 has an issue with 2.5G/5G, but [this user reported it works](https://www.reddit.com/r/mikrotik/comments/dz845w/crs30x_series_support_for_5g_and_25g/f8bf1pg/). It's strange...

By the way, are you using the latest firmware? If not, it might be worth to update.

Siress, 2020-02-07:
Well, I've now acquired the S+RJ10 transceiver and given it a whirl. It's at least negotiating and saturating 1GbE speeds now, so that's an improvement. I've also upgraded the firmware on the MikroTik CRS305. Per MikroTik, the CRS305 with S+RJ10 should auto-negotiate 1/2.5/5/10 GBase-T. The cable works at 10G speeds - verified. Not sure what's going on now... If I turn auto-negotiate off on the CRS305 then I lose the options for 2.5/5 G when using the WinBox gui. Not sure if I can get it run via command line but I'll try that next time I'm digging into it. 

BTW - it seems there's some recurrence of the driver package not responding after the attached RTL device has been left disconnected for a time. The method to restart the driver from before still works fine (restart driver, issue the 'ifconfig eth2 up" command as root).

mervincm, 2020-03-31:
I can confirm using S+RJ10 Mikrotik transceiver I get 2.5g auto-negotiated. please notethat unless you have the latest version of this transceiver you can not use jumbo frames at all, its a known bug.

Siress, 2020-04-01:
Thank you for that information. I will turn them off when I return to debugging.

bb-qq, 2020-05-22:
Did it go well?

Also please make sure using the front port because some users reported instability issues when they use rear ports.

Siress, 2020-05-22:
Hey! Thanks for checking in. I have not yet returned to this project - even with the quarantine...! 

Your comment about the front USB port is not feasible. That is a USB 2.0 port. As such, it is limited to a theoretical maximum of 480 Mbps; a far cry from the target 2500 Mbps.

bb-qq, 2020-06-27:
I concluded Synology devices have a signal quality issue on rear USB ports from the fact while the front USB port is soldered on the main PCB, the rear USB port is not. ([DS918+ tear down](https://www.ifixit.com/Guide/SYNOLOGY+DS918++:+a+complete+disassembly+in+rules/98384))
It might be work if we connect the device to the header pins on the main PCB directly, but it is not realistic...

I will update the document to mention this.

NuKeMtje, 2021-08-06:
### Context
Model(platform): DS412+ (Cedarview) with 2048 MB RAM upgrade
DSM version: 6.2.4-25556
Driver version: 2.15.0-1
Ethernet device: ASUSTOR AS-U2.5G2 (so with 'newer' RLT8156B)
Working status: Works BUT
- does not automatically show in DS GUI (I have to manually do a sudo ipconfig eth2 up)
- slower than build in ethernet ports! CPU/USB bottleneck?
- worked only in the top rear USB3.0 port on my DS412+

Advice: Don't spend your money on this if you have a older DS412+ model or don't get your hopes up

### lusb output
```
 lsusb
|__usb1          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (ehci_hcd 0000:00:1a.7) hub
|__usb2          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (ehci_hcd 0000:00:1d.7) hub
  |__2-1         f400:f400:0100 00  2.00  480MBit/s 98mA 1IF  (Synology Diskstation xxx)
|__usb3          1d6b:0001:0310 09  1.10   12MBit/s 0mA 1IF  (uhci_hcd 0000:00:1a.0) hub
|__usb4          1d6b:0001:0310 09  1.10   12MBit/s 0mA 1IF  (uhci_hcd 0000:00:1d.0) hub
|__usb5          1d6b:0001:0310 09  1.10   12MBit/s 0mA 1IF  (uhci_hcd 0000:00:1d.1) hub
|__usb6          1d6b:0001:0310 09  1.10   12MBit/s 0mA 1IF  (uhci_hcd 0000:00:1d.2) hub
|__usb7          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (xhci_hcd 0000:04:00.0) hub
|__usb8          1d6b:0003:0310 09  3.00 5000MBit/s 0mA 1IF  (xhci_hcd 0000:04:00.0) hub
  |__8-1         0bda:8156:3100 00  3.20 5000MBit/s 256mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 001000001)
```

### iperf3 tests
Connected to a Mikrotik S+RJ10 v1 in a Miktrotik CRS309.
Verified the module works at 2.5Gbit/s on a windows PC getting +2Gbit/s iperf speed results.
I also want to note the previous version of de drivers (2.14.0-3) was working a little bit faster (not more than 100Mbit/s or so)
All tests ran from a TrueNAS system which handles 10Gbit/s fine.

```
 iperf3 -c 192.168.10.181
Connecting to host 192.168.10.181, port 5201
[  5] local 192.168.10.23 port 55300 connected to 192.168.10.181 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  75.1 MBytes   630 Mbits/sec    5   58.4 KBytes
[  5]   1.00-2.00   sec  66.9 MBytes   561 Mbits/sec    5   68.4 KBytes
[  5]   2.00-3.00   sec  70.3 MBytes   590 Mbits/sec    7   52.7 KBytes
[  5]   3.00-4.00   sec  74.5 MBytes   625 Mbits/sec    6   68.3 KBytes
[  5]   4.00-5.00   sec  69.7 MBytes   585 Mbits/sec    8   59.8 KBytes
[  5]   5.00-6.00   sec  75.0 MBytes   629 Mbits/sec    9   71.2 KBytes
[  5]   6.00-7.00   sec  74.4 MBytes   624 Mbits/sec    7   54.1 KBytes
[  5]   7.00-8.00   sec  70.3 MBytes   590 Mbits/sec    6   61.2 KBytes
[  5]   8.00-9.00   sec  75.7 MBytes   635 Mbits/sec    6   46.9 KBytes
[  5]   9.00-10.00  sec  71.2 MBytes   597 Mbits/sec    5   61.2 KBytes
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   723 MBytes   607 Mbits/sec   64             sender
[  5]   0.00-10.15  sec   723 MBytes   598 Mbits/sec                  receiver

 iperf3 -c 192.168.10.181 -P2
Connecting to host 192.168.10.181, port 5201
[  5] local 192.168.10.23 port 64470 connected to 192.168.10.181 port 5201
[  7] local 192.168.10.23 port 40467 connected to 192.168.10.181 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  64.8 MBytes   544 Mbits/sec    8   37.0 KBytes
[  7]   0.00-1.00   sec  13.3 MBytes   111 Mbits/sec   19   17.1 KBytes
[SUM]   0.00-1.00   sec  78.1 MBytes   655 Mbits/sec   27
- - - - - - - - - - - - - - - - - - - - - - - - -
[  5]   1.00-2.00   sec  62.4 MBytes   523 Mbits/sec   13   57.0 KBytes
[  7]   1.00-2.00   sec  18.1 MBytes   152 Mbits/sec   30   15.7 KBytes
[SUM]   1.00-2.00   sec  80.4 MBytes   675 Mbits/sec   43
- - - - - - - - - - - - - - - - - - - - - - - - -
[  5]   2.00-3.00   sec  64.8 MBytes   543 Mbits/sec    9   45.6 KBytes
[  7]   2.00-3.00   sec  13.7 MBytes   115 Mbits/sec   33   14.2 KBytes
[SUM]   2.00-3.00   sec  78.5 MBytes   659 Mbits/sec   42
- - - - - - - - - - - - - - - - - - - - - - - - -
[... 88 lines cut ...]
[SUM]   6.00-7.00   sec  85.1 MBytes   714 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[  5]   7.00-8.00   sec  42.4 MBytes   356 Mbits/sec
[  7]   7.00-8.00   sec  42.5 MBytes   356 Mbits/sec
[SUM]   7.00-8.00   sec  84.9 MBytes   712 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[  5]   8.00-9.00   sec  42.7 MBytes   358 Mbits/sec
[  7]   8.00-9.00   sec  42.8 MBytes   359 Mbits/sec
[SUM]   8.00-9.00   sec  85.4 MBytes   716 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[  5]   9.00-10.00  sec  44.8 MBytes   376 Mbits/sec
[  7]   9.00-10.00  sec  39.9 MBytes   335 Mbits/sec
[SUM]   9.00-10.00  sec  84.7 MBytes   710 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.15  sec   428 MBytes   354 Mbits/sec    0             sender
[  5]   0.00-10.00  sec   427 MBytes   358 Mbits/sec                  receiver
[  7]   0.00-10.15  sec   424 MBytes   350 Mbits/sec    0             sender
[  7]   0.00-10.00  sec   423 MBytes   355 Mbits/sec                  receiver
[SUM]   0.00-10.15  sec   852 MBytes   704 Mbits/sec    0             sender
[SUM]   0.00-10.00  sec   850 MBytes   713 Mbits/sec                  receiver
```
Nothing to write home about.
Total CPU goes up 20-30% during tests.
During SMB transfer CPU goes to 90% at a speed of 60MB/s

Siress, 2023-08-30:
Poked at this again. It's remarkable that this little NAS is still kicking after 11 years of continuous operation. 

Revisiting #7

Constants:
- Model(platform): DS412+ (cedarview)
- DSM version: 6.2-25556
- Driver version: 2.12.0-1  (last cedarview compile without the warning for DSM6.2 users)
- Link speed reported = 2500 Mbps
- Duplex = Full
- Defined client and host IPv4 address on LAN. Host IP set as gateway 
- Test method = iperf3 over certified CAT6 cable to a TrueNAS Scale custom build with Intel 226-V NIC; same port every time.
	
	
Design of Experiments:
- USB RTL8152-based Ethernet device. I have two from CableCreation. A "Wide" and "Slim" 2.5G version with blue USB type A connectors.
- USB Port Used. There are two viable options for >1GbE; rear USB port top and bottom.
- MTU size. 9 options.
- Negotiation. Either Synology is negotiating MTU Manually ("Manual") or automatically ("Auto") while being set by the other device.
- Total Experiments = 72
- Time to reconfigure Auto MTU = ~10sec.
- Time to reconfigure manual MTU = ~5min. (see tip 2)

Learnings / Tips
1. Use telnet into the Synology device. SSH gets dropped when changing MTU, telnet does not.
2. Only change MTU in the non-Syno device and let Syno autonegotiate MTU size. Even though the 'network status' reports MTU size 1500, editing the LAN reveals MTU in greyed out box as 2000. However, it consistently has the best performance if you just let it autonegotiate.

**Results**
It is possible to obtain 1/2 of the desired 2.5GbE speed with iperf3, but there are issues preventing this from being useful at this time.

Issues
1. There's some sort of memory issue. I frequently had eth2 drop with a "SIOCSIFFLANGS: Cannot allocate memory" error. After adding a 2nd 2GB RAM stick (many years ago) and now having proper backups, I finally decided to make the [community-recommended changes](https://community.synology.com/enu/forum/1/post/128338?ref=jussiroine.com) to optimize how the system uses the 4GB of RAM. This has not resulted in keeping the eth driver up. 
2. The connection sometimes starts off with ~600 Mbps and lots of dropped packets. I believe this is largely due to the memory issue (Learning 3) and was resolved by optimizing memory usage. Also, running the iper3 test multiple times often resolved the issue. It's possible this is a minor issue that resolves itself after a few minutes.