bb-qq/r8152 #8: Driver r8152 for Synology DS216play

leriton10, 2020-01-18:
### Description of the problem

Is it possible to get the driver for a Synology DS216play ? It seems to be a monaco procesor 

### Description of your products

> - the product name of your NAS model : 
DS216play
>     - and output of ``uname -a`` command
Linux DSLaurent 3.10.105 #24922 SMP Tue Jul 9 18:27:16 CST 2019 armv7l GNU/Linux synology_monaco_ds216play

>     - DSM version x.x
DSM 6.2.2-24922 Update 4
> - the product name of the ethernet dongle
RTL8152 Fast Ethernet Adapter - Realtek Semiconductor Corp.

### Description of your environment

> - connect directly to a PC or using a hub
Device connected directly to the Synology
>     - PC: NIC model name or chipset name
RTL8152 Fast Ethernet Adapter - Realtek Semiconductor Corp.
>     - Hub: model name
No Hub
> - cable category (Cat.5e or Cat.6 or Cat.7)
Cat.6

### Output of ``dmesg`` command

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 3.10.105 (root@build1) (gcc version 4.9.3 20150311 (prerelease) (crosstool-NG 1.20.0) ) #24922 SMP Tue Jul 9 18:27:16 CST 2019
[    0.000000] CPU: ARMv7 Processor [413fc090] revision 0 (ARMv7), cr=10c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] Machine: STi SoC with Flattened Device Tree, model: STiH410 B2120
[    0.000000] Ignoring memory below PHYS_OFFSET: 0x40000000-0x43000000
[    0.000000] Ignoring memory below PHYS_OFFSET: 0x42000000-0x43000000
[    0.000000] cma: CMA: reserved 16 MiB at 5f000000
[    0.000000] Memory policy: ECC disabled, Data cache writealloc
[    0.000000] On node 0 totalpages: 118784
[    0.000000] free_area_init_node: node 0, pgdat 806ddf40, node_mem_map 8073b000
[    0.000000]   Normal zone: 928 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 118784 pages, LIFO batch:31
[    0.000000] PERCPU: Embedded 7 pages/cpu @80b29000 s6976 r8192 d13504 u32768
[    0.000000] pcpu-alloc: s6976 r8192 d13504 u32768 alloc=8*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 117856
[    0.000000] Kernel command line: console=ttyS0,115200 CONSOLE=/dev/ttyS0 ip=off mtdparts=stm-nand-bch.0:0x500000(boot),-(rfs) bpa2parts=BPA2_Region|aud-coded|aud-transcoded|vid-coded|vid-transcoded|vid-raw-input|vid-output-0|vid-decimated-1|vid-copied-1|vid-macroblock-0|vid-extra-data-0|vid-extra-data-1|v4l2-coded-video-buffers|vid-enc-data-0|vid-enc-scaled-0|vid-encoded-0|v4l2-clut|v4l2-grab|gfx-memory|aud-codec-data|vid-codec-data|aud-output-0|aud-output-1|vid-output-1|vid-decimated-0|vid-copied-0|vid-macroblock-1|blitter|vid-enc-data-1|vid-enc-scaled-1|vid-encoded-1|coredisplay-video|multicom|hades-l3|BPA2_Region0|BPA2_Region1:480M:0x62000000,v4l2-vb2|v4l2-stmc8jpg|v4l2-stmvout|v4l2-compo:24M:0x60800000,coprocs:32M:0x40000000,modules|audio:8M:0x60000000,bigphysarea|te-buffers|m2m-ca3:48M:0x48000000,STMCReserve:32768:0x47FF8000 vmalloc=775m mem=503316480@0x42000000 root=/dev/md0 rw syno_hw_version=DS216play hd_power_on_seq=2 ihd_num=2 netif_num=1 flash_size=8 phys_memsize=1024 rev=r1
[    0.000000] bpa2: partition 'BPA2_Region' created at 0x62000000 (high), size 491520 kB (0x1e000000 B)
[    0.000000] bpa2: partition 'v4l2-vb2' created at 0x60800000 (high), size 24576 kB (0x01800000 B)
[    0.000000] bpa2: partition 'coprocs' created at 0x40000000 (high), size 32768 kB (0x02000000 B)
[    0.000000] bpa2: partition 'modules' created at 0x60000000 (high), size 8192 kB (0x00800000 B)
[    0.000000] bpa2: partition 'bigphysarea' created at 0x48000000 (low), size 49152 kB (0x03000000 B)
[    0.000000] bpa2: partition 'STMCReserve' created at 0x47ff8000 (low), size 32 kB (0x00008000 B)
[    0.000000] Synology Hardware Version: DS216play-j
[    0.000000] Internal HD num: 2
[    0.000000] Internal netif num: 1
[    0.000000] Synology Hardware Revision: r1
[    0.000000] SYNO Transcoding Memory Mode: 3
[    0.000000] PID hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 464MB = 464MB total
[    0.000000] Memory: 393428k/393428k available, 81708k reserved, 0K highmem
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xfff00000 - 0xfffe0000   ( 896 kB)
                   vmalloc : 0x9d800000 - 0xff000000   (1560 MB)
                   lowmem  : 0x80000000 - 0x9d000000   ( 464 MB)
                   modules : 0x7f000000 - 0x80000000   (  16 MB)
[... 882 lines cut ...]
[  132.401678] usbcore: registered new interface driver snd-usb-hiface
[  229.787238] init: pkg-WebStation-userdir main process (17016) terminated with status 1
[  233.381760] init: synocontentextractd main process ended, respawning
[  777.660904] open mtd (RedBoot Config), process=syno_mem_cfg
[  777.667101] close mtd (RedBoot Config), process=syno_mem_cfg
[  777.673434] open mtd (RedBoot Config), process=syno_mem_cfg
[  777.679397] close mtd (RedBoot Config), process=syno_mem_cfg

### Output of ``lsusb`` command

|__usb1          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (ehci_hcd ehci-platform.0) hub
|__usb2          1d6b:0001:0310 09  1.10   12MBit/s 0mA 1IF  (Linux 3.10.105 ohci_hcd Generic Platform OHCI Controller ohci-platform.0) hub
  |__2-1         0403:6001:0600 00  2.00   12MBit/s 90mA 1IF  (RFXCOM RFXtrx433 A1Y0NLCE)
|__usb3          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (Linux 3.10.105 xhci-hcd xHCI Host Controller xhci-hcd.0.auto) hub
  |__3-1         0bda:8152:2000 00  2.10  480MBit/s 100mA 2IFs (Realtek USB 10/100 LAN 00E04C361AB0)
|__usb4          1d6b:0003:0310 09  3.00 5000MBit/s 0mA 1IF  (Linux 3.10.105 xhci-hcd xHCI Host Controller xhci-hcd.0.auto) hub

### Output of ``lsusb`` command

|__usb1          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (ehci_hcd ehci-platform.0) hub
|__usb2          1d6b:0001:0310 09  1.10   12MBit/s 0mA 1IF  (Linux 3.10.105 ohci_hcd Generic Platform OHCI Controller ohci-platform.0) hub
  |__2-1         0403:6001:0600 00  2.00   12MBit/s 90mA 1IF  (RFXCOM RFXtrx433 A1Y0NLCE)
|__usb3          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (Linux 3.10.105 xhci-hcd xHCI Host Controller xhci-hcd.0.auto) hub
  |__3-1         0bda:8152:2000 00  2.10  480MBit/s 100mA 2IFs (Realtek USB 10/100 LAN 00E04C361AB0)
|__usb4          1d6b:0003:0310 09  3.00 5000MBit/s 0mA 1IF  (Linux 3.10.105 xhci-hcd xHCI Host Controller xhci-hcd.0.auto) hub

bb-qq, 2020-01-19:
I added a driver for [monaco](https://github.com/bb-qq/r8152/releases/download/v2.12.0-1/r8152-monaco-2.12.0-1.spk). Please let me know the result.

leriton10, 2020-01-19:
Thank you very much !! 
No problem with the installation.
But I don't see the new interface in the Web GUI.
I tried the command "sudo ifconfig tun0 up" but nothing changed

This is what I get with ifconfig:
eth0      Link encap:Ethernet  HWaddr _xx:xx:xx:xx:xx:xx_
          inet addr:192.168.10.15  Bcast:192.168.10.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:348041 errors:0 dropped:0 overruns:0 frame:0
          TX packets:290628 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:114589544 (109.2 MiB)  TX bytes:143035205 (136.4 MiB)
          Interrupt:130

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:14298 errors:0 dropped:0 overruns:0 frame:0
          TX packets:14298 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:2466580 (2.3 MiB)  TX bytes:2466580 (2.3 MiB)

tun0      Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00
          inet addr:10.8.0.1  P-t-P:10.8.0.2  Mask:255.255.255.255
          UP POINTOPOINT RUNNING NOARP MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:100
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

thanks

bb-qq, 2020-01-19:
Did you see eth1 in the result of ``ifconfig -a``?
If not, please paste the results of ``dmesg`` and ``lsusb`` again.

leriton10, 2020-01-19:
It works !! Thank you !!
I was confuse by the tun0 interface but doing ifconf -a showed me the eth1 interface.
I just did a "sudo ifconfig eth1 up" and it appeared in the Web GUI

Thank you very much !!
Out of curiosity, how do compile your application ? I did tried to compile your driver with spksrc (https://github.com/SynoCommunity/spksrc) before bothering you but I did not succeded. 
I'm just asking because I have a pretty rare processor for a synology and I'm often block by missing release. If you don't have the time I understand.

Thanks again !!

bb-qq, 2020-01-21:
I am using these scripts to build packages.
https://github.com/SynologyOpenSource/pkgscripts-ng

And instructions to build are described here.
https://originhelp.synology.com/developer-guide/create_package/summary.html

leriton10, 2020-01-25:
Thank you very much !

bb-qq, 2020-01-26:
I believe this issue is resolved. Please feel free to reopen the issue if something remains.

closed 2020-01-26 by bb-qq

TataneLeGrand, 2023-08-08:
Hi,

I am experiencing some difficulties with my DS216play (DSM 7.2-64570 Update 3). I managed to install the driver and get an "eth1" entry when displaying the avaliable adapters with "ifconfig -a". But it does not shop up in the web interface and when I try "sudo ifconfig eth1 up" I get the following error message "SIOCSIFFLAGS: Cannot allocate memory".

Is it a reservation issue in the driver or did Synology manage to prevent us from adding an ethernet interface ?

for information the memory usage is : 
                       total        used        free            shared   buff/cache   available
Mem:         413684      216056       76796       13372      120832      153624
Swap:       2097084      194548     1902536

Best regards