bb-qq/r8152 #9: Could you please make ds1618+ (denverton) version?

legion7298, 2020-01-29:
I really really really.. need ds1618+ Version which is used denverton architecture.
Could you please make ds1618+ (denverton) version?

bb-qq, 2020-02-01:
I created the package for denverton.
https://github.com/bb-qq/r8152/releases/download/2.12.0-2/r8152-denverton-2.12.0-2.spk

Please let me know whether it worked.

legion7298, 2020-02-07:
> I created the package for denverton.
> https://github.com/bb-qq/r8152/releases/download/2.12.0-2/r8152-denverton-2.12.0-2.spk
> 
> Please let me know whether it worked.

Yes! It works! Perfectly works. Really Really Thanks. You are my hero ToT

HomerJau, 2020-02-26:
Hi, 

I'm wondering if you can make a new package to support Synology NAS with C2500 Atom for R8152 devices?

I have a DS2415+ with C2538 Atom and would dearly love to add a USB multi gig USB Ethernet adapter. There's quite a few Synology NAS models with this processor.

THX
Garry

HomerJau, 2020-02-26:
I just noticed your support the DS2415+ with the acq111. I can use that as I haven't bought a USB/Ethernet adapter yet. Thanks fo your great work on these packages!!

bb-qq, 2020-04-08:
Closing. Please feel free to re-open the issue.

closed 2020-04-08 by bb-qq