bb-qq/r8152 #10: DS1819+, USB lan lost after reboot

neoswatty, 2020-02-20:
### Description of the problem
DS1819+, USB lan device lost after reboot, but can be restored by manually stop and start the RTL driver package in package center.

### Description of your products
Linux Orion 4.4.59+ #24922 SMP PREEMPT Tue Jan 14 15:20:58 CST 2020 x86_64 GNU/Linux synology_denverton_1819+
DSM 6.22.2-24922 update 5
Asustor AS-U2.5G dongle

### Description of your environment
attached dmesg.
[dmesg.txt](https://github.com/bb-qq/r8152/files/4230344/dmesg.txt)
The symptom is that when the NAS boots up, the eth4 (USB lan) is not found by command ifconfig -a.
but after manually stop and start the RTL driver package in package center, the eth4 show up and the additional lan can be found on control panel.
In the dmesg, I tried to restart the package @ 2xx sec.

bb-qq, 2020-04-10:
This issue should be fixed by this version: https://github.com/bb-qq/r8152/releases/tag/2.12.0-3

bb-qq, 2020-04-12:
Some users reported this kind of bug is fixed. Please feel free to re-open this issue if problems remained.

closed 2020-04-12 by bb-qq