bb-qq/r8152 #11: Confirmed working Club 3D CAC-1420 with DS1815+

smashmouse, 2020-03-22:
### Description of the problem

No problem, just confirmation of working setup.

### Description of your products

synology_avoton_1815+
DSM 6.2.2-24922 Update 4
Club 3D CAC-1420

### Description of your environment
Only rested via direct link between NAS and a Mac Mini with 10Gbe aquantia chipset

### Output of ``dmesg`` command

[ 7283.598338] usb 3-2: new SuperSpeed USB device number 3 using etxhci_hcd-170202
[ 7283.627370] etxhci_hcd-170202 0000:04:00.0: WARN Event TRB for slot 1 ep 2 with no TDs queued?
[ 7283.637036] etxhci_hcd-170202 0000:04:00.0: WARN Event TRB for slot 1 ep 3 with no TDs queued?
[ 7283.646721] etxhci_hcd-170202 0000:04:00.0: WARN Event TRB for slot 1 ep 6 with no TDs queued?
[ 7286.366332] r8152 3-2:1.0 (unregistered net_device): Get ether addr fail
[ 7286.366364] usb 3-2: USB disconnect, device number 3
[ 7286.379605] r8152 3-2:1.0 eth4: v2.12.0 (2019/04/29)
[ 7286.385164] r8152 3-2:1.0 eth4: This product is covered by one or more of the following patents:
               		US6,570,884, US6,115,776, and US6,327,625.

### Output of ``lsusb`` command
Note: had to make sure I connected on USB port with the 5000mbps

|__usb1          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (ehci_hcd 0000:00:16.0) hub
  |__1-1         8087:07db:0002 09  2.00  480MBit/s 0mA 1IF  ( ffffffd1ffffffb2ffffffdbffffffad) hub
    |__1-1.1     f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 6500597B8F6DAE88)
|__usb2          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (Linux 3.10.105 etxhci_hcd-170202 Etron xHCI Host Controller 0000:04:00.0) hub
  |__2-4         051d:0002:0090 00  2.00   12MBit/s 2mA 1IF  (American Power Conversion Back-UPS RS 550G FW:857.L7 .I USB FW:L7   3B1819X27861  )
|__usb3          1d6b:0003:0310 09  3.00 5000MBit/s 0mA 1IF  (Linux 3.10.105 etxhci_hcd-170202 Etron xHCI Host Controller 0000:04:00.0) hub

### Output of ``ifconfig -a`` command

Note: had to bring eth4 up manually.

bb-qq, 2020-04-08:
Thank you for the report. 

Have you encounter a stability problem under heavy load? Some users reported this issue. If you do not have such an issue, I would like to investigate the difference from other users.

smashmouse, 2020-04-08:
I've not tested this in anger yet.  I was just confirming connection whilst I build a backup server for my NAS.

bb-qq, 2020-04-10:
Ok. Closing.

closed 2020-04-10 by bb-qq