bb-qq/r8152 #13: DS918+ w/ TUC-ET2G not binding driver

claytonmark, 2020-03-29:
### Description of the problem

> Kernel modules are loading but driver is not being bound to the device.

### Description of your products

> - DS918+
>     Linux mark-NAS 4.4.59+ #24922 SMP PREEMPT Fri May 10 02:59:42 CST 2019 x86_64 GNU/Linux synology_apollolake_918+
>     - DSM version 6.2.2
> - TUC-ET2G

### Description of your environment

> - connect directly to a PC
>     - PC: TEG-25GECTX
> - Cat.5e

### Output of ``dmesg`` command

````
[    0.000000] Command line: root=/dev/md0 earlyprintk=apl console=ttyS2,115200n8 ihd_num=4 netif_num=2 HddHotplug=1 SataPortMap=23 sata_remap=0>2:1>3:2>0:3>1 syno_hw_version=DS918+ vender_format_version=2 syno_hdd_detect=18,179,176,175 syno_hdd_enable=21,20,19,9 syno_usb_vbus_gpio=13@0000:00:15.0@1,11@0000:00:15.0@2 sn=[1 entry removed] macs=[2 entries removed]
[    0.000000] Kernel command line: root=/dev/md0 earlyprintk=apl console=ttyS2,115200n8 ihd_num=4 netif_num=2 HddHotplug=1 SataPortMap=23 sata_remap=0>2:1>3:2>0:3>1 syno_hw_version=DS918+ vender_format_version=2 syno_hdd_detect=18,179,176,175 syno_hdd_enable=21,20,19,9 syno_usb_vbus_gpio=13@0000:00:15.0@1,11@0000:00:15.0@2 sn=[1 entry removed] macs=[2 entries removed]
[    0.000000]  Gpp#13 - Host: dev_name:usb2        - Port:1 - Polarity: ACTIVE_HIGH
[    0.000000]  Gpp#13 - Host: dev_name:usb1        - Port:3 - Polarity: ACTIVE_HIGH
[    0.000000]  Gpp#11 - Host: dev_name:usb2        - Port:2 - Polarity: ACTIVE_HIGH
[    0.000000]  Gpp#11 - Host: dev_name:usb1        - Port:1 - Polarity: ACTIVE_HIGH
...
[  100.885737] usbcore: registered new interface driver usbfs
[  100.891924] usbcore: registered new interface driver hub
[  100.897963] usbcore: registered new interface driver ethub
[  100.904152] usbcore: registered new device driver usb
[  101.222164] usb usb2: We don't know the algorithms for LPM for this host, disabling LPM.
[  101.529944] usb 1-4: new high-speed USB device number 2 using xhci_hcd
[  101.806030] usb 2-2: new SuperSpeed USB device number 2 using xhci_hcd
[  132.770879] usb-storage 1-4:1.0: USB Mass Storage device detected
[  132.778285] scsi host5: usb-storage 1-4:1.0
[  132.784845] usbcore: registered new interface driver usb-storage
[  132.818027] usbcore: registered new interface driver usblp
[  132.993454] usbcore: registered new interface driver usbhid
[  133.000133] usbhid: USB HID core driver
[  158.232150] usbcore: registered new interface driver r8152
````

### Output of ``lsusb`` command

````
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.59+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation [1 entry removed])
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.59+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__2-2         20f4:e02b:3004 00  3.20 5000MBit/s 512mA 2IFs (Realtek USB 10/100/1G/2.5G LAN 400000001)
````

### Output of ``ifconfig -a`` command

````
docker0   Link encap:Ethernet  HWaddr 02:42:D7:0C:B7:3C  
          inet addr:172.17.0.1  Bcast:172.17.255.255  Mask:255.255.0.0
          inet6 addr: fe80::42:d7ff:fe0c:b73c/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:264 errors:0 dropped:0 overruns:0 frame:0
          TX packets:707 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:40501 (39.5 KiB)  TX bytes:126576 (123.6 KiB)

docker385 Link encap:Ethernet  HWaddr 5E:4B:0D:D1:50:F9  
          inet6 addr: fe80::5c4b:dff:fed1:50f9/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:19 errors:0 dropped:0 overruns:0 frame:0
          TX packets:502 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:1906 (1.8 KiB)  TX bytes:52186 (50.9 KiB)

eth0      Link encap:Ethernet  HWaddr 00:11:32:A6:B2:9D  
          inet addr:169.254.65.116  Bcast:169.254.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

eth1      Link encap:Ethernet  HWaddr 00:11:32:A6:B2:9E  
          inet addr:192.168.1.202  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: fe80::211:32ff:fea6:b29e/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:10494 errors:0 dropped:0 overruns:0 frame:0
          TX packets:99066 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:2007218 (1.9 MiB)  TX bytes:143292007 (136.6 MiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:2898 errors:0 dropped:0 overruns:0 frame:0
          TX packets:2898 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:362172 (353.6 KiB)  TX bytes:362172 (353.6 KiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

````

claytonmark, 2020-03-29:
Additional info:

usb-devices relevant output:

````
T:  Bus=02 Lev=01 Prnt=01 Port=01 Cnt=01 Dev#=  2 Spd=5000 MxCh= 0
D:  Ver= 3.20 Cls=00(>ifc ) Sub=00 Prot=00 MxPS= 9 #Cfgs=  3
P:  Vendor=20f4 ProdID=e02b Rev=30.04
S:  Manufacturer=Realtek
S:  Product=USB 10/100/1G/2.5G LAN
S:  SerialNumber=400000001
C:  #Ifs= 2 Cfg#= 2 Atr=a0 MxPwr=512mA
I:  If#= 0 Alt= 0 #EPs= 1 Cls=02(commc) Sub=0d Prot=00 Driver=(none)
I:  If#= 1 Alt= 0 #EPs= 0 Cls=0a(data ) Sub=00 Prot=01 Driver=(none)
````

Notice that the driver is (none).  I have tried defeating the boot timing by plugging in the USB after boot (and module load), but it still does not get assigned.

bb-qq, 2020-04-07:
Thank you for the information. I will upload a new version of the driver later.

bb-qq, 2020-04-10:
Please try this version: https://github.com/bb-qq/r8152/releases/tag/2.12.0-3

claytonmark, 2020-04-11:
Worked like a charm, didn't even have to add the interface manually.  Thanks.

closed 2020-04-11 by claytonmark