bb-qq/r8152 #15: Confirmed DS1019+ with Club3d usb to lan Adapter

megabyte0469, 2020-04-02:
Ok adapter works here without any problems but only on question.

The driver works only with 1 adapter. 

Is there a possibility to operate 2 adapters?

bb-qq, 2020-04-07:
The kernel module in the driver package supports multiple adapters, but the startup script does not. So you need additional steps to use them such as entering some shell commands and configure the task scheduler.

Also, you may encounter performance problems due to the USB bandwidth limit under heavy conditions.

megabyte0469, 2020-04-07:
ok thats right. usb3 max 5gbit bandwidth but two 2,5 GBIT Adapter work on USB3. 
Wich commands ?? "So you need additional steps to use them such as entering some shell commands and configure the task scheduler." ??

megabyte0469, 2020-04-10:
Test new Driver Version. Works fine here with club3d Adapter. 
Change my startup script from:
------------------------------------------
synoservice --restart pkgctl-r8152
sleep 5
/etc/rc.network restart
sleep 5
------------------------------------------
to with the new Driver:
------------------------------------------
/etc/rc.network restart
sleep 5
------------------------------------------
The "sleep 5" in "script" the new Driver works.

Ok 1 problem here open. 2 Club3 Adapter on same Diskstation.

megabyte0469, 2020-04-10:
Ok test here with 2 Club3d on same Diskstation with the new Driver Version 3 works fine.
All 4 ethernet Adapters with 1GBit connect. The bond with this Adapters to one Ethernet Adapter with 4gbit works fine.
Bond after restart Diskstation works only with the Startup script:

/etc/rc.network restart
sleep 5

closed 2020-04-10 by megabyte0469