bb-qq/r8152 #20: DS416Play success with one remark!

michelhandleitnl, 2020-04-24:
Great work, I used the Club USB-A 3.1 GEN 1 TO RJ45 2.5GB, and it worked instantly!
Only if i out a second Club adapter in the ds416play, the second usb adapter is not working.
The light blinks with interval of 30 secs, but isn't visible in /sys/class/net.
But still great work!

michelhandleitnl, 2020-04-28:
Dmesg output, the second Club 2.5 Gb nic gives the error (unregistered net_device): Get ether addr fail, there is no difference if I use a powered usb 3 hub.

`[353676.188849] usb 4-2: new SuperSpeed USB device number 7 using etxhci_hcd-170202
[353676.215086] drivers/usb/core/ethub.c (2055) Same device found. Change serial to 000000000
[353718.812702] usbcore: registered new interface driver r8152
[353718.820315] etxhci_hcd-170202 0000:04:00.0: WARN Event TRB for slot 1 ep 2 with no TDs queued?
[353718.830092] etxhci_hcd-170202 0000:04:00.0: WARN Event TRB for slot 1 ep 3 with no TDs queued?
[353718.839869] etxhci_hcd-170202 0000:04:00.0: WARN Event TRB for slot 1 ep 6 with no TDs queued?
[353718.951361] usb 2-2: reset SuperSpeed USB device number 56 using xhci_hcd
[353718.970273] usb 2-2: Parent hub missing LPM exit latency info.  Power management will be impacted.
[353718.980575] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff880024604780
[353718.990785] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8800246047c0
[353719.001039] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff880024604800
[353719.097092] r8152 2-2:1.0 eth2: v2.12.0 (2019/04/29)
[353719.102751] r8152 2-2:1.0 eth2: This product is covered by one or more of the following patents:
                                US6,570,884, US6,115,776, and US6,327,625.

[353719.952414] usb 4-2: reset SuperSpeed USB device number 7 using etxhci_hcd-170202
[353719.979416] usb 4-2: device firmware changed
[353719.984359] r8152 4-2:1.0 (unregistered net_device): Get ether addr fail
[353719.992199] r8152 4-2:1.0 eth3: v2.12.0 (2019/04/29)
[353719.997847] r8152 4-2:1.0 eth3: This product is covered by one or more of the following patents:
                                US6,570,884, US6,115,776, and US6,327,625.

[353720.015978] usb 4-2: USB disconnect, device number 7
[353723.900340] IPv6: ADDRCONF(NETDEV_UP): eth2: link is not ready
[353724.073653] IPv6: ADDRCONF(NETDEV_CHANGE): eth2: link becomes ready
[353724.084759] r8152 2-2:1.0 eth2: carrier on
[353726.467587] init: nmbd main process (12390) killed by TERM signal
[353726.679408] init: iscsi_pluginserverd main process (12369) killed by TERM signal
[353726.705189] init: iscsi_pluginengined main process (12353) killed by TERM signal
[353727.672132] init: nmbd main process (12676) killed by TERM signal
[353728.298530] init: iscsi_pluginserverd main process (12538) killed by TERM signal
[353728.370371] init: iscsi_pluginengined main process (12524) killed by TERM signal
[353731.982232] usb 4-2: new SuperSpeed USB device number 8 using etxhci_hcd-170202
[353732.007606] drivers/usb/core/ethub.c (2055) Same device found. Change serial to 000000000
[353732.024032] usb-storage 4-2:1.0: USB Mass Storage device detected
[353732.034318] scsi12 : usb-storage 4-2:1.0
[353733.041434] scsi 12:0:0:0: CD-ROM            Realtek  USB CD-ROM               2.00 PQ: 0 ANSI: 0
[353733.051671] scsi 12:0:0:0: Attached scsi generic sg5 type 5`

michelhandleitnl, 2020-05-29:
Unfortunately I tried with a powered usb 3 hub, but the same issue occurs on a ds416play.

bb-qq, 2020-06-20:
Have you tried the front port? Some users reported stability issues with the rear ports.

michelhandleitnl, 2020-06-20:
I have tried all ports but the same issue stays.

bb-qq, 2020-06-21:
I noticed that the device seemed to switch to USB-storage mode. I don't know that reason, but we may be able to switch its mode again to Ethernet using `usb_modeswitch` ([Related page|https://bbs.archlinux.org/viewtopic.php?id=228195])

Could you try this command?
`usb_modeswitch -v 0bda -p 8152 -g 3-R`

Norinatt, 2025-08-21:
It’s old topic. But where do you write that command?