bb-qq/r8152 #21: DS1019+ and uGreen 2.5G USB Ethernet Adapter (USB-C and USB3 plug) confirmed work

transformcube, 2020-05-01:
### Description of the problem

DS1019+ and uGreen 2.5G USB Ethernet Adapter (USB-C and USB3 plug) confirmed work

> Replace this line with your answer.

### Description of your products

> - the product name of your NAS model
DS1019+
>     - and output of ``uname -a`` command
4.4.59+
>     - DSM version x.x
DSM 6.2.2-24922 Update 6
> - the product name of the ethernet dongle
uGreen USB3.0 to RJ45 2.5G Ethernet Adapter
https://detail.tmall.com/item.htm?id=616242333868&spm=a1z09.2.0.0.4c152e8dX56A1u&_u=15oamkp2a03&skuId=4343251704751

uGreen USB-C to RJ45 2.5G Ethernet Adapter
https://detail.tmall.com/item.htm?id=607985548355&spm=a1z09.2.0.0.4c152e8dX56A1u&_u=15oamkp3217&skuId=4272714473099

### Description of your environment

> - connect directly to a PC or using a hub
>     - PC: NIC model name or chipset name
>     - Hub: model name
> - cable category (Cat.5e or Cat.6 or Cat.7)
Direct connect to DS1019+ (Cat.6)

### Output of ``dmesg`` command

````
(paste output here)
````
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 4.4.59+ (root@build3) (gcc version 4.9.3 20150311 (prerelease) (crosstool-NG 1.20.0) ) #24922 SMP PREEMPT Tue Jan 14 15:19:49 CST 2020
[    0.000000] Command line: root=/dev/md0 earlyprintk=apl console=ttyS2,115200n8 ihd_num=5 netif_num=2 HddHotplug=1 ahci_remap=0>3:1>4:2>0:3>1:4>2:5>5 syno_hw_version=DS1019+ vender_format_version=2 syno_hdd_detect=18,179,176,175,177 syno_hdd_enable=21,20,19,9,173 syno_usb_vbus_gpio=13@0000:00:15.0@1,11@0000:00:15.0@2 syno_spinup_group=3,2 syno_spinup_group_delay=9 sn=1910QXRRQB5M1 macs=0011329b1d9d,0011329b1d9e
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000] x86/fpu: xstate_offset[3]:  960, xstate_sizes[3]:   64
[    0.000000] x86/fpu: xstate_offset[4]: 1024, xstate_sizes[4]:   64
[    0.000000] x86/fpu: Supporting XSAVE feature 0x01: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x02: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x08: 'MPX bounds registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x10: 'MPX CSR'
[    0.000000] x86/fpu: Enabled xstate features 0x1b, context size is 1088 bytes, using 'standard' format.
[    0.000000] e820: BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x0000000000057fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000058000-0x0000000000058fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000059000-0x0000000000085fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000086000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000000fffffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000010000000-0x0000000012150fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000012151000-0x0000000067c17fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000067c18000-0x0000000067c18fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000067c19000-0x0000000067c42fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000067c43000-0x0000000067c7dfff] usable
[    0.000000] BIOS-e820: [mem 0x0000000067c7e000-0x0000000068c7dfff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000068c7e000-0x000000007822efff] usable
[    0.000000] BIOS-e820: [mem 0x000000007822f000-0x000000007aebefff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007aebf000-0x000000007afbefff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000007afbf000-0x000000007affefff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000007afff000-0x000000007affffff] usable
[    0.000000] BIOS-e820: [mem 0x000000007b000000-0x000000007fffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000d3709000-0x00000000d370afff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000e0000000-0x00000000e3ffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed01000-0x00000000fed01fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff800000-0x00000000ffffffff] reserved
[... 922 lines cut ...]

eth2      Link encap:Ethernet  HWaddr 2C:16:DB:A1:21:35
          inet addr:192.168.11.202  Bcast:192.168.11.255  Mask:255.255.255.0
          inet6 addr: fe80::2e16:dbff:fea1:2135/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:41709760 errors:0 dropped:0 overruns:0 frame:0
          TX packets:69123187 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:58157116001 (54.1 GiB)  TX bytes:99661845078 (92.8 GiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:5926 errors:0 dropped:0 overruns:0 frame:0
          TX packets:5926 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:836602 (816.9 KiB)  TX bytes:836602 (816.9 KiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

bb-qq, 2020-05-29:
Thank you for the report!

bb-qq, 2020-06-20:
I added this device to the supported list.

closed 2020-06-20 by bb-qq