bb-qq/r8152 #23: DS218+ Success Report

chikatana, 2020-05-17:
Working perfectly well with RTL8153 based USB card (TP-link TL-UG310)

I intended to set up a link aggregation with the onboard NIC but seems to be blocked by firmware. (create link aggregation option is missing, two LANs are detected and works perfectly fine independently)

xcray, 2020-05-22:
Thanks for your report!
Could you pls help to check which version of DSM running on your nas?
I just got a USB RTL8153 card with 3-ports hub, can't be seen in Control Panel, neither lsusb.
My nas running DSM 6.2.3-25426, while lsusb output is:

> |__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.59+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-2         051d:0002:0106 00  1.10  1.5MBit/s 0mA 1IF  (APC Back-UPS 650 FW:822.A3.I USB FW:A3 3B1816X64418  )
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 7F00957EBACA6090)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.59+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub

chikatana, 2020-05-29:
DSM 6.2.2-24922 Update 6

transformcube, 2020-05-30:
Hi, link aggregation can not get faster network speed on single device.  If you wanted faster network speed on single device use SMBv3 multi-channel.

[https://blog.azureinfra.com/2018/09/14/breaking-the-1gbps-barrier-with-synology-and-windows-10/](url)

chikatana, 2020-06-01:
> Thanks for your report!
> Could you pls help to check which version of DSM running on your nas?
> I just got a USB RTL8153 card with 3-ports hub, can't be seen in Control Panel, neither lsusb.
> My nas [...]

I'm connecting a USB TV tuner to my NAS, which is reported not working on 6.2.3
So, I'm not updating my DSM until new driver is available.
Maybe the USB NIC driver has the same issue with my USB TV tuner which only works on 6.2.2 or before

chikatana, 2020-06-01:
> 
> 
> Hi, link aggregation can not get faster network speed on single device. If you wanted faster network speed on single device use SMBv3 multi-channel.(no need to enable link aggregation settings in NAS, network switch and computer)
> 
> https://blog.azureinfra.com/2018/09/14/breaking-the-1gbps-barrier-with-synology-and-windows-10/

Thanks

chikatana, 2020-06-01:
Consider downgrading your DSM if you really need the USB-NIC to work
You are going to lose all your settings by doing this.
https://www.cliftonsystems.co.uk/how-to-downgrade-synology-dsm/

chikatana, 2020-06-01:
And you should backup all your data in advance

chikatana, 2020-06-21:
Purchased Ugreen CM275 USB 2.5 NIC (RTL8156) for DS218+ (still on DSM 6.2.2-24922 Update 6)
Using TP-Link TL-NG421 2.5Gbps PCI-e for my PC
burst transfer speed up to about 250MBps but only last for a second
steady speed at about 100-110MBps only
burst speed for the onbroad LAN is about 120MBps and steady speed about 90MBps
likely limited by my hard drives
As my DS218+ is already fully occupied by two WD red 8TB drives in RAID 1
I do not have any spare slot to insert an SSD for testing

Using 2.5Gbps NIC is only meaningful if SSD is installed or hard drives in RAID 0
My current build cannot accommodate any more drives so I won't keep using this NIC
but at least it works