bb-qq/r8152 #24: Can't trust data integrity when using this driver with an external adapter.

joelcarlton, 2020-05-21:
### Description of the problem

Wow this is a different one. I rsynced a file between two Synology's. I ran all flavors of sha sum and md5 from the root shell and confirmed the hashes were the same. What was strange is my hash-based duplicate finders were not seeing the files as the same. In my bash terminal I checked the hashes of the files mounted to my computer over smb. What was odd was the hashes matched only one of the Synology's. The hashes produced through this driver and the pluggable 2.5gb adapter were not only different but rotating each time I checked. I disconnected the share and used another IP to mount the share from an internal nic in the nas. Now the hashes match and don't rotate. Disconnected and reconnected a few times to each and confirmed the behavior. 

This is very bad. I moved 1.5TB from one nas to another and used a duplicate finder to easily remove the files from the original nas. Call me paranoid but I like to make sure I'm not deleting something that's not IDENTICAL. So out of 1.5 TB of data, my duplicate finders fond 0 files changed when using the pluggable + driver. When using the built-in nics of the computer and nas, All 1.5 TB is seen to be duplicated.

Here I was hunting down rsync issues with not actually copying the data bit for bit. I don't know that I can trust that the data I get and send through this adapter is actually intact. The whole reason I bought this nic was to speed up deduping data over smb.

Any idea what could be causing this?

### Description of your products

> - the product name of your NAS model
Linux 3.10.105 #24922 SMP Wed Jul 3 16:37:24 CST 2019 x86_64 GNU/Linux synology_avoton_1815+
DSM 6.2.2-24922-4
Plugable USBC-2500

### Description of your environment

> - connect directly to a PC or using a hub
Plugable to iMac 10.14.6

### Output of ``dmesg`` command

````
(paste output here)
````

### Output of ``lsusb`` command

````
|__usb3          1d6b:0003:0310 09  3.00 5000MBit/s 0mA 1IF  (Linux 3.10.105 etxhci_hcd-170202 Etron xHCI Host Controller 0000:04:00.0) hub
  |__3-1         0bda:8156:3000 00  3.20 5000MBit/s 512mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 000000001)
  |__3-4         06ca:b1b1:0100 00  3.00 5000MBit/s 0mA 1IF  ( 00001234567890B3)
````

### Output of ``ifconfig -a`` command

````
eth4      Link encap:Ethernet  HWaddr xx.xx.xx.xx.xx.xx
          inet addr:xx  Bcast:xx  Mask:255.255.255.252
          inet6 addr: xx
          UP BROADCAST RUNNING MULTICAST  MTU:9000  Metric:1
          RX packets:86394 errors:0 dropped:0 overruns:0 frame:0
          TX packets:38406 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:22524266 (21.4 MiB)  TX bytes:4249053 (4.0 MiB)
````

bb-qq, 2020-05-22:
I believe this corruption issue does not relate to the device driver or the device itself because TCP/IP and Ethernet has a checksum in each layer independently. If corruption is detected in a packet, the packet is discarded immediately.

I've heard that rsync has a problem with error handling when disconnecting and that you should use the --checksum and --append-verify options. Are you using these options?

Also if you are facing a disconnection issue when this driver please try to use the front port.

joelcarlton, 2020-05-22:
Maybe I didn't describe the issue properly. Rsync worked fine. In the cli of each nas the hashes matched every time with no rotation; so everything good with rsync. 

When hashing the same file on the nas using a bash terminal on another machine over mounted share is when I see the issue.

# Local NAS
- ssh NAS > teminal sha512: ...00x07

# SMB

- smb NAS built in nic > terminal local share sha512: ...00x07
- smb NAS through pluggable and r8152 > terminal local share sha512: ...043er
- smb NAS through pluggable and r8152 > terminal local share sha512: ...043er
- smb NAS through pluggable and r8152 > terminal local share sha512: ...043er
- smb NAS built in nic > terminal local share sha512: ...00x07

# AFP

- smb NAS built in nic > terminal local share sha512: ...00x07
- smb NAS built in nic > terminal local share sha512: ...00x07
- smb NAS built in nic > terminal local share sha512: ...00x07
- smb NAS through pluggable and r8152 > terminal local share sha512: ...043er
- smb NAS through pluggable and r8152 > terminal local share sha512: ...043er
- smb NAS through pluggable and r8152 > terminal local share sha512: ...043er
- smb NAS built in nic > terminal local share sha512: ...00x07
- smb NAS built in nic > terminal local share sha512: ...00x07
- smb NAS built in nic > terminal local share sha512: ...00x07

bb-qq, 2020-05-29:
How did you calculate hashes for mounted files? Is it generic programs like sha1sum/sha256sum?

transformcube, 2020-05-31:
The problem may cause by the USB3 (superspeed?) port in DS1815+ not enough power to drive the 2.5G dongle. Try to find a external USB3 superspeed Hub with power supply may help🤔.

bb-qq, 2020-06-20:
Even if there wasn't enough power, it's weird to see data corruption happen, because there are many mechanisms to guarantee data integrity in upper communication protocols. 

If the corruption also occurs with sha1sum/sha256sum, we should suspect hardware issues in CPU or RAM.

joelcarlton, 2020-06-20:
Unfortunately the 1815 does not have a USB port on the front. I could test another USB port.

bb-qq, 2020-06-27:
Could you compare the original file with the corrupted file using the `cmp` command? If there are some "pattern of corruption", it might be a hint to resolve the issue.

robloh, 2020-07-02:
I've seen this before with Synology and it turned out one of the 2 NAS' I was using had faulty memory. If you install the Windows Synology tool you can run a memory check (takes a long time, and you may have to check the log after the unit comes back up for the result: cat /var/log/messages | grep 'Memtest')

bb-qq, 2022-01-22:
I agree with the expectation that memory failure is the cause. 
There have been no other similar reports so far, so I'm closing this issue, but please feel free to reopen it if you have additional comments.

closed 2022-01-22 by bb-qq