bb-qq/r8152 #25: UGREEN-CM275 not work

liuquan321, 2020-05-23:
### Description of the problem

> Replace this line with your answer.

DS916+ an the ethernet Adapter [UGREEN-CM275](https://www.lulian.cn/product/673-cn.html) not work,CM275 on DS916 + can only work normally in 1G mode, the packet loss rate in 2.5G mode is above 70%
````
请求超时。
来自 192.168.65.56 的回复: 字节=32 时间=1ms TTL=64
请求超时。
请求超时。
来自 192.168.65.216 的回复: 无法访问目标主机。
来自 192.168.65.216 的回复: 无法访问目标主机。
请求超时。
请求超时。
请求超时。
请求超时。
请求超时。
请求超时。
请求超时。
来自 192.168.65.56 的回复: 字节=32 时间=1ms TTL=64
请求超时。
请求超时。

192.168.65.56 的 Ping 统计信息:
    数据包: 已发送 = 173，已接收 = 40，丢失 = 133 (76% 丢失)，
往返行程的估计时间(以毫秒为单位):
    最短 = 1ms，最长 = 1008ms，平均 = 73ms
Control-C
^C
````
the specific link is as follows：
````
PC(192.168.65.216) -> AX200 -> AX86U -> AX86U_2.5G_lan -> CM275 -> DS916+_usb(192.168.65.56)
````
### Description of your products

> - the product name of your NAS model
>     - and output of ``uname -a`` command
>     - DSM version 6.2.3-25426
> - the product name of the ethernet dongle

### Description of your environment

> - connect directly to a PC or using a hub
>     - PC: Intel ax200
>     - Hub: ASUS RT-AX86U
> - cable category (Cat.5e or Cat.6 or Cat.7)
Cat.5e
### Output of ``dmesg`` command

````
[57465.776490] usbcore: deregistering interface driver r8152
[57466.458147] etxhci_hcd-170202 0000:04:00.0: WARN Event TRB for slot 1 ep 3 with no TDs queued?
[57467.570385] usb 4-1: reset SuperSpeed USB device number 2 using etxhci_hcd-170202
[57468.200068] r8152 4-1:1.0 eth2: v2.12.0 (2019/04/29)
[57468.205650] r8152 4-1:1.0 eth2: This product is covered by one or more of the following patents:
               		US6,570,884, US6,115,776, and US6,327,625.
               
[57468.223283] usbcore: registered new interface driver r8152
[57468.295091] iSCSI:iscsi_target.c:612:iscsit_del_np CORE[0] - Removed Network Portal: 192.168.65.56:3260 on iSCSI/TCP
[57468.328312] iSCSI:iscsi_target.c:612:iscsit_del_np CORE[0] - Removed Network Portal: [fe80::2e16:dbff:fea1:2065]:3260 on iSCSI/TCP
[57469.496138] init: scsi_plugin_server main process (6066) killed by TERM signal
[57469.768463] init: iscsi_pluginserverd main process (24207) killed by TERM signal
[57469.836534] init: iscsi_pluginengined main process (24197) killed by TERM signal
[57470.104026] init: scsi_plugin_server main process (24225) killed by TERM signal
[57471.178215] init: synowsdiscoveryd main process ended, respawning
[57472.369190] init: iscsi_pluginserverd main process (24291) killed by TERM signal
[57472.388844] init: iscsi_pluginengined main process (24272) killed by TERM signal
[57472.480132] init: scsi_plugin_server main process (24326) killed by TERM signal
[57473.497174] IPv6: ADDRCONF(NETDEV_UP): eth2: link is not ready
[57473.557766] IPv6: ADDRCONF(NETDEV_CHANGE): eth2: link becomes ready
[57473.566920] r8152 4-1:1.0 eth2: carrier on

````

### Output of ``lsusb`` command

````
|__usb1          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (xhci_hcd 0000:00:14.0) hub
  |__1-5         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 65006B08B0E41663)
|__usb2          1d6b:0003:0310 09  3.00 5000MBit/s 0mA 1IF  (xhci_hcd 0000:00:14.0) hub
|__usb3          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (Linux 3.10.105 etxhci_hcd-170202 Etron xHCI Host Controller 0000:04:00.0) hub
|__usb4          1d6b:0003:0310 09  3.00 5000MBit/s 0mA 1IF  (Linux 3.10.105 etxhci_hcd-170202 Etron xHCI Host Controller 0000:04:00.0) hub
  |__4-1         0bda:8156:3000 00  3.20 5000MBit/s 512mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 000000001)
  |__4-2         174c:55aa:0100 00  3.00 5000MBit/s 144mA 1IF  (YuanTech PD2500 002014A002AF)

````

### Output of ``ifconfig -a`` command

````
docker0   Link encap:Ethernet  HWaddr 02:42:62:E9:3F:DA  
          inet addr:172.17.0.1  Bcast:172.17.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

eth0      Link encap:Ethernet  HWaddr xxx  
          inet6 addr: xxx Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:2758238 errors:0 dropped:0 overruns:0 frame:0
          TX packets:24833019 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:248392022 (236.8 MiB)  TX bytes:36913114294 (34.3 GiB)

eth1      Link encap:Ethernet  HWaddr xxx
          inet6 addr: xxx
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:650985 errors:0 dropped:0 overruns:0 frame:0
          TX packets:3371684 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:67638991 (64.5 MiB)  TX bytes:4704031265 (4.3 GiB)

eth2      Link encap:Ethernet  HWaddr 2C:16:DB:A1:20:65  
          inet addr:192.168.65.56  Bcast:192.168.65.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:1942 errors:0 dropped:0 overruns:0 frame:0
          TX packets:796 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:559976 (546.8 KiB)  TX bytes:139937 (136.6 KiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:2278111 errors:0 dropped:0 overruns:0 frame:0
          TX packets:2278111 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:280598364 (267.5 MiB)  TX bytes:280598364 (267.5 MiB)

ovs-syste Link encap:Ethernet  HWaddr EA:5C:E3:50:88:B6  
          BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

ovs_bond0 Link encap:Ethernet  HWaddr 00:11:32:69:70:4B  
          inet addr:192.168.65.54  Bcast:192.168.65.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:25022 errors:0 dropped:0 overruns:0 frame:0
          TX packets:22488 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:4222477 (4.0 MiB)  TX bytes:28416228 (27.0 MiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

tun0      Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  
          inet addr:10.233.0.1  P-t-P:10.233.0.2  Mask:255.255.255.255
          UP POINTOPOINT RUNNING NOARP MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:6 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:100 
          RX bytes:0 (0.0 B)  TX bytes:4352 (4.2 KiB)
````
````
ethtool eth2
Settings for eth2:
	Supported ports: [ MII ]
	Supported link modes:   10baseT/Half 10baseT/Full 
	                        100baseT/Half 100baseT/Full 
	                        1000baseT/Full 
	                        2500baseX/Full 
	Supported pause frame use: No
	Supports auto-negotiation: Yes
	Advertised link modes:  10baseT/Half 10baseT/Full 
	                        100baseT/Half 100baseT/Full 
	                        1000baseT/Full 
	                        2500baseX/Full 
	Advertised pause frame use: Symmetric Receive-only
	Advertised auto-negotiation: Yes
	Link partner advertised link modes:  100baseT/Full 
	                                     1000baseT/Full 
	Link partner advertised pause frame use: No
	Link partner advertised auto-negotiation: Yes
	Speed: 2500Mb/s
	Duplex: Full
	Port: MII
	PHYAD: 32
	Transceiver: internal
	Auto-negotiation: on
	Supports Wake-on: pumbg
	Wake-on: g
	Current message level: 0x00007fff (32767)
			       drv probe link timer ifdown ifup rx_err tx_err tx_queued intr tx_done rx_status pktdata hw wol
````

bb-qq, 2020-05-29:
- Are you using front ports? If not, please try it.
- Please try to use another cable if you have any.

liuquan321, 2020-05-30:
> * Are you using front ports? If not, please try it.
> * Please try to use another cable if you have any.
@bb-qq @transformcube Thank you very much, when I checked it to the front port, it worked. I tested all USB ports last week, and I can only guess that it may be related to the dust in the USB ports. Now I can say that DS916 + and [UGREEN-CM275](https://www.lulian.cn/product/673-cn.html) are confirmed to work.
````
lsusb
|__usb1          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (xhci_hcd 0000:00:14.0) hub
  |__1-5         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 65006B08B0E41663)
|__usb2          1d6b:0003:0310 09  3.00 5000MBit/s 0mA 1IF  (xhci_hcd 0000:00:14.0) hub
  |__2-2         0bda:8156:3000 00  3.20 5000MBit/s 512mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 000000001)
|__usb3          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (Linux 3.10.105 etxhci_hcd-170202 Etron xHCI Host Controller 0000:04:00.0) hub
  |__3-1         051d:0002:0106 00  1.10  1.5MBit/s 0mA 1IF  (APC Back-UPS 650 FW:822.A3.I USB FW:A3 3B1637X04127  )
|__usb4          1d6b:0003:0310 09  3.00 5000MBit/s 0mA 1IF  (Linux 3.10.105 etxhci_hcd-170202 Etron xHCI Host Controller 0000:04:00.0) hub
  |__4-2         174c:55aa:0100 00  3.00 5000MBit/s 144mA 1IF  (YuanTech PD2500 002014A002AF)
````
````
64 bytes from 192.168.65.56: icmp_seq=500 ttl=63 time=1.85 ms
64 bytes from 192.168.65.56: icmp_seq=501 ttl=63 time=1.20 ms
64 bytes from 192.168.65.56: icmp_seq=502 ttl=63 time=1.31 ms
64 bytes from 192.168.65.56: icmp_seq=503 ttl=63 time=1.13 ms
64 bytes from 192.168.65.56: icmp_seq=504 ttl=63 time=1.47 ms
64 bytes from 192.168.65.56: icmp_seq=505 ttl=63 time=1.07 ms
64 bytes from 192.168.65.56: icmp_seq=506 ttl=63 time=1.18 ms
64 bytes from 192.168.65.56: icmp_seq=507 ttl=63 time=1.28 ms
64 bytes from 192.168.65.56: icmp_seq=508 ttl=63 time=2.32 ms
64 bytes from 192.168.65.56: icmp_seq=509 ttl=63 time=1.06 ms
64 bytes from 192.168.65.56: icmp_seq=510 ttl=63 time=1.03 ms
64 bytes from 192.168.65.56: icmp_seq=511 ttl=63 time=1.83 ms
64 bytes from 192.168.65.56: icmp_seq=512 ttl=63 time=1.25 ms
64 bytes from 192.168.65.56: icmp_seq=513 ttl=63 time=1.54 ms
^C
--- 192.168.65.56 ping statistics ---
513 packets transmitted, 513 received, 0% packet loss, time 512824ms
rtt min/avg/max/mdev = 1.034/1.621/19.169/0.902 ms
````

closed 2020-06-19 by liuquan321