bb-qq/r8152 #27: OK vs 8153 but not vs 8152B

phile74, 2020-06-13:
Hi,
First thanks a lot for this nice driver.
I tested 2 adapters on a Synology DS415+., USB 3.0 output.
1. Syncwire 1Gb adapter, model No. SW-AD061, Chipset RTL8153 => success !
2. CableMatters, USB 2.0 FastEthernet adapter, model No.202023, chipset RTL8152B (not 8152 afaik) => the NAS is not seen by the router (direct RJ45 link, I rebooted the NAS, ...)
Is your package ready for RTL8152 but not for RTL8152B ? or simply it should work ?
BRgds

bb-qq, 2020-06-20:
RTL8152B should work. I guess that the cause is not listed VID/PID of the device in the Linux driver.
I can add the VID/PID to the driver if you paste a result of `lsusb`.

phile74, 2020-06-23:
Hi,
thanks for your reply.
Could you please give me the procedure to paste a "result of lsusb" ? because I absolutly don't know what you mean, I'm a bit a noob I admit...
It would be very kind to add it to the driver to make it work
Rgds

phile74, 2020-06-24:
Hi bb-qq, 
I think I manage to get what you need, the result of lsusb.
This is the result based on a FastEthernet adapter, bsed on RTL8152 (chinese stuff @ 3Eur). I already sent back to A'zon the CableMatters based on RTL8152B.
Anyway, same issue with both FastEthernet-adapters : they do not show up thru control panel.

2 results : 
First with the 1Gb USB-Adapter Syncwire only, this one works very well, based on RTL8153 => 

|__usb1          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (ehci_hcd 0000:00:16.0) hub
  |__1-1         8087:07db:0002 09  2.00  480MBit/s 0mA 1IF  ( ffffffd1ffffffb2ffffffdbffffffad) hub
    |__1-1.1     f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskSataion x)
    |__1-1.3     0bda:8153:3100 00  2.10  480MBit/s 350mA 1IF  (Realtek USB 10/100/1000 LAN 001000001)
|__usb2          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (Linux 3.10.105 etxhci_hcd-170202 Etron xHCI Host Controller 0000:04:00.0) hub
|__usb3          1d6b:0003:0310 09  3.00 5000MBit/s 0mA 1IF  (Linux 3.10.105 etxhci_hcd-170202 Etron xHCI Host Controller 0000:04:00.0) hub

then I plugged a FastEth adapter based on RTL8152 I could not make it work :

|__usb1          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (ehci_hcd 0000:00:16.0) hub
  |__1-1         8087:07db:0002 09  2.00  480MBit/s 0mA 1IF  ( ffffffd1ffffffb2ffffffdbffffffad) hub
    |__1-1.1     f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskSataion x)
    |__1-1.3     0bda:8153:3100 00  2.10  480MBit/s 350mA 1IF  (Realtek USB 10/100/1000 LAN 001000001)
|__usb2          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (Linux 3.10.105 etxhci_hcd-170202 Etron xHCI Host Controller 0000:04:00.0) hub
  |__2-1         0fe6:9700:0101 00  1.10   12MBit/s 120mA 1IF  ( ffffffc6ffffffa9ffffffddffffffad)
|__usb3          1d6b:0003:0310 09  3.00 5000MBit/s 0mA 1IF  (Linux 3.10.105 etxhci_hcd-170202 Etron xHCI Host Controller 0000:04:00.0) hub

Then I rebooted the NAS, with both USB-adapters plugged
same result like above

Hope these results will enable you to fix this issue.
Just ask if you need more feedback
Rgds

bb-qq, 2020-06-27:
Thank you for sharing the result.

I found the line for the not working device
> |__2-1 0fe6:9700:0101 00 1.10 12MBit/s 120mA 1IF ( ffffffc6ffffffa9ffffffddffffffad)
and its VID/PID is 0fe6:9700.

According to the information on this page, the device appears to be powered by a chip called DM9601.
https://devicehunt.com/view/type/usb/vendor/0FE6/device/9700

So the problem with this device is a different cause than the issue of CableMatters, we need a completely different driver.

phile74, 2020-06-28:
Hi,
ok, no big deal regarding this 3eur chinese adapter, they claim it's RTL8152 but it's not.. fake news so :-)

Most important it is that your driver works well with 1Gb RTL8153 chip based like Syncwire etc : thanks a lot for your job.

I realize I did not explain why I focus on 1Gb or 100Mb adapter : it's for hifi ! :-)
You developed this driver to get data out of a Syno using multiGb USB-adapters. There is a demand for that, Very cool.
In my case, I use my NAS as a server to stream audio files to a UPnP/DLNA renderer.
The point is that the NAS (Syno & others) are just basic small computers, all integrated in 1 board. For hifi purpose, we can power the NAS by a linear power supply (LPS) to avoid the noise of the SMPS to pollute the board. As a result of that, the ethernet stream out of the NAS is far less noisy, sound quality is better.
In my case, I don't have LPS on my NAS.
When I saw your driver, I wanted to test to get data out of my NAS from the USB instead of the RJ45 socket. And the result is : complete success ! Far less noisy from USB than RJ45 => perfect for audio.
Then we can go even further easily thru USB, using a cheap adapter, to cut the 5VDC from the USB socket & provide cleaner 5VDC to the USB-Ethernet adapter using an external power supply (a power bank for smartphone works perfectly).

Many thanks for your work ! Thanks to you, music sounds far better ! Great !
BRgds

bb-qq, 2020-07-04:
I have not ever noticed such purpose of this driver, and I am glad to hear you feel it is useful :-)

rizwanranjha, 2023-08-25:
Hi, I am also able to find **Davicom DM9601 USB 2.0 To Fast Ethernet Adapter** information for **USB\VID_0FE6&PID_9700&REV_0101**  on https://www.devicekb.com/en/hardware/usb-vendors/vid_0fe6-pid_9700-rev_0101 but where to find its driver???

bb-qq, 2023-08-26:
Are you looking for a driver for Synology NAS? 
I have not developed that driver and it probably does not exist in this world.