bb-qq/r8152 #28: New r8152.c from realtek

Barlos, 2020-06-19:
Hello,

I noticed that there is a new r8152.c driver form Realtek released 2020-Apr-20. File size is 556KB.
ver. r8152-2.13.0

The current r8152.c file size is 367KB. r8152-2.12.0

[r8152-2.13.0.zip](https://github.com/bb-qq/r8152/files/4802026/r8152-2.13.0.zip)

bb-qq, 2020-06-20:
Thank you for the information. I will update the driver later.

There seem to be many changes in this release...
https://github.com/bb-qq/r8152/commit/eef1e4e435cc5e78a4cee12c3ca0c1638536a4c7

bb-qq, 2020-07-04:
I uploaded new drivers based on v2.13.0.
https://github.com/bb-qq/r8152/releases/tag/2.13.0-1
 
NOTE: Older platforms are not supported due to the kernel version.

closed 2020-07-04 by bb-qq