bb-qq/r8152 #29: RTL8156 VID: 0bda PID:8156

Romanazh, 2020-06-27:
Hi!
First of all, I want to say Thank you for the work you are doing. 

Please add a USB network card RTL8156 VID: 0bda PID:8156 
https://aliexpress.ru/item/4000982394521.html?spm=a2g0s.9042311.0.0.180e33edhvrTPm&_ga=2.95220319.257406693.1593255459-264153218.1577284876

bb-qq, 2020-07-02:
0bda/8156 is already listed in the driver. 

https://github.com/bb-qq/r8152/blob/master/r8152.c#L12956
```
	{REALTEK_USB_DEVICE(VENDOR_ID_REALTEK, 0x8050)},
	{REALTEK_USB_DEVICE(VENDOR_ID_REALTEK, 0x8152)},
	{REALTEK_USB_DEVICE(VENDOR_ID_REALTEK, 0x8153)},
	{REALTEK_USB_DEVICE(VENDOR_ID_REALTEK, 0x8156)},
```

The current driver did not work for your device?

Romanazh, 2020-07-02:
No

My nas 918+

bb-qq, 2020-07-02:
That combination should work. Could you paste outputs of `dmesg` and `lsusb` here?

Romanazh, 2020-07-03:
> 
> 
>That combination should work. Could you paste outputs of dmesg and lsusb here?
 
 
--
Роман Ажищев
 

admin@Cloud:/$ dmesg
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 4.4.59+ (root@build3) (gcc version 4.9.3 20150311 (prerelease) (crosstool-NG 1.20.0) ) #24922 SMP PREEMPT Tue Jan 14 15:19:49 CST 2020
[    0.000000] Command line: syno_hdd_powerup_seq=1 HddHotplug=0 syno_hw_version=DS918+ vender_format_version=2 console=ttyS0,115200n8 withefi elevator=elevator quiet syno_hdd_detect=0 syno_port_thaw=1 root=/dev/md0 sn=1780PDN781212 mac1=0011327B27B5 mac2=0011327B27B6 netif_num=2 vid=0x05e3 pid=0x0727
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: Supporting XSAVE feature 0x01: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x02: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x04: 'AVX registers'
[    0.000000] x86/fpu: Enabled xstate features 0x7, context size is 832 bytes, using 'standard' format.
[    0.000000] e820: BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000100-0x000000000009d7ff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009d800-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000e0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x00000000c486afff] usable
[    0.000000] BIOS-e820: [mem 0x00000000c486b000-0x00000000c4871fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000c4872000-0x00000000c4cc5fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000c4cc6000-0x00000000c5103fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000c5104000-0x00000000d8676fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000d8677000-0x00000000d8700fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000d8701000-0x00000000d8717fff] ACPI data
[    0.000000] BIOS-e820: [mem 0x00000000d8718000-0x00000000d8c56fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000d8c57000-0x00000000d9ffefff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000d9fff000-0x00000000d9ffffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000db000000-0x00000000df1fffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000f8000000-0x00000000fbffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed00000-0x00000000fed03fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed1c000-0x00000000fed1ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000011fdfffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] SMBIOS 2.7 present.
[    0.000000] DMI: ASUS All Series/H87I-PLUS, BIOS 2003 11/05/2014
[    0.000000] Kernel/User page tables isolation: disabled
[    0.000000] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000000] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000000] e820: last_pfn = 0x11fe00 max_arch_pfn = 0x400000000
[    0.000000] MTRR default type: uncachable
[    0.000000] MTRR fixed ranges enabled:
[    0.000000]   00000-9FFFF write-back
[    0.000000]   A0000-BFFFF uncachable
[    0.000000]   C0000-CFFFF write-protect
[    0.000000]   D0000-E7FFF uncachable
[    0.000000]   E8000-FFFFF write-protect
[    0.000000] MTRR variable ranges enabled:
[    0.000000]   0 base 0000000000 mask 7F00000000 write-back
[    0.000000]   1 base 0100000000 mask 7FE0000000 write-back
[    0.000000]   2 base 00E0000000 mask 7FE0000000 uncachable
[    0.000000]   3 base 00DC000000 mask 7FFC000000 uncachable
[    0.000000]   4 base 00DB000000 mask 7FFF000000 uncachable
[    0.000000]   5 base 011FE00000 mask 7FFFE00000 uncachable
[    0.000000]   6 disabled
[    0.000000]   7 disabled
[    0.000000]   8 disabled
[    0.000000]   9 disabled
[    0.000000] x86/PAT: Configuration [0-7]: WB  WT  UC- UC  WB  WT  UC- UC
[... 743 lines cut ...]
[   24.800018] iSCSI:target_core_file.c:153:fd_attach_hba ODX Token Manager is enabled.
[   24.800027] iSCSI:target_core_multi_file.c:91:fd_attach_hba RODSP plugin for multifile is enabled.
[   24.800033] iSCSI:target_core_ep.c:813:ep_attach_hba RODSP plugin for epio is enabled.
[   24.800034] iSCSI:target_core_ep.c:820:ep_attach_hba ODX Token Manager is enabled.
[   25.019042] capability: warning: `nginx' uses 32-bit capabilities (legacy support in use)
[   25.261329] loop: module loaded
[   29.464199] init: synoscheduler-vmtouch main process (5191) killed by TERM signal
[   29.796405] init: synoscheduler-vmtouch main process (13349) killed by TERM signal
[   30.108014] init: synoscheduler-vmtouch main process (13511) killed by TERM signal
[   30.263616] init: synoscheduler-vmtouch main process (13591) killed by TERM signal
admin@Cloud:/$

login as: admin
admin@192.168.1.34's password:
Could not chdir to home directory /var/services/homes/admin: No such file or dir                                                                    ectory
admin@Cloud:/$ lsusb
[... cut ...]
  |__3-1         8087:8008:0005 09  2.00  480MBit/s 0mA 1IF  ( ffffffd1ffffffb2f                                                                    fffffdbffffffad) hub
|__usb4          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (ehci_hcd 0000:00:1                                                                    d.0) hub
  |__4-1         8087:8000:0005 09  2.00  480MBit/s 0mA 1IF  ( ffffffd1ffffffb2f                                                                    fffffdbffffffa0) hub
admin@Cloud:/$ ^C
admin@Cloud:/$

Romanazh, 2020-07-03:
admin@Cloud:/$ lsusb
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.59+ xhci                                                              -hcd xHCI Host Controller 0000:00:14.0) hub
  |__1-4         f400:f400:0207 00  2.00  480MBit/s 500mA 1IF  (Generic USB Stor                                                              age 000000000207)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.59+ xhci                                                              -hcd xHCI Host Controller 0000:00:14.0) hub
  |__2-3         0bda:8156:3100 00  3.20 5000MBit/s 256mA 2IFs (Realtek USB 10/1                                                              00/1G/2.5G LAN 001000001)
|__usb3          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (ehci_hcd 0000:00:1                                                              a.0) hub
  |__3-1         8087:8008:0005 09  2.00  480MBit/s 0mA 1IF  ( ffffffd1ffffffb2f                                                              fffffdbffffffad) hub
|__usb4          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (ehci_hcd 0000:00:1                                                              d.0) hub
  |__4-1         8087:8000:0005 09  2.00  480MBit/s 0mA 1IF  ( ffffffd1ffffffb2f                                                              fffffdbffffffa0) hub
admin@Cloud:/$ dmesg
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 4.4.59+ (root@build3) (gcc version 4.9.3 20150311 (                                                              prerelease) (crosstool-NG 1.20.0) ) #24922 SMP PREEMPT Tue Jan 14 15:19:49 CST 2                                                              020
[    0.000000] Command line: syno_hdd_powerup_seq=1 HddHotplug=0 syno_hw_version                                                              =DS918+ vender_format_version=2 console=ttyS0,115200n8 withefi elevator=elevator                                                               quiet syno_hdd_detect=0 syno_port_thaw=1 root=/dev/md0 sn=1780PDN781212 mac1=00                                                              11327B27B5 mac2=0011327B27B6 netif_num=2 vid=0x05e3 pid=0x0727
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: Supporting XSAVE feature 0x01: 'x87 floating point regis                                                              ters'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x02: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x04: 'AVX registers'
[    0.000000] x86/fpu: Enabled xstate features 0x7, context size is 832 bytes,                                                               using 'standard' format.
[    0.000000] e820: BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000100-0x000000000009d7ff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009d800-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000e0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x00000000c486afff] usable
[    0.000000] BIOS-e820: [mem 0x00000000c486b000-0x00000000c4871fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000c4872000-0x00000000c4cc5fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000c4cc6000-0x00000000c5103fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000c5104000-0x00000000d8676fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000d8677000-0x00000000d8700fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000d8701000-0x00000000d8717fff] ACPI data
[    0.000000] BIOS-e820: [mem 0x00000000d8718000-0x00000000d8c56fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000d8c57000-0x00000000d9ffefff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000d9fff000-0x00000000d9ffffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000db000000-0x00000000df1fffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000f8000000-0x00000000fbffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed00000-0x00000000fed03fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed1c000-0x00000000fed1ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000011fdfffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] SMBIOS 2.7 present.
[    0.000000] DMI: ASUS All Series/H87I-PLUS, BIOS 2003 11/05/2014
[    0.000000] Kernel/User page tables isolation: disabled
[    0.000000] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000000] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000000] e820: last_pfn = 0x11fe00 max_arch_pfn = 0x400000000
[    0.000000] MTRR default type: uncachable
[    0.000000] MTRR fixed ranges enabled:
[    0.000000]   00000-9FFFF write-back
[    0.000000]   A0000-BFFFF uncachable
[    0.000000]   C0000-CFFFF write-protect
[    0.000000]   D0000-E7FFF uncachable
[    0.000000]   E8000-FFFFF write-protect
[    0.000000] MTRR variable ranges enabled:
[    0.000000]   0 base 0000000000 mask 7F00000000 write-back
[    0.000000]   1 base 0100000000 mask 7FE0000000 write-back
[... cut ...]
[ 1344.241102] Bluetooth: HCI socket layer initialized
[ 1344.241104] Bluetooth: L2CAP socket layer initialized
[ 1344.241109] Bluetooth: SCO socket layer initialized
[ 1344.242844] Bluetooth: RFCOMM TTY layer initialized
[ 1344.242848] Bluetooth: RFCOMM socket layer initialized
[ 1344.242852] Bluetooth: RFCOMM ver 1.11
[ 1344.244180] Bluetooth: HIDP (Human Interface Emulation) ver 1.2
[ 1344.244183] Bluetooth: HIDP socket layer initialized
[ 1344.249481] usbcore: registered new interface driver btusb
[ 1547.081389] usb 1-10: U

bb-qq, 2020-07-04:
```
[ 1344.068484] r8152 1-10:2.0: Unknown version 0x7410
```
This line indicates this device has a newer version of the r8156 chip. Try the new driver I created today.
https://github.com/bb-qq/r8152/releases/download/2.13.0-1/r8152-apollolake-2.13.0-1.spk

Romanazh, 2020-07-04:
it's working !
Thank you very much!

piraterx, 2022-02-13:
Hey sorry to reply but I am having same issue in pop os! Any thing can be done. it can detect but does't work. How to I make it work using same file you posted here?

bb-qq, 2022-03-06:
Later releases already support the chip revision 0x7410. So I would like you to make a new Issue and post the output of dmesg there if you need help.