bb-qq/r8152 #30: DS414 driver request

pacificwater2, 2020-06-27:
I had a Synology DS414 based on  Marvell Armada XP MV78230,  can you help to update a driver for this type CPU?

Many thanks:)

bb-qq, 2020-07-04:
I added the driver for armadaxp platforms.
https://github.com/bb-qq/r8152/releases/download/2.12.0-3/r8152-armadaxp-2.12.0-3.spk

I appreciate if you report whether it works.

Quotic, 2021-05-28:
@pacificwater2 were you able to test it?

dreimer1986, 2021-06-04:
As we wait for a reply I decided to just try it out. I just bought 2x CLUB 3D CAC-1420 from Amazon.de for my two DS414. These will not run at 2.5G at the beginning as I just started to upgrade my internal Network, but monday I can tell you if they work at all at least.

Quotic, 2021-06-05:
@dreimer1986 You are still on DSM 6 or? Which switch are you using for 2.5G? I assume the one from QNAP?

dreimer1986, 2021-06-05:
Yes, I had very bad experiences with BETA Synology firmwares in the past, so I stay where I am until final. Right now I still have everything Gigabit. That's what I mean with I can verify it working for now, not on 2.5G though. Switches (as I need 4 of them for my whole home LAN...) will still follow very soon. Regarding verification. Amazon was fast! I just installed version 2.14.0-3 and did a reboot. 3rd LAN shows up and can be set up easily. I use static IP for that.
[image]
Shares work, everything works. 2.5G check will follow soon.

dreimer1986, 2021-06-08:
Small update that you know whats up over here. I have 4 of these CLUB 3D CAC-1420 in my network now. Aka 2x Synology DS414, 2x Intel NUC. Still waiting for the already ordered 3x QNAP QSW-1105-5T and a Startech ST2GPEX PCI Express 2.5Gbps LAN card fot the main PC. The switches are sold out in Germany as it seems, but should be available in the next few days again. So they are paid and ordered, we just have to wait a lil longer. ^^ Keeping you updated.

Quotic, 2021-06-11:
I upgraded my NAS to a QNAP ts-653d in meantime (good deal in germany, and this nas has 2.5 by default). But maybe I will upgrade the synology with this 3d club also, for a backup solution.

dreimer1986, 2021-06-12:
I was curious and just updated both NAS to 2.15.0-1. Works too. With Jumbo Frames. I still don't get why 4070 fits perfect into the packet size I set on Windows driver with 4088, but.... one more or less confirms that this is the size I need. Both WIndows devices are set to it and can go up to 4046 (tested with ping 192.168.181.50 -f -l 4046) and with that setting the same counts for ping 192.168.181.168 -f -l 4046. One less and we are down to 4045 as last one working.
[image]
[image]

bb-qq, 2024-06-24:
The size of header of a ICMP packet is 28 bytes. (IP:20 + ICMP:8)

By the way, matching MTUs within LAN is not that important. This is because UDP and ICMP rarely handle packets that large, and in the case of TCP, packets are communicated in packets of the appropriate size through MSS negotiation.

The driver itself is working, so I would like to close this issue. If you still have questions, you can create a new thread in the forum to gather knowledge.

closed 2024-06-24 by bb-qq