bb-qq/r8152 #31: DS920+

pmousarris, 2020-06-28:
Will the driver work for the new nas DS920+ ?

davidgirard, 2020-06-28:
Hello there - I have the same question for the DS220+. Both 920 and 220 sport Gemini Lake cores so I don’t think support is available yet, but it would be awesome if it could be added to a later release. I’ll stick to my DS216+II for now, the driver works great with it (from USB port).

Cheers,
David

bb-qq, 2020-07-02:
Currently, building packages for Geminilake architecture is not supported, because toolchains for them are not released by Synology yet.
https://sourceforge.net/projects/dsgpl/files/DSM%206.2.2%20Tool%20Chains/

davidgirard, 2020-07-02:
Gotcha. Thanks for the hard work!

randompasser, 2020-08-04:
Geminilake toolchain is out: https://sourceforge.net/projects/dsgpl/files/DSM%206.2.2%20Tool%20Chains/Intel%20x86%20Linux%204.4.59%20%28GeminiLake%29/

davidgirard, 2020-08-04:
Great news!

bb-qq, 2020-08-14:
Toolchains for geminilake is certainly released, but the build environment including kernel sources is not yet...

```
pkgscripts-ng/EnvDeploy -v 6.2 --list
Available platforms: 6281 alpine alpine4k apollolake armada370 armada375 armada37xx armada38x armadaxp avoton braswell broadwell broadwellnk bromolow cedarview comcerto2k denverton dockerx64 evansport grantley hi3535 kvmx64 monaco qoriq rtd1296 x64
```

PatrickPTC, 2020-08-17:
try on ds920+ can show the card, but can not transfer data even in direct link mode or bridge mode. Nas will drop link,shut down.  May I ask when it could work. Thanks.

Will0822, 2020-09-02:
DS920+ report: it fails to work normally, DS920+ will crash, must restart the device and delete the driver to work normally

silverko2, 2020-10-10:
Hello bb-qq, 
after last toolkit update now available geminilake platform, please can You try to compile the driver for DS920+
Thank You

```
/toolkit/pkgscripts/EnvDeploy -v 6.2 --list
Available platforms: ........ geminilake ...........
```

davidgirard, 2020-10-11:
I did a quick diff between the apollo and gemini lake toolchains, here you go:
15c15
< PLAT_FLAGS="-DSYNO_GEMINILAKE"
---
> PLAT_FLAGS="-DSYNO_APOLLOLAKE"
101c101
< SynoKernelConfig="geminilake"
---
> SynoKernelConfig="apollolake"

The same goes for all the other files, only the platform names were added/changed.

In other words, basically the same config. The apollo lake driver was notoriously unstable when installed on a gemini lake NAS, so I'm not holding my breath at this point.

bb-qq, 2020-10-12:
Thank you for the information!
I created [the driver for the geminilake platform](https://github.com/bb-qq/r8152/releases/download/2.13.0-1/r8152-geminilake-2.13.0-1.spk) with following modification of pkgscript-ng.

```
diff --git a/include/pkg_util.sh b/include/pkg_util.sh
index 2ae8852..8208ad1 100755
--- a/include/pkg_util.sh
+++ b/include/pkg_util.sh
@@ -63,6 +63,7 @@ pkg_get_platform() { # [path of env.mak (default: /env.mak)]
                        DENVERTON)              arch="denverton" ;;
                        REALTEK_RTD1296)        arch="rtd1296" ;;
                        APOLLOLAKE)             arch="apollolake" ;;
+                       GEMINILAKE)             arch="geminilake" ;;
                        CEDARVIEW)              arch="cedarview" ;;
                        AVOTON)                 arch="avoton" ;;
                        BRASWELL)               arch="braswell" ;;
@@ -92,8 +93,8 @@ plat_to_unified_plat() {
        local unified_plat=
 
        case "$plat" in
-               x86 | bromolow | cedarview | avoton | braswell | broadwell | dockerx64 | kvmx64 | grantley | denverton | apollolake | broadwellnk)
-                       unified_plat="x86 bromolow cedarview avoton braswell broadwell dockerx64 kvmx64 grantley denverton apollolake broadwellnk"
+               x86 | bromolow | cedarview | avoton | braswell | broadwell | dockerx64 | kvmx64 | grantley | denverton | apollolake | broadwellnk | geminilake)
+                       unified_plat="x86 bromolow cedarview avoton braswell broadwell dockerx64 kvmx64 grantley denverton apollolake broadwellnk geminilake"
                        ;;
                # alpine and alpine4k use same define.
                alpine | alpine4k )
@@ -111,7 +112,7 @@ plat_to_family() {
        local family=
 
        case "$plat" in
-               x86 | bromolow | cedarview | avoton | braswell | broadwell | dockerx64 | kvmx64 | grantley | denverton | apollolake | broadwellnk)
+               x86 | bromolow | cedarview | avoton | braswell | broadwell | dockerx64 | kvmx64 | grantley | denverton | apollolake | broadwellnk | geminilake)
                        family="x86_64"
                        ;;
                evansport )
@@ -216,7 +217,7 @@ pkg_get_spk_unified_platform() { # [path of env.mak (default: /env.mak)]
                88f6281)
                        spk_unified_platform="88f628x"
                        ;;
-               x86 | bromolow | cedarview | avoton | braswell | broadwell | dockerx64 | kvmx64 | grantley | denverton | apollolake | broadwellnk)
+               x86 | bromolow | cedarview | avoton | braswell | broadwell | dockerx64 | kvmx64 | grantley | denverton | apollolake | broadwellnk | geminilake)
                        spk_unified_platform="x64"
                        ;;
                alpine | alpine4k )
diff --git a/include/platforms b/include/platforms
index b0372eb..bce4c89 100644
--- a/include/platforms
+++ b/include/platforms
@@ -47,7 +47,7 @@ AllPlatformOptions () {
 }
 
 Is64BitPlatform() {
-       local all64BitPlatforms="X64 BROMOLOW CEDARVIEW AVOTON BRASWELL BROADWELL DOCKERX64 KVMX64 GRANTLEY DENVERTON REALTEK_RTD1296 APOLLOLAKE BROADWELLNK"
+       local all64BitPlatforms="X64 BROMOLOW CEDARVIEW AVOTON BRASWELL BROADWELL DOCKERX64 KVMX64 GRANTLEY DENVERTON REALTEK_RTD1296 APOLLOLAKE BROADWELLNK GEMINILAKE"
        CheckInList $BUILD_TARGET $all64BitPlatforms && return 0 || return 1
 }
 ```

punkw800c, 2020-10-13:
Downloaded the driver for geminilake platform。It worked on ds220+，but not very stable
[image]

kleuter, 2020-10-13:
What exactly you mean by the word "stable"?

punkw800c, 2020-10-13:
Sorry，I’m not a native speaker of English。
The network to my NAS was down while I am downloading a movie.

kleuter, 2020-10-13:
Network can be down to reasons not related to the driver. Does it reproduce of did it happen only once?

punkw800c, 2020-10-13:
It happened just once. But it was only used for less than 2 hours.

kleuter, 2020-10-13:
You need to make sure it happens more than once and only when using the driver (not default 1GBe ethernet).

punkw800c, 2020-10-13:
The driver was used only. The default 1GBe ethernet was not connected with my router.

punkw800c, 2020-10-13:
And I will keep testing this driver.

bb-qq, 2021-01-04:
Thank you for test the driver.

I would like to close this issue. And I would appreciate it if create a new issue using the template for bug reports if you have any problems. Thanks.

closed 2021-01-04 by bb-qq