bb-qq/r8152 #32: 2x 918+ with two CLUB 3D CAC-1420 - Success Report - 2.13.0-1

megabyte0469, 2020-07-04:
Hallo,
the new driver version 2.13.0-1 works perfect with 2 DS918+ (2 CLUB 3D CAC-1420 adapters on everyone DS918+)

Thanks for all.

Great Job !!

Barlos, 2020-07-05:
Great.
I wish my DS716+II Braswell CPU can be supported ASAP.

davidgirard, 2020-07-05:
Hey there - My Braswell-based DS216+II works well with the Braswell driver. The DS716+II sports a very similar CPU, main difference is # of cores, there’s no reason it shouldn’t work. Have you tried all USB ports? Did you grab the right driver?

Cheers,
David

stefandhn, 2022-11-27:
Working on latest DSM using the back USB port on DS918+. Setup:

- 2 CLUB 3D's, 1 in the PC and 1 in the NAS, connected directly to eachother
- Reaching roughly a constant ~230 MB/s write speed to the NAS (RAID10) with large files
- Stability of copying files to the NAS improved when using jumbo frames (less dips, more constant ~230 MB/s)