bb-qq/r8152 #36: D218+ UGREEN USB A to 2.5G(RTL8156)  issue

ryanchen789, 2020-07-09:
Synology D218+
UGREEN USB A to 2.5G Ethernet Adapter (RTL8156)
Not working properly

Unable to work with Asus AX86U **2.5G LAN port**，
Unable to assign IP address automatically。
---------------------------------------------------------
Work properly

work with Asus AX86U **1G LAN port**
Get IP address normally

ryanchen789, 2020-07-09:
Test equipment: DS218 +, Ugreen USB a to 2.5G (RTL8156), ASUS AX86U

When the network cable is inserted into the 2.5G LAN port of the router, the 2.5G network can be recognized, but it can not work normally.
Unable to Ping, unable to access NAS

When the network cable is inserted into the 1g LAN port of the router, the 1g network can be identified and the NAS can be accessed normally.

liuquan321, 2020-07-11:
I have report this problem to AUSU support , thay told me that it would be fixed soon, please be patient.

XsplitS, 2020-07-12:
Try to configure manually your interface (eth1)

ex: ifconfig eth1 192.x.x.x netmask 255.x.x.x up

blackie333, 2020-07-22:
Hello guys, thank you a lot for this project.
I have DS218 and the same ugreen adaptor on the way from aliexpress :(
Have you managed to fix the issue ?

bb-qq, 2020-07-23:
Please try connecting the dongle to your Windows PC to isolate the problem.
I guess this is an auto-negotiation issue of the router and it can be solved by fixing the link-speed manually.

seasonsolt, 2020-08-02:
> Test equipment: DS218 +(DSM 6.2.3 25426), Ugreen USB A to 2.5G (RTL8156), ASUS AX86U
> 
> When the network cable is inserted into the 2.5G LAN port of the router, the 2.5G network can be recognized, but it can not work normally.
> Unable to Ping, unable to access NAS
> 
> When the network cable is inserted into the 1g LAN port of the router, the 1g network can be identified and the NAS can be accessed normally.

ASUS update it's firmware  v.3.0.0.4.384.9283 , it might helps

ryanchen789, 2020-08-02:
FIRMWARE v.3.0.0.4.384.9283, WORK WELL,THINK'S VERY MUCH

closed 2020-08-02 by ryanchen789