bb-qq/r8152 #42: Test with DS720+ (Gemini Lake) successful

Cretanios, 2020-08-11:
### Description of the problem

I simply renamed the line
    arch= "apollolake"
         with
    arch="geminilake"
in the r8152-apollolake-2.13.0-1.spk file with NotePad++ and saved the file.

After Installation on the DS720+ the driver worked well and I have proper LAN connection
(see Daniels Video: https://www.youtube.com/watch?v=AyU7bB7BOsc) 

### Description of your products

- DS720+ with 10GB RAM
- DSM version 6.2.3-25426 Update 2
- 3D CAC-1420 Dongle (USB A)

marco712, 2020-08-11:
It work on the 920+ thans

PatrickPTC, 2020-08-17:
PatrickPTC commented 9 minutes ago

try on ds920+ can show the card, but can not transfer data or login in admin page. even in direct link mode or net bridge mode. Nas will drop link, shut down. May I ask when it could work. Thanks.

Anyone who can make it work on 920,is there any special setup or ststus ?  I try lots of time and nas must be hard power off when connect fall, too shame. Hope could use it soon.

aoleg, 2020-08-20:
DS420+ report: works, not stable after a very short while, the device must be restarted. Removed the driver.

Will0822, 2020-09-02:
DS920+ report: it fails to work normally, DS920+ will crash, must restart the device and delete the driver to work normally

aminebizid, 2020-12-15:
I got this message with dmesg
r8152: module verification failed: signature and/or required key missing - tainting kernel

Sjekke, 2020-12-15:
> I got this message with dmesg
> 
> r8152: module verification failed: signature and/or required key missing - tainting kernel

Did you use the latest driver ...

bb-qq, 2021-01-04:
I have already created [the driver for geminilake built with the official toolchains](https://github.com/bb-qq/r8152/releases/download/2.14.0-2/r8152-geminilake-2.14.0-2.spk). So I would appreciate it if you try it first and create a new issue using the template for bug report if you have any problems. Thanks.

closed 2021-01-04 by bb-qq