bb-qq/r8152 #43: [Success] DS1815+ (avaton) confirmed working with Club 3D CAC-1420

dankao, 2020-08-21:
**NAS:** DS1815+ (avaton) with 16GB RAM (2x8GB) w/two DX513's (Fully Loaded 18 drives)**.
DSM Version:** 6.2.3-25426 Update 2
**Driver version:** r8152-avoton-2.12.0-3.spk
**Switch:** Netgear MS510TXPP (port g4, 2.5Gbps)

**Notes:** Informal SMB transfer speed to a Windows 10 x64 PC with NVMe SSD connected to g9 (10Gbps) QNAP QXG-10G1T (Aquantia AQtion AQC107 Chipset) results in 220MB+/sec speeds.

Since I'm running Open vSwitch for Virtual Machines on the DS1815+, I had to SSH into the box and manually run:

`ifconfig eth4 up`

Before if would show up in the Control Panel, allowing me to configure.

```
eth4      Link encap:Ethernet  HWaddr A0:CE:C8:xx:xx:xx
          inet addr:xxx.xxx.xxx.xxx  Bcast:xxx.xxx.xxx.xxx  Mask:xxx.xxx.xxx.xxx
          UP BROADCAST RUNNING MULTICAST  MTU:9000  Metric:1
          RX packets:833215 errors:0 dropped:0 overruns:0 frame:0
          TX packets:892799 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:3725217175 (3.4 GiB)  TX bytes:6681140918 (6.2 GiB)
```

Relevant dmesg:

```
[Thu Aug 20 23:32:11 2020] r8152: module verification failed: signature and/or required key missing - tainting kernel
[Thu Aug 20 23:32:11 2020] usbcore: registered new interface driver r8152
[Thu Aug 20 23:34:47 2020] usb 3-1: new SuperSpeed USB device number 2 using etxhci_hcd-170202
[Thu Aug 20 23:34:47 2020] etxhci_hcd-170202 0000:04:00.0: WARN Event TRB for slot 2 ep 2 with no TDs queued?
[Thu Aug 20 23:34:47 2020] etxhci_hcd-170202 0000:04:00.0: WARN Event TRB for slot 2 ep 3 with no TDs queued?
[Thu Aug 20 23:34:47 2020] etxhci_hcd-170202 0000:04:00.0: WARN Event TRB for slot 2 ep 6 with no TDs queued?
[Thu Aug 20 23:34:48 2020] usb 3-1: reset SuperSpeed USB device number 2 using etxhci_hcd-170202
[Thu Aug 20 23:34:48 2020] r8152 3-1:1.0 eth4: v2.12.0 (2019/04/29)
[Thu Aug 20 23:34:48 2020] r8152 3-1:1.0 eth4: This product is covered by one or more of the following patents:
                                        US6,570,884, US6,115,776, and US6,327,625.

[Thu Aug 20 23:34:50 2020] ip_tables: (C) 2000-2006 Netfilter Core Team
...
[Thu Aug 20 23:37:58 2020] r8152 3-1:1.0 eth4: carrier on
[Thu Aug 20 23:38:17 2020] usb 2-4: ep 0x81 - rounding interval to 64 microframes, ep desc says 80 microframes
```

dankao, 2020-08-31:
**Update:** I had some weird wonkiness if IPv6 was left on including carrier flapping and worst-case scenario, what seems to be a kernel panic and the system becoming non-responsive requiring a hard reboot. After disabling IPv6 on the dongle, the system remained stable.

If an IPv6 address still shows up on the dongle even if you have it set to off in DSM under Network, run as root/sudo in shell:

`echo 1 > /proc/sys/net/ipv6/conf/eth4/disable_ipv6`

Going on 24 hours now with heavy traffic and no issues now.

dankao, 2020-09-03:
**Update #2:** Synology finally crashed.

```
2020-09-02T17:46:59-07:00 bertha kernel: [53137.143460] ------------[ cut here ]------------
2020-09-02T17:46:59-07:00 bertha kernel: [53137.148633] WARNING: at net/sched/sch_generic.c:255 dev_watchdog+0x24b/0x260()
2020-09-02T17:46:59-07:00 bertha kernel: [53137.163910] Modules linked in: r8152(OF) tcm_loop(O) iscsi_target_mod(O) target_core_ep(O) target_core_multi_file(O) target_core_file(O) t
arget_core_iblock(O) target_core_mod(O) syno_extent_pool(PO) rodsp_ep(O) cifs udf isofs loop nfsd exportfs rpcsec_gss_krb5 hid_generic usbhid hid usblp usb_storage nf_conntrack x_tab
les avoton_synobios(PO) leds_lp3943 exfat(O) btrfs synoacl_vfs(PO) zlib_deflate hfsplus md4 hmac libcrc32c mlx5_core(O) mlx4_en(O) mlx4_core(O) mlx_compat(O) compat(O) qede(O) qed(O)
 atlantic(O) tn40xx(O) i40e(O) ixgbe(O) be2net(O) igb(O) i2c_algo_bit e1000e(O) vxlan fuse vfat fat crc32c_intel aesni_intel glue_helper lrw gf128mul ablk_helper arc4 cryptd ecryptfs
 sha256_generic sha1_generic ecb aes_x86_64 authenc des_generic ansi_cprng cts md5 cbc cpufreq_powersave cpufreq_performance acpi_cpufreq mperf processor thermal_sys cpufreq_stats fr
eq_table dm_snapshot crc_itu_t crc_ccitt quota_v2 quota_tree psnap p8022 llc sit tunnel4 ip_tunnel ipv6 zram(C) sg etxhci_hcd xhci_hcd uhci_hcd ehci_pci ehci_hcd usbcore usb_common [
last unloaded: ip_tables]
2020-09-02T17:46:59-07:00 bertha kernel: [53137.269073] CPU: 1 PID: 0 Comm: swapper/1 Tainted: PF        C O 3.10.105 #25426
2020-09-02T17:46:59-07:00 bertha kernel: [53137.277344] Hardware name: Insyde MohonPeak/Type2 - Board Product Name1, BIOS M.111 2017/05/09
2020-09-02T17:46:59-07:00 bertha kernel: [53137.286975]  ffffffff814af01b ffffffff81036b08 0000000000000000 ffff88047fc43e68
2020-09-02T17:46:59-07:00 bertha kernel: [53137.295271]  0000000000000001 0000000000000001 0000000000000060 ffffffff81036b69
2020-09-02T17:46:59-07:00 bertha kernel: [53137.303571]  ffffffff817378c0 0000000000000030 ffff88047fc43e78 ffff88047fc43e38
2020-09-02T17:46:59-07:00 bertha kernel: [53137.311868] Call Trace:
2020-09-02T17:46:59-07:00 bertha kernel: [53137.314601]  <IRQ>  [<ffffffff814af01b>] ? dump_stack+0xc/0x15
2020-09-02T17:46:59-07:00 bertha kernel: [53137.321145]  [<ffffffff81036b08>] ? warn_slowpath_common+0x58/0x70
2020-09-02T17:46:59-07:00 bertha kernel: [53137.328060]  [<ffffffff81036b69>] ? warn_slowpath_fmt+0x49/0x50
2020-09-02T17:46:59-07:00 bertha kernel: [53137.334682]  [<ffffffff810696f3>] ? trigger_load_balance+0x53/0x200
2020-09-02T17:46:59-07:00 bertha kernel: [53137.341693]  [<ffffffff814096cb>] ? dev_watchdog+0x24b/0x260
2020-09-02T17:46:59-07:00 bertha kernel: [53137.348023]  [<ffffffff81409480>] ? dev_graft_qdisc+0x70/0x70
2020-09-02T17:46:59-07:00 bertha kernel: [53137.354450]  [<ffffffff81041a0c>] ? call_timer_fn.isra.32+0x1c/0x80
2020-09-02T17:46:59-07:00 bertha kernel: [53137.361460]  [<ffffffff81041bda>] ? run_timer_softirq+0x16a/0x1f0
2020-09-02T17:46:59-07:00 bertha kernel: [53137.368277]  [<ffffffff8103aca6>] ? __do_softirq+0xc6/0x1a0
2020-09-02T17:46:59-07:00 bertha kernel: [53137.374511]  [<ffffffff814b7c3c>] ? call_softirq+0x1c/0x1a0
2020-09-02T17:46:59-07:00 bertha kernel: [53137.380743]  [<ffffffff81004635>] ? do_softirq+0x35/0x70
2020-09-02T17:46:59-07:00 bertha kernel: [53137.386684]  [<ffffffff8103aeb5>] ? irq_exit+0x95/0xa0
2020-09-02T17:46:59-07:00 bertha kernel: [53137.392431]  [<ffffffff81022027>] ? smp_apic_timer_interrupt+0x67/0xa0
2020-09-02T17:46:59-07:00 bertha kernel: [53137.399732]  [<ffffffff814b6bbf>] ? apic_timer_interrupt+0x15f/0x170
2020-09-02T17:46:59-07:00 bertha kernel: [53137.406838]  <EOI>  [<ffffffff813cad59>] ? cpuidle_enter_state+0x49/0xb0
2020-09-02T17:46:59-07:00 bertha kernel: [53137.414349]  [<ffffffff813cad55>] ? cpuidle_enter_state+0x45/0xb0
2020-09-02T17:46:59-07:00 bertha kernel: [53137.421176]  [<ffffffff813cae68>] ? cpuidle_idle_call+0xa8/0x160
2020-09-02T17:46:59-07:00 bertha kernel: [53137.427897]  [<ffffffff8100b3f5>] ? arch_cpu_idle+0x5/0x20
2020-09-02T17:46:59-07:00 bertha kernel: [53137.434025]  [<ffffffff8107b50d>] ? cpu_startup_entry+0x7d/0x160
2020-09-02T17:46:59-07:00 bertha kernel: [53137.440743] ---[ end trace a20c13f1602fc3df ]---
2020-09-02T17:46:59-07:00 bertha kernel: [53137.445908] r8152 3-1:1.0 eth4: Tx timeout
2020-09-02T17:46:59-07:00 bertha kernel: [53137.455140] r8152 3-1:1.0 eth4: Tx status -2
2020-09-02T17:46:59-07:00 bertha kernel: [53137.459973] r8152 3-1:1.0 eth4: Tx status -2
2020-09-02T17:46:59-07:00 bertha kernel: [53137.464801] r8152 3-1:1.0 eth4: Tx status -2
2020-09-02T17:46:59-07:00 bertha kernel: [53137.469607] r8152 3-1:1.0 eth4: Tx status -2
2020-09-02T17:50:07-07:00 bertha kernel: [53325.752201] INFO: task snmpd:15172 blocked for more than 120 seconds.
2020-09-02T17:50:07-07:00 bertha kernel: [53325.759409] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
2020-09-02T17:50:07-07:00 bertha kernel: [53325.776107]  ffff88043aca7d50 0000000000000086 000000000000c000 ffff88043aca7fd8
2020-09-02T17:50:07-07:00 bertha kernel: [53325.784417]  ffff88043aca7fd8 ffff88046a140800 ffff88046a140800 ffff88046f002980
2020-09-02T17:50:07-07:00 bertha kernel: [53325.792710]  ffff88046a140800 ffffffff81854f40 ffffffff81854f44 00000000ffffffff
```

It seems related to this issue: https://bugzilla.kernel.org/show_bug.cgi?id=198931#c6

But I'm not sure if there's any way to disable LPM or to enable `usb.quirks` with the kernel that Synology uses, especially since I don't know how far Linux kernel 3.10.105 is backported by them.

bb-qq, 2020-09-13:
I investigated Synology's startup script and found that we could set parameters to the kernel module by editing the `/etc/modules.local.conf` in my environment(DS918+) as follows.

```
module_usbcore="quirks=0bda:8153:k"
```

But [Linux kernel 3.10.105 does not have the capability of quirk for LPM](https://github.com/torvalds/linux/blob/v3.10/drivers/usb/core/quirks.c) unfortunately...

war59312, 2021-03-16:
Thanks, same issue here with my DS1815+, was wondering why it kept crashing when transferring files.

@bb-qq So does this mean, we can't use with DS1815+?

bb-qq, 2021-03-21:
> So does this mean, we can't use with DS1815+?

I have no idea what causes it, so for now I can only say that it does not work...

> /etc/modules.local.conf does not exist for me. There is however, /etc/modules.conf

Yes, modules.local.conf should be created manually. modules.conf will be overwritten by Synology programs.

war59312, 2021-03-21:
10-4 Thank you. I appreciate you.

war59312, 2021-06-13:
Sadly still an issue with version 2.15.0-1

kekegsm, 2022-04-17:
Hi! Do you have any update on this? There is some avaton based Synology at my partners, and they like some network improvization...

bb-qq, 2022-04-23:
I made a test driver disable LPM to investigate the problem on other platforms, but the symptoms did not improve.

Since LPM is also suspected to cause crashes in this Issue, I have also made a driver for avton, but it will probably not improve the symptoms.
https://github.com/bb-qq/r8152/releases/download/2.15.0-6/r8152-avoton-2.15.0-7nolpm2.spk

kekegsm, 2022-04-24:
Thx,

I will try it later, (waiting for new HDD-s), and I will write back with
status.

THX a lot
Gabriel

bb-qq ***@***.***> ezt írta (időpont: 2022. ápr. 23., Szo,
6:51):

> I made a test driver disable LPM to investigate the problem on other
> platforms, but the symptoms did not improve.
>
> Since LPM is also suspected to cause crashes in this Issue, I have also
> made a driver for avton, but it will probably not improve the symptoms.
>
> https://github.com/bb-qq/r8152/releases/download/2.15.0-6/r8152-avoton-2.15.0-7nolpm2.spk
>
> <https://github.com/bb-qq/r8152/issues/43#issuecomment-1107361256>, or
> unsubscribe

kekegsm, 2022-05-05:
Hi,

A tried it, but no success.:
-Installed the driver
-I can see it in DSM, but didn't get IP with DHCP
-I set manually an IP, but can't ping it/cant reach it. In my Mikrotik
router, I see the IP, but no MAC address on this card.

Any idea? Maybe my Mikrotik router or my mikrotik switch is the problem?

Keserű Gábor ***@***.***> ezt írta (időpont: 2022. ápr. 24., V,
23:34):

> Thx,
>
> I will try it later, (waiting for new HDD-s), and I will write back with
> status.
>
> THX a lot
> Gabriel
>
> bb-qq ***@***.***> ezt írta (időpont: 2022. ápr. 23., Szo,
> 6:51):
>
>> I made a test driver disable LPM to investigate the problem on other
>> platforms, but the symptoms did not improve.
>>
>> Since LPM is also suspected to cause crashes in this Issue, I have also
>> made a driver for avton, but it will probably not improve the symptoms.
>>
>> https://github.com/bb-qq/r8152/releases/download/2.15.0-6/r8152-avoton-2.15.0-7nolpm2.spk
>>
>> <https://github.com/bb-qq/r8152/issues/43#issuecomment-1107361256>, or
>> unsubscribe
>>
>
>
>
> *Keserű Gábor*

war59312, 2022-05-05:
Try using 169.254.x.x IP addresses so don't need DHCP ?

bb-qq, 2022-05-28:
That could be a problem on the driver side.
If you can provide the dmesg output I can look into it.

war59312, 2022-08-23:
Great news to report.

Looks like this is for sure a case of DS1815+ USB ports not having enough power.

Now using a USB hub with power and it's stable. 🥇 

I've gotten it fully stable and seeing as high as 230MB/s transfer speeds.

Average closer to 165MB/s, but that is still a good 50MB/s faster, I'll take it.

> sudo ifconfig eth4
> eth4      Link encap:Ethernet  HWaddr A0:CE:C8:CD:44:CF
>           inet addr:169.254.100.105  Bcast:169.254.100.255  Mask:255.255.255.0
>           inet6 addr: fe80::a2ce:c8ff:fecd:44cf/64 Scope:Link
>           UP BROADCAST RUNNING MULTICAST  MTU:9000  Metric:1
>           RX packets:8675499 errors:0 dropped:0 overruns:0 frame:0
>           TX packets:3615385 errors:0 dropped:0 overruns:0 carrier:0
>           collisions:0 txqueuelen:1000
>           RX bytes:67096356692 (62.4 GiB)  TX bytes:22196930825 (20.6 GiB)

Zero errors and 0 dropped after 62.4GiB sent and 20.6GiB received. Just messing around and that over a good 15 min period.

Do see dmesg showing:

```
[1035956.474575] r8152 3-2.2:1.0 eth4: get_registers -71
[1036197.917767] r8152 3-2.2:1.0 eth4: get_registers -71
[1036210.955125] r8152 3-2.2:1.0 eth4: get_registers -71
[1036248.011926] r8152 3-2.2:1.0 eth4: get_registers -71
[1036313.169018] r8152 3-2.2:1.0 eth4: get_registers -71
[1036367.309472] r8152 3-2.2:1.0 eth4: get_registers -71
[1036712.867978] r8152 3-2.2:1.0 eth4: get_registers -71
[1036738.961701] r8152 3-2.2:1.0 eth4: get_registers -71
[1036767.096728] r8152 3-2.2:1.0 eth4: get_registers -71
[1036781.136903] r8152 3-2.2:1.0 eth4: get_registers -71
[1036928.807372] r8152 3-2.2:1.0 eth4: get_registers -71
[1036982.485838] r8152 3-2.2:1.0 eth4: get_registers -71
```

But everything seems fine.

Running lsusb shows correctly the hardware 🥇 

```
|__usb1          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (ehci_hcd 0000:00:16.0) hub
  |__1-1         8087:07db:0002 09  2.00  480MBit/s 0mA 1IF  ( ffffffd1ffffffb2ffffffdbffffffad) hub
    |__1-1.1     f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 65004AAA37852190)
|__usb2          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (Linux 3.10.105 etxhci_hcd-170202 Etron xHCI Host Controller 0000:04:00.0) hub
  |__2-2         05e3:0610:0655 09  2.10  480MBit/s 100mA 1IF  (GenesysLogic USB2.1 Hub ffffffd6ffffffa3ffffffebffffffcb) hub
  |__2-3         0764:0501:0001 00  2.00   12MBit/s 2mA 1IF  (CP1500PFCLCD CRCA102-3I1 CPS)
|__usb3          1d6b:0003:0310 09  3.00 5000MBit/s 0mA 1IF  (Linux 3.10.105 etxhci_hcd-170202 Etron xHCI Host Controller 0000:04:00.0) hub
  |__3-2         05e3:0626:0655 09  3.20 5000MBit/s 0mA 1IF  (GenesysLogic USB3.1 Hub ffffffd6ffffffa3ffffffebffffffc0) hub
    |__3-2.2     0bda:8156:3000 00  3.20 5000MBit/s 512mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 000000001)

```

I'll keep testing, for now reporting my findings and hoping others can duplicate and enjoy.

Hardware And Software:

I have cat 5e cable running from my NAS to my Win10 Computer just for testing.

1.) DS1815+ running DM 6.2.4-25556 Update 6
2.) RTL8152/RTL8153 Driver Version 2.15.0-1
3.) SABRENT 4-Port USB 3.0 Hub with Individual LED Lit Power Switches, Includes 5V/2.5A Power Adapter (HB-UMP3) from https://smile.amazon.com/gp/product/B00TPMEOYM/
4.) Club 3D CAC-1420 (https://smile.amazon.com/gp/product/B07Q626XK2/) Plugged Into The HUB with Power into USB on the NAS
5.) A 2nd Club 3D CAC-1420 (https://smile.amazon.com/gp/product/B07Q626XK2/) plugged into my Computer's USB 3.0 slot for testing, letting it use default 169.254.100.x IP and I set adapter to use MTU of 9,000, as in enable Jump Frame of 9,014 Bytes
6.) On the NAS, run `sudo ifconfig eth4 up` to bring the Club 3D Adapter online in Control Panel > Network > Network Interface. I set the Interface to use a 169.254.100.x address and using MTU of 9,000

Then start testing. It's stable for me now, even 30 minutes later. First time ever.

I'll get it working properly with DHCP and what not later, have to grab a longer cable first haha.

Anyways, hope that helps someone.

war59312, 2022-08-25:
Holding stable. 💯 

```
ifconfig eth4
eth4      Link encap:Ethernet  HWaddr A0:CE:C8:CD:44:CF
          inet addr:192.168.10.145  Bcast:192.168.10.255  Mask:255.255.255.0
          inet6 addr: fe80::a2ce:c8ff:fecd:44cf/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:30795591 errors:0 dropped:0 overruns:0 frame:0
          TX packets:14333790 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:165381600930 (154.0 GiB)  TX bytes:77290907842 (71.9 GiB)

```

So doesn't work great with Jumbo frames, it's slower. Instead just using standard 1,500 MTU.

And DHCP is working as expected.

Make sure to set Control Panel > Network > Network Interface > Manage > Service Order and make LAN 5 first if you want your 2.5Gbps network to be the default.

I recommend keeping the built in LAN 1 still as a backup, just in case.

war59312, 2022-08-31:
Also tried with Plugable 2.5G USB C and USB to Ethernet Adapter, 2-in-1 Adapter @ https://smile.amazon.com/dp/B084L4JL9K and same problem. :(

```
lsusb
|__usb1          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (ehci_hcd 0000:00:16.0) hub
  |__1-1         8087:07db:0002 09  2.00  480MBit/s 0mA 1IF  ( ffffffd1ffffffb2ffffffdbffffffad) hub
    |__1-1.1     f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 65004AAA37852190)
|__usb2          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (Linux 3.10.105 etxhci_hcd-170202 Etron xHCI Host Controller 0000:04:00.0) hub
  |__2-1         05e3:0610:0655 09  2.10  480MBit/s 100mA 1IF  (GenesysLogic USB2.1 Hub ffffffd6ffffffa3ffffffebffffffcb) hub
  |__2-3         0764:0501:0001 00  2.00   12MBit/s 2mA 1IF  (CP1500PFCLCD CRCA102-3I1 CPS)
|__usb3          1d6b:0003:0310 09  3.00 5000MBit/s 0mA 1IF  (Linux 3.10.105 etxhci_hcd-170202 Etron xHCI Host Controller 0000:04:00.0) hub
  |__3-1         05e3:0626:0655 09  3.20 5000MBit/s 0mA 1IF  (GenesysLogic USB3.1 Hub ffffffd6ffffffa3ffffffebffffffc0) hub
    |__3-1.2     0bda:8156:3100 00  3.20 5000MBit/s 256mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 001000001)

```

```
[1741685.032519] r8152 3-1.2:1.0 eth4: get_registers -71
[1741888.349569] r8152 3-1.2:1.0 eth4: get_registers -71
[1742717.690155] r8152 3-1.2:1.0 eth4: get_registers -71
[1743349.415659] r8152 3-1.2:1.0 eth4: Tx timeout
[1743349.425369] r8152 3-1.2:1.0 eth4: Tx status -2
[1743349.430573] r8152 3-1.2:1.0 eth4: Tx status -2
[1743349.435801] r8152 3-1.2:1.0 eth4: Tx status -2
[1743349.441037] r8152 3-1.2:1.0 eth4: Tx status -2
[1743351.863779] r8152 3-1.2:1.0 eth4: get_registers -71
[1743351.881935] r8152 3-1.2:1.0 eth4: get_registers -71
[1743351.900061] r8152 3-1.2:1.0 eth4: get_registers -71
[1743351.918194] r8152 3-1.2:1.0 eth4: get_registers -71
[1743351.936330] r8152 3-1.2:1.0 eth4: get_registers -71
[1743351.954473] r8152 3-1.2:1.0 eth4: get_registers -71
[1743351.972612] r8152 3-1.2:1.0 eth4: get_registers -71
[1743351.990751] r8152 3-1.2:1.0 eth4: get_registers -71
[1743352.014273] r8152 3-1.2:1.0 eth4: get_registers -71
[1743352.032413] r8152 3-1.2:1.0 eth4: get_registers -71
[1743352.050551] r8152 3-1.2:1.0 eth4: get_registers -71
[1743352.068691] r8152 3-1.2:1.0 eth4: get_registers -71
[1743352.086824] r8152 3-1.2:1.0 eth4: get_registers -71
[1743352.104971] r8152 3-1.2:1.0 eth4: get_registers -71
[1743352.123115] r8152 3-1.2:1.0 eth4: get_registers -71
[1743352.141250] r8152 3-1.2:1.0 eth4: get_registers -71
[1743352.923342] usb 3-1.2: reset SuperSpeed USB device number 9 using etxhci_hcd-170202
```

rickenwing, 2023-02-09:
NAS: DS1815+ (avaton) with 16GB RAM (2x8GB) 
DSM Version: DSM 7.1.1-42962 Update 3
Driver version: r8152-avoton-2.16.3-3.spk
Switch: Netgear MS510TXM (2.5GBe port)
Adapter: USB-C2500 Asus 2.5G ethernet usb adapter

DHCP worked just fine, albeit the nic originally was configured in as static, switched in DSM UI and worked. 

Seeing similar issues per other users with more recent driver. Speeds drop on transfer, but nothing fails and it picks back up soon after, both on iperf and a regular SMB transfer. No transfer or dropped packets at the nic level or switch level seen, just the expected Link down events from the USB reset. 

No crashes or manual intervention seen thus far with this type of event. 

iperf3 drop:

[  4]  46.00-47.00  sec   265 MBytes  2.22 Gbits/sec
[  4]  47.00-48.00  sec   241 MBytes  2.01 Gbits/sec
[  4]  48.00-49.01  sec  0.00 Bytes  0.00 bits/sec
[  4]  49.01-50.01  sec  0.00 Bytes  0.00 bits/sec
[  4]  50.01-51.01  sec  0.00 Bytes  0.00 bits/sec
[  4]  51.01-52.01  sec  0.00 Bytes  0.00 bits/sec
[  4]  52.01-53.00  sec   152 MBytes  1.29 Gbits/sec
[  4]  53.00-54.00  sec   278 MBytes  2.33 Gbits/sec
[  4]  54.00-55.00  sec   278 MBytes  2.33 Gbits/sec
[  4]  55.00-56.00  sec   278 MBytes  2.33 Gbits/sec
...

- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-300.00 sec  79.2 GBytes  2.27 Gbits/sec                  sender
[  4]   0.00-300.00 sec  79.2 GBytes  2.27 Gbits/sec                  receiver

relevent dmesg logs:
[ 3315.235310] r8152 3-2:1.0 eth4: get_registers -71
[ 3328.260778] r8152 3-2:1.0 eth4: get_registers -71
[ 3329.507506] r8152 3-2:1.0 eth4: get_registers -110
[ 3331.839302] etxhci_hcd-170202 0000:04:00.0: WARN Event TRB for slot 1 ep 2 with no TDs queued?
[ 3331.848981] etxhci_hcd-170202 0000:04:00.0: WARN Event TRB for slot 1 ep 3 with no TDs queued?
[ 3331.858708] r8152 3-2:1.0 eth4: get_registers -108
[ 3331.876519] r8152 3-2:1.0 eth4: get_registers -71
[ 3331.894271] r8152 3-2:1.0 eth4: get_registers -71
[ 3331.912043] r8152 3-2:1.0 eth4: get_registers -71
[ 3331.929805] r8152 3-2:1.0 eth4: get_registers -71
[ 3331.947579] r8152 3-2:1.0 eth4: get_registers -71
[ 3331.965339] r8152 3-2:1.0 eth4: get_registers -71
[ 3331.983091] r8152 3-2:1.0 eth4: get_registers -71
[ 3332.002121] r8152 3-2:1.0 eth4: get_registers -71
[ 3332.019861] r8152 3-2:1.0 eth4: get_registers -71
[ 3332.037632] r8152 3-2:1.0 eth4: get_registers -71
[ 3332.055416] r8152 3-2:1.0 eth4: get_registers -71
[ 3332.073157] r8152 3-2:1.0 eth4: get_registers -71
[ 3332.090929] r8152 3-2:1.0 eth4: get_registers -71
[ 3332.108686] r8152 3-2:1.0 eth4: get_registers -71
[ 3332.126462] r8152 3-2:1.0 eth4: get_registers -71
[ 3332.962362] usb 3-2: reset SuperSpeed USB device number 2 using etxhci_hcd-170202
[ 3346.579653] r8152 3-2:1.0 eth4: get_registers -71

ifconfig stats:
eth4      Link encap:Ethernet  HWaddr A0:36:BC:D9:36:AC
          inet addr:192.168.1.230  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: fe80::a236:bcff:fed9:36ac/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:538942837 errors:0 dropped:0 overruns:0 frame:0
          TX packets:101616259 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:798484270978 (743.6 GiB)  TX bytes:91837578772 (85.5 GiB)

Otherwise solid performance and thus far no major issues. Not sure if this is the same experience as others or not but I consider this good.