bb-qq/r8152 #46: [Error] DS2415+ (avaton) with Club 3D CAC-1420

tmkmuc, 2020-08-28:
### Description of the problem
Adapter installation worked. Read 224 GB/s, Write 175 GB/s. Read seems stable, write goes to 0 MB/s after a few minutes, then adapter was not reachable any more with ping. After reboot normal again, 2 minutes later next crash.

### Description of your products

DS2415+, 12x 4TB RAID 6, 4GB RAM. Worked stable with 1GB for 4 years.
Linux SRVDS2415 3.10.105 #25426 SMP Wed Jul 8 03:19:33 CST 2020 x86_64 GNU/Linux synology_avoton_2415+
DSM 6.2.3-25426 Update 2
 Club 3D CAC-1420
Direct connection to ASUS mainboard with 2.5GB onboard, CAT6.

messages after crash:
2020-08-28T11:41:46+02:00 SRVDS2415 smbd: ../source3/lib/recvfile.c:216 recvfile failed(-1): Broken pipe, count=[1048576] total tx/tx=[856800/856800] rwbytes=[856800/856800]
### Output of ``dmesg`` command

### Output of ``lsusb`` command
|__usb1          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (ehci_hcd 0000:00:16.0) hub
  |__1-1         8087:07db:0002 09  2.00  480MBit/s 0mA 1IF  ( ffffffd1ffffffb2ffffffdbffffffad) hub
    |__1-1.1     f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 65004C24D3020557)
|__usb2          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (Linux 3.10.105 etxhci_hcd-170202 Etron xHCI Host Controller 0000:04:00.0) hub
|__usb3          1d6b:0003:0310 09  3.00 5000MBit/s 0mA 1IF  (Linux 3.10.105 etxhci_hcd-170202 Etron xHCI Host Controller 0000:04:00.0) hub
  |__3-1         0bda:8156:3000 00  3.20 5000MBit/s 512mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 000000001)

````

### Output of ``ifconfig -a`` command
eth0      Link encap:Ethernet  HWaddr 00:11:32:3C:CB:EB
          inet addr:192.168.17.126  Bcast:192.168.17.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:16313 errors:0 dropped:0 overruns:0 frame:0
          TX packets:35513 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:2017748 (1.9 MiB)  TX bytes:24269030 (23.1 MiB)

eth1      Link encap:Ethernet  HWaddr 00:11:32:3C:CB:EC
          inet addr:169.254.173.118  Bcast:169.254.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

eth2      Link encap:Ethernet  HWaddr 00:11:32:3C:CB:ED
          inet addr:169.254.18.80  Bcast:169.254.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

eth3      Link encap:Ethernet  HWaddr 00:11:32:3C:CB:EE
          inet addr:169.254.156.120  Bcast:169.254.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

eth4 is CLUB3D adapter
eth4      Link encap:Ethernet  HWaddr A0:CE:C8:CD:45:9C
          inet addr:192.168.16.126  Bcast:192.168.16.255  Mask:255.255.255.0
          inet6 addr: fe80::a2ce:c8ff:fecd:459c/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:12951755 errors:0 dropped:0 overruns:0 frame:0
          TX packets:909047 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:19605287541 (18.2 GiB)  TX bytes:51418960 (49.0 MiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:3252 errors:0 dropped:0 overruns:0 frame:0
          TX packets:3252 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:471094 (460.0 KiB)  TX bytes:471094 (460.0 KiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
````

tmkmuc, 2020-08-28:
Tested with SAME Club3D adapter and SAME cable between two PCs, Windows 10, latest Realtek driver: 283 GB/s, Read and Write, completly stable. So the problem is the DS2415+ and/or the driver.

tmkmuc, 2020-08-28:
In the first posting, GB/s is wrong, should be MB/s of course.

bb-qq, 2020-08-29:
Thank you for the detailed explanation.

I am now suspecting [USB LPM issue](https://bugzilla.redhat.com/show_bug.cgi?id=1713657), but I could not find a way to change kernel parameters in Synology environment...

So could you try to change the autosuspend settings by this command instead?

```
echo -1 > /sys/module/usbcore/parameters/autosuspend
```

dankao, 2020-08-30:
```
10:45am xxxxx@xxxxx:/var/services/homes/xxxxx% cat /sys/module/usbcore/parameters/autosuspend
cat: /sys/module/usbcore/parameters/autosuspend: No such file or directory

10:45am xxxxx@xxxxx:/var/services/homes/xxxxx% ls -l /sys/module/usbcore/parameters/
total 0
-rw-r--r-- 1 root root 4096 Aug 30 10:46 authorized_default
-r--r--r-- 1 root root 4096 Aug 30 10:46 blinkenlights
-rw-r--r-- 1 root root 4096 Aug 30 10:46 initial_descriptor_timeout
-r--r--r-- 1 root root 4096 Aug 30 10:46 nousb
-rw-r--r-- 1 root root 4096 Aug 30 10:46 old_scheme_first
-rw-r--r-- 1 root root 4096 Aug 30 10:46 usbfs_memory_mb
-rw-r--r-- 1 root root 4096 Aug 30 10:46 usbfs_snoop
-rw-r--r-- 1 root root 4096 Aug 30 10:46 use_both_schemes
```

bb-qq, 2020-09-03:
How about `/sys/bus/usb/devices/3-1/power/autosuspend`? (`3-1` is obtained by output of `lsusb`)