bb-qq/r8152 #48: driver issues under sustained full duplex load on rtd1296

johndavisnz, 2020-09-05:
has anyone sucesfully used the rtd1296 driver - I'm seeing speed drop to zero (requiring if down/up to clear) testing it with full duplex loads (either 'real world' copying large files simultaneously in both directions, or synthetically testing using iperf3)

MKHB1981, 2020-09-06:
I tested it with a Synology DS418 and a CAC-1420. Unfortunately it's the same problem. The data rate drops to 0 after a few seconds. The Ethernet adapter still has a link, but is no longer displayed as an interface in the NAS.

When I use on a 1gbits switch it works for a couple of minutes. If I reduce the speed further with a USB2.0 extension cable, then data can be transmitted stably. I tried for about half an hour. Supporting the power supply with a Y-cable does not help. The behavior on both USB ports is also the same.

johndavisnz, 2020-09-06:
I'm on a ds418 as well - I've tested with both a direct link to a matching cablecreations 2.5gb usb3 adaptor on a pc, and via a 1gbe switch

At both 2.5gbe and 1gbe iperf3 will test sustained fine with the ds418 only receiving traffic, at 2.5gbe (direct) any loading of transmit traffic brings the link down, at 1gbe I can test transmit only fine, but and combined loading of transmit and receive brings the link down after a couple of seconds. 

Testing using a direct connection I tried manually forcing link speed down using eththool - at 100mbs fdx the link is stable loaded simultaneously in both directions, at 1000mbs fdx if I control test bandwidth using the --bandwidth option in iperf3 it's stable up to about 300mbs, beyond that again the link collapses

So looks to primarily be a transmit related issue

Limoges65, 2021-01-10:
The same happens to me using DS118. Link/Transfer collapses when under full speed, one single direction works. Latest 2.14.** RTD2196 used.

LuxKeiwoker, 2021-01-25:
I'm having the same issue also with DSM 7 and a DS418j. Did anyone find a resolution for this?

bb-qq, 2021-02-23:
I do not have ideas to resolve this situation. But I am interested in the result if the dongle connected via a self-powered USB hub with an AC adapter.

bb-qq, 2022-12-04:
I decided to merge topics on rtd1296 into issue #275.
If you notice anything new, please comment there.

closed 2022-12-04 by bb-qq