bb-qq/r8152 #50: I don't understand how to install on the NAS from Synology

mrAlexRus, 2020-09-27:
I tried different releases: r8152-monaco-2.12.0-1.spk and r8152-rtd1296-2.13.0-1.spk. 
Used different DSM: V. 6.1.7-15284 update 3 (DS3615xs) and V. 6.2.3-25426 update 2 (DS916+) 
I get an error everywhere: the package is not supported on the DiskStation platform or is not compatible with the current DSM version. 
So how do I install this driver? 
Network card on the RTL8156B chip, manufacturer OPEN-SMART

bb-qq, 2020-10-12:
What is the DSM version are you using? It seems that the error caused by version mismatch of the kernel.

bb-qq, 2021-01-04:
Close this issue due to no response. Please feel free to re-open the issue if you notice my comment.

closed 2021-01-04 by bb-qq