bb-qq/r8152 #51: Succes : DS916+ and PLUGABLE 2.5GBPS USB ETHERNET ADAPTER

Sjekke, 2020-10-09:
Working with DS916+ and PLUGABLE 2.5GBPS USB ETHERNET ADAPTER
After installation I had to restart the Synology NAS for it to show up.

Synology Version: DSM 6.2.3-25426 Update 2
Driver Version: 2.12.0-3

Sjekke, 2020-11-23:
It seems the USB ethernet sometimes stops working for 2 minutes and restarts. Suggestions?

bb-qq, 2020-11-29:
* Are you using the front USB port?
* What was the output of dmesg when the problem occurred?

Sjekke, 2020-11-29:
Yesterday I performed a factory reset of my ASUS AX11000 router and the Synology didn't stopped anymore since then. It's to early to confirm that this was the issue. I will see what happens in the few next days and keep you informed. 

It's a pity you can't make a new package for the Braswell platform with the latest driver.

Sjekke, 2020-11-30:
> * Are you using the front USB port?
> * What was the output of dmesg when the problem occurred?

[ 3837.279747] r8152 4-1:1.0 eth2: Tx timeout
[ 3837.288895] r8152 4-1:1.0 eth2: Tx status -2
[ 3837.293702] r8152 4-1:1.0 eth2: Tx status -2
[ 3837.298519] r8152 4-1:1.0 eth2: Tx status -2
[ 3837.303339] r8152 4-1:1.0 eth2: Tx status -2
[ 3840.910719] usb 4-1: reset SuperSpeed USB device number 2 using etxhci_hcd-170202
[11623.983582] init: sshd main process (9052) killed by TERM signal
[34437.086107] r8152 4-1:1.0 eth2: Tx timeout
[34437.094402] r8152 4-1:1.0 eth2: Tx status -2
[34437.099205] r8152 4-1:1.0 eth2: Tx status -2
[34437.104045] r8152 4-1:1.0 eth2: Tx status -2
[34437.108868] r8152 4-1:1.0 eth2: Tx status -2
[34440.673858] usb 4-1: reset SuperSpeed USB device number 2 using etxhci_hcd-170202

Sjekke, 2020-12-01:
Happend again ... 
[  134.921499] r8152 4-1:1.0 eth2: v2.12.0 (2019/04/29)
[  134.927083] r8152 4-1:1.0 eth2: This product is covered by one or more of the following patents:
               		US6,570,884, US6,115,776, and US6,327,625.
               
[  135.865461] init: nmbd main process (12681) killed by TERM signal
[  136.118925] init: scsi_plugin_server main process (11599) killed by TERM signal
[  136.432287] init: nmbd main process (13324) killed by TERM signal
[  137.893409] IPv6: ADDRCONF(NETDEV_UP): eth2: link is not ready
[  138.526818] IPv6: ADDRCONF(NETDEV_CHANGE): eth2: link becomes ready
[  138.535565] r8152 4-1:1.0 eth2: carrier on
[  139.813356] init: nmbd main process (13402) killed by TERM signal
[  140.262798] init: scsi_plugin_server main process (13270) killed by TERM signal
[  140.532865] init: nmbd main process (14147) killed by TERM signal
[  141.082092] init: nmbd main process (14429) killed by TERM signal
[  141.477127] init: iscsi_pluginserverd main process (14119) killed by TERM signal
[  141.489693] init: iscsi_pluginengined main process (14115) killed by TERM signal
[  141.539402] init: scsi_plugin_server main process (14132) killed by TERM signal
[  142.003179] init: nmbd main process (14672) killed by TERM signal
[  142.591212] init: iscsi_pluginserverd main process (14631) killed by TERM signal
[  142.618037] init: iscsi_pluginengined main process (14627) killed by TERM signal
[  142.688833] init: scsi_plugin_server main process (14645) killed by TERM signal
[  143.344665] usbcore: registered new interface driver snd-usb-audio
[  143.527075] usbcore: registered new interface driver snd-usb-hiface
[  143.544213] aufs 3.10.x-20141110
[  143.657233] Bridge firewalling registered
[  149.595044] init: scsi_plugin_server main process (15062) killed by TERM signal
[  152.551396] init: iscsi_pluginserverd main process (16550) killed by TERM signal
[  152.561721] init: iscsi_pluginengined main process (16545) killed by TERM signal
[  152.591235] init: scsi_plugin_server main process (16561) killed by TERM signal
[  152.998765] IPv6: ADDRCONF(NETDEV_UP): docker0: link is not ready
[  158.342924] device dockerb64fada entered promiscuous mode
[  158.349188] IPv6: ADDRCONF(NETDEV_UP): dockerb64fada: link is not ready
[  161.387879] cgroup: runc (17932) created nested cgroup for controller "blkio" which has incomplete hierarchy support. Nested cgroups may change behavior in the future.
[  161.894721] IPv6: ADDRCONF(NETDEV_CHANGE): dockerb64fada: link becomes ready
[  161.902690] docker0: port 1(dockerb64fada) entered forwarding state
[  161.909736] docker0: port 1(dockerb64fada) entered forwarding state
[  161.917061] IPv6: ADDRCONF(NETDEV_CHANGE): docker0: link becomes ready
[  164.433107] FAT-fs (synoboot2): fat_set_state: set=1 process=synoddsmd pid=18046
[  164.444524] FAT-fs (synoboot2): fat_set_state: set=0 process=synoddsmd pid=18046
[  176.944632] docker0: port 1(dockerb64fada) entered forwarding state
[  280.583094] init: scsi_plugin_server main process (17330) killed by TERM signal
[ 2908.687439] perf interrupt took too long (2503 > 5000), lowering kernel.perf_event_max_sample_rate to 50000
[43947.633198] ------------[ cut here ]------------
[43947.638380] WARNING: at net/sched/sch_generic.c:255 dev_watchdog+0x24b/0x260()
[43947.646458] NETDEV WATCHDOG: eth2 (r8152): transmit queue 0 timed out
[43947.653698] Modules linked in: bridge stp aufs snd_usb_hiface macvlan veth snd_pcm_oss xt_conntrack snd_mixer_oss xt_addrtype snd_usb_audio ipt_MASQUERADE snd_pcm xt_REDIRECT snd_timer snd_hwdep snd_usbmidi_lib xt_nat snd_rawmidi iptable_nat snd_seq_device nf_nat_ipv4 snd nf_nat snd_page_alloc soundcore r8152(OF) cifs udf tcm_loop(O) iscsi_target_mod(O) isofs loop target_core_ep(O) target_core_multi_file(O) target_core_file(O) target_core_iblock(O) target_core_mod(O) syno_extent_pool(PO) rodsp_ep(O) nfsd exportfs rpcsec_gss_krb5 syno_hddmon(P) hid_generic usbhid hid usblp usb_storage uhci_hcd ehci_pci ehci_hcd xt_mac xt_geoip(O) compat_xtables(O) nf_conntrack_ipv6 nf_defrag_ipv6 ip6table_filter ip6_tables xt_recent xt_iprange xt_limit xt_state xt_tcpudp xt_multiport xt_LOG nf_conntrack_ipv4 nf_defrag_ipv4
[43947.733909]  nf_conntrack iptable_filter ip_tables x_tables braswell_synobios(PO) leds_lp3943 exfat(O) btrfs i915 drm_kms_helper cfbfillrect cfbcopyarea cfbimgblt output drm fb fbdev intel_agp intel_gtt agpgart video backlight button synoacl_vfs(PO) zlib_deflate hfsplus md4 hmac libcrc32c compat(O) igb(O) i2c_algo_bit e1000e(O) vxlan fuse vfat fat crc32c_intel aesni_intel glue_helper lrw gf128mul ablk_helper arc4 cryptd ecryptfs sha256_generic sha1_generic ecb aes_x86_64 authenc des_generic ansi_cprng cts md5 cbc cpufreq_powersave cpufreq_performance acpi_cpufreq mperf processor thermal_sys cpufreq_stats freq_table dm_snapshot crc_itu_t crc_ccitt quota_v2 quota_tree psnap p8022 llc sit tunnel4 ip_tunnel ipv6 sg etxhci_hcd xhci_hcd usbcore usb_common [last unloaded: braswell_synobios]
[43947.810798] CPU: 0 PID: 0 Comm: swapper/0 Tainted: PF          O 3.10.105 #25426
[43947.819071] Hardware name: Synology Inc. DS916+/Type2 - Board Product Name, BIOS M.215 3/2/2016
[43947.828801]  ffffffff814b5a6b ffffffff81036b08 0000000000000000 ffff88027fc03e68
[43947.837105]  0000000000000001 0000000000000000 00000000000000d0 ffffffff81036b69
[43947.845410]  ffffffff8173c6d8 ffff880200000030 ffff88027fc03e78 ffff88027fc03e38
[43947.853738] Call Trace:
[... cut ...]
[43947.999186] r8152 4-1:1.0 eth2: Tx status -2
[43948.004009] r8152 4-1:1.0 eth2: Tx status -2
[43951.602358] usb 4-1: reset SuperSpeed USB device number 2 using etxhci_hcd-170202
[60688.080796] r8152 4-1:1.0 eth2: Tx timeout
[60688.089981] r8152 4-1:1.0 eth2: Tx status -2
[60688.094795] r8152 4-1:1.0 eth2: Tx status -2
[60688.099638] r8152 4-1:1.0 eth2: Tx status -2
[60688.104450] r8152 4-1:1.0 eth2: Tx status -2
[60691.674298] usb 4-1: reset SuperSpeed USB device number 2 using etxhci_hcd-170202
admin@InfectedNAS:/$

Sjekke, 2020-12-07:
No info?

Sjekke, 2020-12-07:
It seems that using the front usb is the solution. Nevertheless what is your opinion if you see my feedback?

bb-qq, 2020-12-13:
Some users reported stability issues when they use front ports, so I would have like to confirm that at first.
However, some users also reported performance/stability issues on old platforms caused by USB host controller/driver. If using front ports does not work for you, your issue may be the same one.

Sjekke, 2020-12-13:
> Some users reported stability issues when they use front ports, so I would have like to confirm that at first.
>
> However, some users also reported performance/stability issues on old platforms caused [...]

Since I use the front port, my adapter is stable for a week now. No drops at all. 

- Can you make a new driver with latest version. 
- Any idea if your drivers will work on DSM 7.0?

bb-qq, 2020-12-13:
Please check [the new release](https://github.com/bb-qq/r8152/releases/tag/2.14.0-2) and #68.

Sjekke, 2020-12-13:
> Please check [the new release](https://github.com/bb-qq/r8152/releases/tag/2.14.0-2) and #68. 

Ok I will test them and provide feedback.

Sjekke, 2020-12-13:
> Please check [the new release](https://github.com/bb-qq/r8152/releases/tag/2.14.0-2) and #68. 

Task Scheduler has completed a triggered task.

Task: Eth 2.5
Start time: Sun, 13 Dec 2020 11:02:50 GMT
Stop time: Sun, 13 Dec 2020 11:02:51 GMT
Current status: 0 (Normal)
Standard output/error:
eth2: ERROR while getting interface flags: No such device
Copyright (c) 2003-2020 Synology Inc. All rights reserved.

Usage: synonet (Version 25426)
   --help 
   --show 
   --dhcp iface
   --manual iface ip mask
   --check ip {mask}
   --la_new bondX {1|2} ethX ethX 
   --la_del bondX
   --la_enum
   --la_get bondX
   --la_exist
   --set_gateway gateway
   --set_dns dns
   --dns_update
   --set_mtu iface MTU
   --set_hostname new_hostname
   --get_hostname
   --vlan_enum
   --vlan_create device vlan_id {dhcp| ip mask}
   --vlan_remove vlan_interface
   --vlan_exist
   --set_pppoe iface username password
   --get_if pppoe
     (only support pppoe now)
   --wake xx:xx:xx:xx:xx:xx ethX
   --get_block ethX
   --set_block ethX {ha|aha|virtualization|mailplus} {true|false}
   --dhcp_server_change_interface old_interface new_interface
   --get_mac_addr iface {original | ha}
   --set_ip -4/-6 ethx {flush | add | del} {empty | ip/prefix_len}
       ex1    : --set_ip -4 eth0 add 192.168.55.66/16
       ex2    : --set_ip -6 eth0 del 2001:b011:4002:f0:211:32ff:fe22:efcb/64
       ex3    : --set_ip -4 eth0 flush
       ex4    : --set_ip -6 eth0 flush
       note    : it's different from --manual, ifcfg doesn't modified
   --ipv6-init (Do NOT use this after system booting)

It's not getting an IP.

Sjekke, 2020-12-13:
> Please check [the new release](https://github.com/bb-qq/r8152/releases/tag/2.14.0-2) and #68.

It works ... but ... it was not easy to get it back working. 

Before you install the latest driver I advise to 👍 

- De-Install old drivers
- Reboot Synology
- Install new drivers
- Reboot Synology

Normally everything should work. I had problems with the DHCP server in my router but finally after using and not using the DHCP server it works.

phantomtypist, 2020-12-14:
@Sjekke good to hear.  I'm going to try it out on my DS716+ II in a couple of days.

Sjekke, 2020-12-14:
> @Sjekke good to hear.  I'm going to try it out on my DS716+ II in a couple of days.

Good luck!

phantomtypist, 2020-12-17:
@bb-qq @Sjekke I've verified the latest beta driver (as of writing this) for Braswell is working with the DS716+ II.  I used the bottom rear USB port (top was occupied by thumb drive).  Been solid for at least two days and I also copied roughly 7TB over the network to it.  I am using the CableCreations USB 3.0 to 2.5GbE network adapter that has the Realtek RTL8156 chipset inside it.  On the other end I paired it with a FLYPROFiber 10G SFP+ to RJ45 transceiver, which is plugged into a MikroTik CRS305-1G-4S+IN 10GbE switch.

phantomtypist, 2020-12-17:
Forgot to mention I'm running DSM 6.2.3-25426 Update 2

Sjekke, 2020-12-17:
> @bb-qq @Sjekke I've verified the latest beta driver (as of writing this) for Braswell is working with the DS716+ II.  I used the bottom rear USB port (top was occupied by thumb drive).  Been solid [...]

Same here. I'm testing with the rear USB ports and till now everything stays stable. Yes!

phantomtypist, 2020-12-17:
I have an RS217 I can think I can test, but it'll be a while before I get around to that one.

Sjekke, 2020-12-18:
Still working perfect without instability on my DS916+ after 4 days.

phantomtypist, 2020-12-18:
I had a blip this morning, but I'm not really sure what caused it.  The Synology was unresponsive over the network for about 15 minutes give or take a bit, but came back on its own.  It didn't reboot.  I guess I'll just wait to see if it happens again.

Sjekke, 2020-12-18:
> I had a blip this morning, but I'm not really sure what caused it.  The Synology was unresponsive over the network for about 15 minutes give or take a bit, but came back on its own.  It didn't reboot.  I guess I'll just wait to see if it happens again. 

That's what I experienced with the old drivers.

phantomtypist, 2020-12-29:
Kept getting those random network drops with the rear bottom USB 3.0 port.  Was happening like once every day or two.  Nothing in dmesg about what was happening.

Switched it to the front USB 3.0 port on the DS716+ II.  Will see if that fares better.

Sjekke, 2020-12-29:
> Kept getting those random network drops with the rear bottom USB 3.0 port.  Was happening like once every day or two.  Nothing in dmesg about what was happening.
> 
> 
> 
> Switched it to the front USB 3.0 port on the DS716+ II.  Will see if that fares better.

I don't have network hick-ups. My adapter is connected on the back.

Sjekke, 2021-01-04:
This can be closed. 
My DS916+ is working perfectly with the latest drivers

bb-qq, 2021-02-23:
Thank you. I will leave this issue open until I create the wiki to summarize working reports.

Sjekke, 2021-05-30:
I installed the latest test driver, so far so good. But when I use the DScam app on my Ipad the NAS stops working. After 5min. the adapter comes available again ... Can I test something?

bb-qq, 2021-07-11:
It is unlikely, but the USB port may not be supplying enough power. In that case, you can solve the problem by using a self-powered USB hub with an AC adapter.

Sjekke, 2021-07-11:
> It is unlikely, but the USB port may not be supplying enough power. In that case, you can solve the problem by using a self-powered USB hub with an AC adapter.

In the meanwhile I'm on DSM7 ... and works perfectly with the new SPK DSM7

bb-qq, 2021-07-11:
Thank you for the reply. I close this issue.

closed 2021-07-11 by bb-qq