bb-qq/r8152 #53: RTL8156 on Synology DS1812+ Cedarview

bryanhunwardsen, 2020-10-21:
Breaking w/ template format just to cross post the aquantia chip/driver has the same issue as this one does on this box.
It is pretty well detailed there, but anyone searching for the issue chip/driver can refer to that issue log for details.

https://github.com/bb-qq/aqc111/issues/14#issuecomment-714002426

bb-qq, 2020-11-01:
Yes, I am suspecting that old platforms have some issues of with the USB Host Controller layer(software or hardware). 

But I found [a thread of r8152 related to performance issue in the kernel Bugzilla]( https://bugzilla.kernel.org/show_bug.cgi?id=205923). This patch may have something to do with the issue. Could you try this test version? https://github.com/bb-qq/r8152/releases/tag/2.12.0-3_test

bryanhunwardsen, 2020-11-01:
I returned all 5 of my RT 2.5 adapters, I just ordered another that should arrive tomorrow to test it. But, I dont have high hopes, I did just a pure USB data test on the ports from a usb 3.2 Gen 2(PCIe NVME) that I have gotten 800MBps+Read/600MBps+Write connected to my Desktop, the the 1812+ usb3 ports barely managed 70MbpsRead/55MbpsWrite with it that leads me t believe there is something very insidious going on with the Syno hardware or Kernel/Driver that is not specific to just ethernet adapters. Searching the net is littered with posts showing this issue across even current Syno products, but I dont have those to confirm against, Ill report here as soon as Ive run a test on the test driver. Thanks!

bryanhunwardsen, 2020-11-04:
So I got another (Sabrent) RTL8156 2.5GBE usb adapater, and tested. No improvement with the patched driver.
Max speed was about 110MBps but averaged 60-70MBps on a write from a 2.5+GBps source reading off a PCI-E gen3 nvme ssd  over SMB protocol. Read peaked at 80MBps and averaged about 75MBps. It seems like the port is being artificially limited by hardware(chip/chip fw) or software (kernel/OS). The fact that Synology has not addressed it in any manner over the last 8 years leads me to believe they put sub-spec hardware in and advertised it as USB3.0 compliant.

Im not entirely certain there is any point in keeping this or the AQ driver open for syno cedarview, if it was just the network driver Id say lets keep digging, but the equally poor usb drive performance to me rules out a network driver issue. But Im willing to hold out hope until they release their 2021 model and I kiss this garbage usb 3  synology goodbye.

bb-qq, 2020-11-29:
Sorry for my late reply and thank you for testing the modified driver.

> the 1812+ usb3 ports barely managed 70MbpsRead/55MbpsWrite with it

This verification convicted me it is a host controller issue. Thank you again.