bb-qq/r8152 #54: Realtek 8156 Version 2.14 is on Realtek web site

tmkmuc, 2020-10-22:
Please update the driver, and I will try it once more.

bb-qq, 2020-11-01:
I uploaded 2.14 driver packages. (but it only supports recent platforms.)

LeniReinfe, 2020-11-10:
pls bb-qq could you compile to Broadwell ? (Virtual Machine doesn't works with 21-12.0-3) >thk so much!

LeniReinfe, 2020-11-10:
> I uploaded 2.14 driver packages. (but it only supports recent platforms.)

pls bb-qq could you compile 2.14 to Broadwell ? (Virtual Machine doesn't works with 21-12.0-3) >thk so much!

Sjekke, 2020-11-14:
Is the latest driver already available for the Braswell platform?

bb-qq, 2020-11-29:
Unfortunately, the latest Realtek drivers do not support old kernels. Technically we can modify the driver to support old kernels, but I do not have much time for doing that. Sorry...

bb-qq, 2020-12-13:
Please check: https://github.com/bb-qq/r8152/releases/tag/2.14.0-2

closed 2021-05-16 by bb-qq