bb-qq/r8152 #56: DS214play Success Report

k-vq, 2020-10-26:
DS214play [Evansport] connect to Windows directly via a pair of UGreen adapters

iperf3 with builtin 1000Mbps: 940Mbps consistently
iperf3 with USB 2.5Gbe (MTU 1500): 1.5-1.8Gbps, quite varied
iperf3 with USB 2.5Gbe (MTU 9000): 1.9Gbps consistently

Windows SMB copy speed now is about 140MB/s (was 110MB/s). It is not very different between 1500 and 9000 MTU. I suspect the limitation is because of the SATA speed of DS214play (I don't use RAID).

Thank @bb-qq  a lot for the great work.

k-vq, 2020-10-27:
After using a multi-gig switch, the Windows copy speed now is about 160MB/s @ 1500MTU. I tried 9000MTU but the speed is very unstable.

jplourde71, 2023-05-31:
I set the MTU to 9000 manually. But MTU is up at 1500 on DS214play running DSM 7.1.1-42962 Update 5

Network Status: 2500 Mbps, Full duplex, MTU 1500

devosdan, 2024-01-17:
@ jplourde71. Same settings as yours and my speed is maximum 80 MB/s. Have 2 HDD 4TB Western Digital in Raid 1 . Impossible to force network status to 9000MTU.