bb-qq/r8152 #57: Report - DS1520+  Working

noblemd00, 2020-10-26:
AJA benchmark 
Avg. 250 MB/Sec Read/Write 
4GB XAVC 4K file

[image]

DS1520+
DSM 6.2.3-25426 Update 2

Connected via NETGEAR 10G Switch (XS508M) with Cat 8 cables.

r8152-geminilake-2.13.0-1
CableCreation USB 3.0 to 2.5 Gigabit LAN
RTL8156

[https://smile.amazon.com/gp/product/B07VNFLTLD](url)

ASUS AM4 TUF Gaming X570-Plus (Wi-Fi) Motherboard
Ryzen 3950x 16 GB memory
2.5GBase-T PCIe Network Adapter - Realtek RTL8125-CG

davidgirard, 2020-10-26:
Any stability issues?

Thanks,
David

[201026 1835-00]<https://nam05.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F16325865%2F97235844-265e4400-17ba-11eb-8d24-6e79806a28be.png&data=04%7C01%7C%7C059cac5dc73546308c5008d87a025b21%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637393497814303936%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C1000&sdata=%2FFIOEIuUNpiSjbC9EUIhJAEby8U%2BTMSBjorJ%2BzNj8rE%3D&reserved=0>

noblemd00, 2020-10-26:
No stability issues so far (1 day - numerous files transferred and multiple videos - movies/TV - watched). 

Also, Active Backup for Business completed scheduled tasks without problem and Virtual Machines accessed without issue.

davidgirard, 2020-10-26:
Thanks for sharing! I’ll try on my ds220+ later tonight (also geminilake).

noblemd00, 2020-10-27:
Update: Virtual Machines do not recognize the USB NIC - does not show in NIC list. For VMs to access network/internet must have at least one built-in NIC connected.

[image]

hstriepe, 2024-12-18:
Working on my DS 1520+ with WAVLINK USB-C 5 Gigabit Adapter WL-NWU340G connected to Mokerlink 10GB Managed switch.
Getting about 350MBps from BTRFS spinner arrays.

dy1945, 2025-07-20:
> Update: Virtual Machines do not recognize the USB NIC - does not show in NIC list. For VMs to access network/internet must have at least one built-in NIC connected.
> 
> [image]

Fixed by reinstall virtual machine manager.  
 step 1:  uninstall vmm 
 step 2:  shutdown the open vswitch on the network interface settings(btw, vmm requires ovs and prevent to shut it down)
 step 3: reinstall  vmm， it will set ovs for all network interfaces.

backup your VMs before reinstall， then uninstall will not remove them.

[image]

[image]