bb-qq/r8152 #59: Report - DS418j not Working

awolswat, 2020-11-04:
Hi bb-qq,

Thank you for your contribution.

I try to install "r8152-rtd1296-2.14.0-1.spk" into my DS418j.
It seen install OK and the system can found the "UGREEN USB 2.5G Ethernet Adapter".
But, when i try to send files to nas, after a few second NAS will be halt. 
Can you help to solve this problem?
Thanks again.

---------------------------------------------------
Model 	: DS418j
CPU	: Realtek RTD1293	Dual Core
Package Arch : rtd1296
UGREEN USB 2.5G Ethernet Adapter
---------------------------------------------------

LuxKeiwoker, 2021-01-25:
I'm having the same issue even with DSM 7.0 beta

Marc176, 2021-09-10:
I also have a DS420j with RTD1296 and same issue. We need a working driver for rtd1296

bb-qq, 2021-09-24:
I dropped support of rtd1296. Please see #152 for details.

bb-qq, 2022-02-05:
I created an experimental driver with low memory consumption. Could you try this?
https://github.com/bb-qq/r8152/releases/download/2.15.0-5/r8152-rtd1296-2.15.0-6_rxbuf-test.spk

bb-qq, 2022-12-04:
I decided to merge topics on rtd1296 into issue #275.
If you notice anything new, please comment there.

closed 2022-12-04 by bb-qq