bb-qq/r8152 #60: RTL8152 2.5gb > bb-qq Driver Works with bromolow Synology 3617xs ! (but doesnt works Virtual Machine)

LeniReinfe, 2020-11-06:
### Description of the problem
Driver RTL8152 - 2.12.0-3 Works! DSM 6.2.3 Synology detected USB2.5 ethernet (CableCreation)

> Replace this line with your answer.
Virtual Machine Manager can't find IP address USB2.5 Realtek

### Description of your products

> - the product name of your NAS model 3617xs
>     - and output of ``uname -a`` command
>     - DSM version 6.2.3 update2
> - the product name of the ethernet dongle
CABLECREATION usb2.5g

### Description of your environment
Xpenology Ok
Virtual Machine (Windows10 and Linux) doesn't works detect USB

> - connect directly to a PC or using a hub -- DIRECTLY
>     - PC: NIC model name or chipset name - REALTEK
>     - Hub: model name
> - cable category (Cat.5e or Cat.6 or Cat.7) - CAT6

### Output of ``dmesg`` command

````
(paste output here)
````

### Output of ``lsusb`` command

````
(paste output here)
````

### Output of ``ifconfig -a`` command

````
(paste output here)
````

bb-qq, 2020-11-29:
The current start-up script does not support creating bond interfaces due to driver loading order.
Modifying some config files to load the driver at an earlier stage may resolve the issue, but I have no idea about that.

bb-qq, 2020-11-29:
The method written in issue #63 seems to be a workaround for this issue.

LeniReinfe, 2020-11-29:
Understood bb-qq,  Your compilation was useful to me.  If you made an Realtek upgrade for "broadwell" will be appreciate. thk so much for your effort!

bb-qq, 2025-01-01:
This Issue is closed because the current version supports Open vSwitch and can also be used with VMM. If you have any questions, please feel free to re-open.

closed 2025-01-01 by bb-qq