bb-qq/r8152 #65: synology_bromolow_3615xs can use this driver?

greenmon86, 2020-11-19:
### Description of the problem
I want try the UGREEN USB3.0 2.5GB USB Lan for my 3615xs, but I cannot find the driver for synology_bromolow_3615xs.

The driver not available for bromolow?

Linux xxxxxx 3.10.105 #23739 SMP Tue Jul 10 00:15:23 CST 2018 x86_64 GNU/Linux synology_bromolow_3615xs
DSM 6.2-23739 U2

imontesp, 2020-11-23:
Hello. The driver wtitten by "bbqq" working ok with this adapter. I am using it on an Apollolake Synology 218+. I do not know if there is any for your platform.

greenmon86, 2020-11-23:
Hi, 

i using this adapter success using on 3617xs, but not 3615xs due to cannot find the correct for bromolow.
https://github.com/bb-qq/r8152/releases/tag/2.12.0-3

bb-qq, 2021-02-23:
I created the driver for Bromolow based platforms.
https://github.com/bb-qq/r8152/releases/download/2.14.0-2/r8152-bromolow-2.14.0-2.spk
// The SDK of this platform was broken for a long time, but I just tried it and it was fixed, so I was able to create it.

Please let me know whether it works.

bb-qq, 2021-07-11:
I close this issue. Create a new issue if you have problems, please.

closed 2021-07-11 by bb-qq