bb-qq/r8152 #66: Nuevo icono

imontesp, 2020-11-22:
Hello. Congratulations for a good job. It is running ok on 218+. Can yoy change the icon of the package fot a square image? An ethernet plug icon should be perfect in my opinion. Thanks.

imontesp, 2020-11-23:
[image]

imontesp, 2020-11-27:
Hello again, I try to compile the spk package with the icon published below, but my knowledge about this topic is very poor... Can anybody that know about this topic compile the spk file with this icon for the apollolake platform fo me? I would very grateful for this. Thanks.

Sjekke, 2020-11-28:
I don't think the author will answer. I like the drivers and I can use them ... but when people ask questions it stays a question. It's a pity the author of this github is not communicative.

bb-qq, 2020-11-29:
Sorry for my late reply. I will update the icon in the next release. Thank you.
By the way, is this icon originally created by you or downloaded from some free material site?

imontesp, 2020-12-03:
> Lo siento por mi respuesta tardía. Actualizaré el icono en la próxima versión. Gracias.
> Por cierto, ¿este icono fue creado originalmente por usted o descargado de algún sitio de material gratuito?

Hello. Thanks for your answer. The icon was found on free libraries in internet. I have a mini updade of your project. I have maked a update of the original package (spk) file. I change the ICON_PACKAGE and re-zip it in TAR format. Then i have renamed with SPK extension and it works great.

bb-qq, 2020-12-13:
Thank you for your answer. Where did you get the image? I would like to confirm its license.

imontesp, 2020-12-16:
Hello again. I do not remember where I found the icon, but it be free license. I am sure of that.

bb-qq, 2025-06-21:
In the recently released 2.20.1, I changed the icon design to match the look and feel of the latest Synology icons. I tried to make it so that it doesn't look out of place when placed next to other icons.

I hope you like it. Thank you for your suggestion.

closed 2025-06-21 by bb-qq