bb-qq/r8152 #68: Driver can not work on DSM7.0 Beta (release date 12/8/2020)

PatrickPTC, 2020-12-08:
DS920+ could normally use latest version drivers,  today when update NAS to DSM7.0, driver can not start (actually start serval seconds then auto turn off) . And do you have good way to make it rework?  NAS can not be downgrade.

Sjekke, 2020-12-08:
So it means DSM7.0 doesn't support USB ethernet drivers. So it no step forward. I hope you get an answer from the author because he is not very responsive which is a bummer because he is doing a great job.

Garfield999, 2020-12-08:
> NAS can not be downgrade.

Wrong. There's a workaround to downgrade to 6.2.3 using putty and editing the version file in etc.defaults

bb-qq, 2020-12-13:
Probably we can build the driver for DSM 7, but unfortunately I noticed that I do not have much enough disk space on my Linux development machine to download SDKs and build the drivers at this time. Please wait just moments...

LuxKeiwoker, 2021-01-18:
> Probably we can build the driver for DSM 7, but unfortunately I noticed that I do not have much enough disk space on my Linux development machine to download SDKs and build the drivers at this time. Please wait just moments...

I'd be interested as well. I was dumb enough to update my RTD based 418j to DSM 7b and just realized that there is a 2.5 GBe driver available for the RTD devices as well *facepalm*. Now I'm missing even more...

BeatSkip, 2021-01-23:
It's doable with a bit of fiddling to get it working. #88

Garfield999, 2021-03-30:
> Probably we can build the driver for DSM 7, but unfortunately I noticed that I do not have much enough disk space on my Linux development machine to download SDKs and build the drivers at this time. Please wait just moments...

I hope you'll find some free disk space :)

BeatSkip, 2021-03-30:
> Probably we can build the driver for DSM 7, but unfortunately I noticed that I do not have much enough disk space on my Linux development machine to download SDKs and build the drivers at this time. Please wait just moments...

Just an update, getting the install working is fine, but the driver itself has some issues with DSM 7. Something changed and multiple people trying my compiled versions get either a boot loop or connection cutting out. i have tried multiple driver versions and multiple toolkits. I'm afraid it isn't just a simple "build with new SDK" solution.

Garfield999, 2021-03-31:
I tried compiling module sources (wget r8152 2.14) myself for apollolake and ended in a reboot after insmod the module on a ds918+

bb-qq, 2021-07-08:
I'm sorry to keep you waiting. I've busy so I can't make any promises, but I will try to upgrade the NAS and examine the solution mentioned in #88 hopefully this weekend.

bb-qq, 2021-07-11:
I released new drivers supporting DSM 7.0. Closing.

closed 2021-07-11 by bb-qq