bb-qq/r8152 #73: Pluggable USBC-E2500 working on DS920+

TC72, 2020-12-15:
I bought 2 of the pluggable devices to connect my M1 Mac Mini direct to the Synology NAS without any router.
After I installed the driver I had to restart the NAS to see the network interface.

I have it connected to the rear USB port and no issues yet.
Transfer speeds have gone from 108/96 to 270/250.

The only slight issue is my Mac isn't allowing me to configure it for jumbo packets.
In the Network panel if I try to set it up manually there is no option for 2.5G and the max packet size is limited to 1500.

Thank you so much, huge upgrade in speed for only £60.

aminebizid, 2020-12-16:
Hello,
Could you please send us the link to the driver you installed on your NAS.

TC72, 2020-12-16:
https://github.com/bb-qq/r8152/releases/tag/2.14.0-1

The one I used was geminilake:
https://github.com/bb-qq/r8152/releases/download/2.14.0-1/r8152-geminilake-2.14.0-1.spk

hero2017, 2023-08-19:
Just bought this yesterday and have run a few tests on my DS920+. Connected to back USB. Works well with the latest DSM which as of today is DSM 7.2-64570 Update 1. iperf3 test results:

`
[image]

`

luanbui, 2025-02-07:
Anyone try with the WAVLINK 5Gbps USB C to Ethernet Adapter?

https://www.amazon.com/dp/B0DBY23BL9