bb-qq/r8152 #76: is ds218j support?

centralhardware, 2020-12-17:


detsuku, 2021-01-13:
I also want to use this NAS.

LuxKeiwoker, 2021-01-25:
I guess the DS218j has the rtd1296 Realtek SoC? If so, you can try and install the rtd1296 driver.

detsuku, 2021-02-14:
Hello.

Thanks for the reply.
Also sorry for the late reply.

Where can I get the rtd1296 driver from?

I can't access the official page with the specs on it.
Reviews are the best source of information, but it looks like the CPU of the DS218j is a Marvell Armada 385 88F6820.

LuxKeiwoker, 2021-02-14:
You can find the appropriate drivers in the release page

https://github.com/bb-qq/r8152/releases

bb-qq, 2021-02-23:
This is the direct link to the driver.
https://github.com/bb-qq/r8152/releases/download/2.14.0-2/r8152-rtd1296-2.14.0-2.spk

I close this issue. If you have another question, please create a new issue.

closed 2021-02-23 by bb-qq