bb-qq/r8152 #78: Driver request for Armada375

csihama, 2020-12-25:
Hi,

Could anyone implemet driver for this model?:

DS215j | Marvell Armada 375 88F6720 | Dual Core | 2 | Armada375 | DDR3 512 MB

It would be very nice if I could make my nas 2.5gbit/s fast.

Have a nice day :)

bb-qq, 2021-01-04:
I uploaded the driver for the armada375 platform.
https://github.com/bb-qq/r8152/releases/download/2.14.0-2/r8152-armada375-2.14.0-2.spk

I would appreciate it if you report whether it works or not.

csihama, 2021-01-04:
Hi,

Thank you for the driver, I'll buy a compatible usb to ethernet adapter (from ebay or somewhere), then I'll test and report. 

Thank you very much, if it works, then I don't need to buy new NAS.
And Happy New Year!

csihama, 2021-01-07:
Hi,

I can confirm it works, but it shows up as 1000Base ethernet only, I will test the speed via iperf3, when I have time.

bb-qq, 2021-02-20:
Thank you for testing. I close this issue.
If you have any problems, please create another issue for it.

closed 2021-02-20 by bb-qq