bb-qq/r8152 #80: Driver request for Intel Core i5-4570

vinklerma, 2021-01-04:
Hi,

Could anyone implemet driver for this model?

DS3615XS| XPEnology Intel Core i5-4570| DDR3 16GB

Have a nice day

bb-qq, 2021-02-23:
I created the driver for Bromolow based platforms. 
https://github.com/bb-qq/r8152/releases/download/2.14.0-2/r8152-bromolow-2.14.0-2.spk
// The SDK of this platform was broken for a long time, but I just tried it and it was fixed, so I was able to create it.

Please let me know whether it works.

vinklerma, 2021-02-23:
Greetings,

Thanks for the driver, but it not works with: pci-e 8125

What information do you need for add this hardware for the driver list?

bb-qq, 2021-02-27:
Oops, sorry, I didn't notice that it was the model number for the PCIe version. Its driver is completely different from the driver for USB one and I do not have a plan to create it currently.

closed 2021-07-11 by bb-qq