bb-qq/r8152 #82: DS716 II driver works nearly 100 %

Cess872, 2021-01-09:
The driver is working, but devices can only connect with static IP. 
DHCP devices have no access to the nas.

My environment:
DS716 II
DSM version 6.2.3-25426 Update 3
Port: USB3.0 at back
Club 3D CAC-1420
driver: r8152-braswell 2.14.0-2

Sjekke, 2021-01-09:
> The driver is working, but devices can only connect with static IP. 
> 
> DHCP devices have no access to the nas.
> 
> 
> 
> My environment:
> 
> DS716 II
> 
> DSM version 6.2.3-25426 Update 3
> 
> Port: USB3.0 at back
> 
> Club 3D CAC-1420
> 
> driver: r8152-braswell 2.14.0-2

This has nothing to do with the driver. Check your DHCP settings and network settings on the NAS.

Cess872, 2021-01-09:
Error found. Problem solved. 
Everything works fine now.
Thank you

Sjekke, 2021-01-09:
> Error found. Problem solved.
> Everything works fine now.
> Thank you

What was the error?

Cess872, 2021-01-09:
SSH: Remote Host identification has changed.
I corrected it, then it worked.

Sjekke, 2021-01-09:
> SSH: Remote Host identification has changed.
> 
> I corrected it, then it worked.

👍