bb-qq/r8152 #84: Works for a while on DS420+ rear port

mrbroons, 2021-01-09:
Issue:
Installs and works for a while but after a few hours, and possibly when under load copying files, the device disconnects from the network. The dongle looks fine with connection light lit and activity light flashing but nothing is working.

My environment:
DS420+
DSM 6.2.3-25426 Update 3
Port: USB3.0 at back
USBC-E2500
driver: r8152-geminilake-2.14.0-2.spk

bb-qq, 2021-01-10:
Can you try the front port? Some users reported stability issues when they use rear ports.

mrbroons, 2021-01-10:
Sure thing. I was wondering why that might be given the ports are the same.
I will or course give it a try and report back, thank you.

Sjekke, 2021-01-10:
Strange my 916+ worked flawlessly for weeks ... today I had 2 disconnects ...

bb-qq, 2021-02-23:
> Sure thing. I was wondering why that might be given the ports are the same.
> I will or course give it a try and report back, thank you.

What were the results? @mrbroons 
The front ports are soldered directly to the motherboard, while the rear ports are via the daughter card. This seems to affect the signal quality or something else.

> Strange my 916+ worked flawlessly for weeks ... today I had 2 disconnects ...

After all, how often is a disconnection likely to occur?

Sjekke, 2021-02-23:
> 
> 
> 
> > Strange my 916+ worked flawlessly for weeks ... today I had 2 disconnects ...
> 
> 
> 
> After all, how often is a disconnection likely to occur?

I didn't had disconnects in normal usage but when I'm transferring a lot of files then sometimes it will disconnect

bb-qq, 2024-06-24:
I would like to close this issue as I have not heard back from you in a long time.
If you have any questions, feel free to re-open.

FYI: If the cause of this problem is overheating of the device, then using a product with the latest 8156BG chip may solve the problem.