bb-qq/r8152 #86: SUCCESS: Club 3D CAC-1420 working on DS412+

Rabadahu, 2021-01-13:
DS412+
DSM version 6.2.3-25426 Update 3
Club 3D CAC-1420, rear, down USB port
r8152-cedarview-2.14.0-2.spk

il-logical-mind, 2024-11-18:
List of devices i tested today... I used UsbTreeView on windows 11 to get informations...
i tried also with powered usb 3.0 hub...

iperf3 test results always 1,29/1,0 GBIT in send/recive with all!

CLUB 3D ‎cac-1420 (RTL8156)

Vendor ID                : 0x0BDA (Realtek Semiconductor Corp.)
Product ID               : 0x8156
Manufacturer String      : Realtek
Product String           : USB 10/100/1G/2.5G LAN
Serial                   : 401000001
USB Version              : 3.2 Gen 1
Port maximum Speed       : SuperSpeed
Device maximum Speed     : SuperSpeed
Device Connection Speed  : SuperSpeed
Self powered             : no
Demanded Current         : 256 mA
Used Endpoints           : 4

D-Link DUB-E250 (RTL8156)

Vendor ID                : 0x2001 (D-Link Corporation)
Product ID               : 0xB301
Manufacturer String      : D-Link
Product String           : DUBE250 2.5GbE Adapter
Serial                   : 001000001
USB Version              : 3.2 Gen 1
Port maximum Speed       : SuperSpeed
Device maximum Speed     : SuperSpeed
Device Connection Speed  : SuperSpeed
Self powered             : no
Demanded Current         : 256 mA
Used Endpoints           : 4

Belkin ‎INC012btBK (RTL8156)

Vendor ID                : 0x0BDA (Realtek Semiconductor Corp.)
Product ID               : 0x8156
Manufacturer String      : Realtek
Product String           : USB 10/100/1G/2.5G LAN
Serial                   : 4013000001
USB Version              : 2.1
Port maximum Speed       : High-Speed (Companion Port 3-26 is doing the SuperSpeed)
Device maximum Speed     : SuperSpeed
Device Connection Speed  : High-Speed
Self powered             : no
Demanded Current         : 200 mA
Used Endpoints           : 4

‎StarTech ‎US2GC30 (RTL8156)

Vendor ID                : 0x0BDA (Realtek Semiconductor Corp.)
Product ID               : 0x8156
Manufacturer String      : Realtek
Product String           : USB 10/100/1G/2.5G LAN
Serial                   : 4013000001
USB Version              : 3.2 Gen 1
Port maximum Speed       : SuperSpeed
Device maximum Speed     : SuperSpeed
Device Connection Speed  : SuperSpeed
Self powered             : no
Demanded Current         : 256 mA
Used Endpoints           : 4

ASUS USB-C2500 (RTL8156)

Vendor ID                : 0x0B05 (ASUSTek Computer Inc.)
Product ID               : 0x1976
Manufacturer String      : Realtek
Product String           : USB 10/100/1G/2.5G LAN
Serial                   : 5013000001
USB Version              : 3.2 Gen 1
Port maximum Speed       : SuperSpeed
Device maximum Speed     : SuperSpeed
Device Connection Speed  : SuperSpeed
Self powered             : no
Demanded Current         : 256 mA
Used Endpoints           : 4

uni ‎UA176-24 (RTL8156)

Vendor ID                : 0x0BDA (Realtek Semiconductor Corp.)
Product ID               : 0x8156
Manufacturer String      : Realtek
Product String           : USB 10/100/1G/2.5G LAN
Serial                   : 4013000001
USB Version              : 3.2 Gen 1
Port maximum Speed       : SuperSpeed
Device maximum Speed     : SuperSpeed
Device Connection Speed  : SuperSpeed
Self powered             : no
Demanded Current         : 256 mA
Used Endpoints           : 4

il-logical-mind, 2024-11-21:
> Cedarview is old, so this speed may be a limitation of hardware performance.

yes, so do you agree that "slow" should be written instead of "work" otherwise another crazy person like me will try for a long time in vain without succeeding? When someone says that something works it should be documented..
Thanks for your work!