bb-qq/r8152 #87: DS216play - Success but speeds limited to 1Gb

szczottie, 2021-01-18:
Hi,

I connected my Synology Nas DS216play with Plugable 2.5G USB C and USB to Ethernet Adapter directly to my MacMini with Sabrent USB Type-A or Type-C to 5-Gigabit Ethernet Adapter and when I run iperf3 I only get 1 Gb speeds. On NAS adapter does showes up as 2500. I checked and connected same adapter to my MacBook Pro instead of NAS and I managed to get full 2.5GbE speeds. 

What am I doing wrong?
[image]
[image]
[image]

closed 2021-01-18 by szczottie

bb-qq, 2021-01-18:
I guess there is a CPU bottleneck with this platform so increasing MTU to reduce load might improve throughput. When you could find a solution, I appreciate it if you share it here.

takeh-t, 2023-10-16:
It may be against the rules to comment on a closed issue, but I am commenting because I encountered the same event.
In the case of the USB adapter I purchased, it linked up at 1Gbps before the driver successfully recognized the adapter, and when the driver worked, it stayed at 1Gbps. So, in Managed Switch, I forced it to be set to be recognized at 2.5Gbps, but it did not Link up. So I changed the USB port and it linked up at 2.5Gbps before the driver recognized the adapter. Then I stopped/executed the driver and it recognized it at 2.5Gbps. Once recognized at 2.5Gbps, it continues to work at 2.5Gbps even when reconnected to an Unmanaged Switch.

I do not know if this will be a solution for everyone, since an environment with a fixed 2.5Gbps link-up is required at least once, but those who can prepare the same environment should give it a try.
(DSM 7.2, Monaco)