bb-qq/r8152 #88: DSM 7.0 - Got it actually working

BeatSkip, 2021-01-23:
Not necessarily an issue, just wanted to let you guys know that i got this driver actually working
on the DSM 7.0 Beta with a DS415Play

My workflow (proof of concept):
1. Install DSM toolkit on clean ubuntu VM and set up according to the official DSM 7.0 Beta PDF
2. Edit the pkgscripts-ng/include/applyEnv file. Line 40 from:
KERNEL_SEARCH_PATH="$SysRootLib/modules/$PRODUCT-$DSM_SHLIB_MAJOR.$DSM_SHLIB_MINOR/build"
to:
KERNEL_SEARCH_PATH="$SysRootLib/modules/DSM-$DSM_SHLIB_MAJOR.$DSM_SHLIB_MINOR/build"
(this fixes build errors)
3. Edit the required INFO.sh file and add the line:
os_min_ver="7.0-40000"
4. compile SPK with: 
sudo ./PkgCreate.py -v 7.0 -p evansport -c r8152 --min-sdk 7.0
5. extract the resulting SPK file as a zip and extract the r8152.ko file.
6. put it on the synology and use insmod to load the module
7. enable interface with: 
ifconfig eth1 up

[image]
The result! yay! no downgrade needed. Now just a script to load it at boot. if anyone can point me to tips how to do this.

LuxKeiwoker, 2021-01-23:
I guess the same process would also apply to ARM based Synology devices?

BeatSkip, 2021-01-23:
Honestly couldn't say, but I would suppose it wouldn't be any different

LuxKeiwoker, 2021-01-23:
Ok I've been able to recompile the file as well for rtd1296 and insmod did not give me an error at least. Now I just need to get the USB adapter to confirm that it's working. But boy oh boy - I remember now why I hate working with linux and it's command lines. I've been able to fill in the "gaps" of your instructions, but it took me like 50 browser tabs to overcome a lot of hurdles. Well in any case: thanks again :). I'll come back an confirm if it's working as soon as I've received the adapter

BeatSkip, 2021-01-23:
Yeah, sorry if I wrote the entire step by step instructions It would have taken an entire essay. And as It's just a proof-of-principle for me it would give me more trouble from people not knowing what they did getting errors. Just wanted to proof that it is possible and get people in their way to do it themselves. I'll probably make a fork later on that's a bit more usable. But I want to try to get that in a working spk and not extracting and copying the module manually.

BeatSkip, 2021-01-23:
But great that it worked @LuxKeiwoker!

wingsuiting, 2021-01-23:
Can anyone confirm working on a synology 720+ NAS? If so it may be time to get a youtube tutorial together now that the beta's playing nice.

BeatSkip, 2021-01-24:
> Can anyone confirm working on a synology 720+ NAS? If so it may be time to get a youtube tutorial together now that the beta's playing nice.

If i send you a SPK, would you be willing to test? if you can confirm it as working i'll release it on my fork as a release. but i would like to have it confirmed tested before i do that. i'm setting up the build environment right now

wingsuiting, 2021-01-24:
I 

> If i send you a SPK, would you be willing to test? if you can confirm it as working i'll release it on my fork as a release. but i would like to have it confirmed tested before i do that. i'm [...]

Thank you - I would but I'm still on a QNAP - it's the 1gbe that's been putting me (and many others) off moving to the 720+. Does anyone else have an [x]20 unit they can test on?

davidgirard, 2021-01-24:
I have a 220+ (J4025) and can test

BeatSkip, 2021-01-24:
@davidgirard 

https://unity.polyform.dev:5551/sharing/kZG2iH4KU

how soon can you test? soon going to bed haha

BeatSkip, 2021-01-24:
For anyone willing to test, here's the full first release:

[R8152-All-2.14.0-11-SPK.zip](https://github.com/bb-qq/r8152/files/5861521/R8152-All-2.14.0-11-SPK.zip)

if everyone that tests it would kindly reply if it did or didn't work. then i can upload it as a full release tomorrow :D

DJNoblesse, 2021-01-24:
> For anyone willing to test, here's the full first release:
> 
> [R8152-All-2.14.0-11-SPK.zip](https://github.com/bb-qq/r8152/files/5861521/R8152-All-2.14.0-11-SPK.zip)
> 
> if everyone that tests it would kindly reply if it did or didn't work. then i can upload it as a full release tomorrow :D

Tested on DS216+II(Braswell) and it works great!

BeatSkip, 2021-01-24:
@DJNoblesse thanks! That's awesome. Seems like I'm one of the first on the internet to get device drivers working on dsm 7.0 :sunglasses: at least I couldn't find anything at all

lerry, 2021-01-24:
> For anyone willing to test, here's the full first release:
> 
> [R8152-All-2.14.0-11-SPK.zip](https://github.com/bb-qq/r8152/files/5861521/R8152-All-2.14.0-11-SPK.zip)
> 
> if everyone that tests it would kindly reply if it did or didn't work. then i can upload it as a full release tomorrow :D

My DSM was broken after install this package and reboot , DS218+.  It messed up nginx conf file in /etc/nginx/conf.d/, many packages can't start. Finally I fixed it manually

BeatSkip, 2021-01-24:
@lerry you did use the Apollolake package? Really curious what happened there, as there's nothing to do with nginx in the package

Limoges65, 2021-01-24:
Did you guys use newer R8152-drivers than in the last 2.14.0-2 spk? There is still some malfunction with rtd1296-chipsets, not yet covered

BeatSkip, 2021-01-24:
@Limoges65 no it is the same one as the 2.14-2, as that is also the latest one available from Realtek

Limoges65, 2021-01-24:
> 
> 
> @Limoges65 no it is the same one as the 2.14-2, as that is also the latest one available from Realtek

Do you expect any improvement with the Club3d adapter, connected to Synologies with rtd1296 with DSM7.0 and the new package?

LuxKeiwoker, 2021-01-24:
> For anyone willing to test, here's the full first release:
> 
> [R8152-All-2.14.0-11-SPK.zip](https://github.com/bb-qq/r8152/files/5861521/R8152-All-2.14.0-11-SPK.zip)
> 
> if everyone that tests it would kindly reply if it did or didn't work. then i can upload it as a full release tomorrow :D

With installing the spk, is the driver persistent? So no need to reinstall after restart of the NAS?

> > @Limoges65 no it is the same one as the 2.14-2, as that is also the latest one available from Realtek
> 
> Do you expect any improvement with the Club3d adapter, connected to Synologies with rtd1296 with DSM7.0 and the new package?

I think no one would know, as I believe I've been the first one to repack the rtd1296 driver for DSM7.0 in this project. Don't have the adapter yet, but it's in the mail.

BeatSkip, 2021-01-24:
@LuxKeiwoker yes, it's persistent. Just like the original DSM 6 version. That is the reason I continued working on it.

BeatSkip, 2021-01-24:
@Limoges65 I would have the faintest clue about performance expectations. I just worked on the repack and adaptation to the new DSM platform

LuxKeiwoker, 2021-01-24:
> @LuxKeiwoker yes, it's persistent. Just like the original DSM 6 version. That is the reason I continued working on it.

out of curiosity: what was necessary to be able to install the *.spk ? I mean the whole process you described in the first post was giving me an SPK, but i extracted the *.ko file out of it and installed with insmod. I never actually tried to use the *.spk

BeatSkip, 2021-01-24:
1. Include extra privilege file that states the package should run as user and not root
2. Include that privilege file in package
3.Edit post install script to edit the  privilege file back to root permissions after install (dumb security bug in my mind)
4. Compile with new DSM 7.0 sdk

BeatSkip, 2021-01-24:
@lerry do you have any additional info? I don't want to be the responsible for people f'ing up their nas :flushed:

lerry, 2021-01-24:
Yes I used the Apollolake package. I tried twice, at first I compiled the package with your forked repo, the second time I use your package. After installing, it can't work, nothing happened, so I reboot my NAS, and it restart once and once again itself. So I unplugged the USB adapter, this time it booted successfully, but almost all packages can't up, such as  Photos, Cloud Sync,  The log shows there was something wrong with nginx conf, I found many conf in  /etc/nginx/conf.d became gibberish,  I removed and  reinstalled them, and the system became normal. I tried twice, and it behaved the same

BeatSkip, 2021-01-24:
Hmm weird, and have you installed this driver from @bb-qq on DSM 6.2 before with success?

BeatSkip, 2021-01-24:
I checked, and the word 'nginx' isn't even once in the entire source code of the package :sweat_smile:, so I'm baffled how that file could be corrupted, and any other files related to that

BeatSkip, 2021-01-24:
@lerry could you provide me with a zip of all the logs, including /var/log/packages/r8152 logs and
The debug.dat
https://jeremyverda.net/export-synology-nas-system-logs/
You can post it here or mail it to oscar_de_boy@msn.com (this is my spambox mail, not sure if I want my good mail posted all around)

lerry, 2021-01-24:
> Hmm weird, and have you installed this driver from @bb-qq on DSM 6.2 before with success? Trying to figure out if it was a problem before my modifications or that i introduced it with my edits

Yes, it's ok  on DSM 6.2. I don't why ,but it just happened  

Errors in Log Center
it appeared when system is broken.

`错误 | 系统 | 2021/01/24 12:50:01 | SYSTEM | Task Scheduler Output results path is inaccessible, please check output result folder is accessible.`

`/var/log$ sudo grep nginx *`

```
synobootup.log:Jan 24 11:22:36 LerryStation synopkgctl[21076]: systemd_reload.cpp:17 synosystemd: [nginx] reloading ...
synobootup.log:Jan 24 11:22:36 LerryStation synopkgctl[21076]: systemd_reload.cpp:21 synosystemd: [nginx] reloaded.
synobootup.log:Jan 24 11:22:36 LerryStation synow3tool[24087]: nginx_deploy.cpp:906 Nginx config has problem, change to abnormal mode
synobootup.log:Jan 24 11:22:38 LerryStation synow3tool[24087]: nginx: [emerg] too long parameter "
synobootup.log:Jan 24 11:22:38 LerryStation systemd[1]: nginx.service: control process exited, code=exited status=1
synobootup.log:Jan 24 11:23:02 LerryStation synopkgctl[12689]: systemd_reload.cpp:17 synosystemd: [nginx] reloading ...
synobootup.log:Jan 24 11:23:02 LerryStation synopkgctl[12689]: systemd_reload.cpp:21 synosystemd: [nginx] reloaded.
synobootup.log:Jan 24 11:23:03 LerryStation synow3tool[26300]: nginx_deploy.cpp:906 Nginx config has problem, change to abnormal mode
synobootup.log:Jan 24 11:23:04 LerryStation systemd[1]: nginx.service: control process exited, code=exited status=1
synobootup.log:Jan 24 11:23:04 LerryStation synow3tool[26300]: nginx: [emerg] too long parameter "
synobootup.log:Jan 24 11:23:06 LerryStation synopkgctl[22086]: systemd_reload.cpp:17 synosystemd: [nginx] reloading ...
synobootup.log:Jan 24 11:23:06 LerryStation synopkgctl[22086]: systemd_reload.cpp:21 synosystemd: [nginx] reloaded.
synobootup.log:Jan 24 11:23:06 LerryStation synow3tool[26642]: nginx_deploy.cpp:906 Nginx config has problem, change to abnormal mode
synobootup.log:Jan 24 11:23:08 LerryStation synow3tool[26642]: nginx: [emerg] too long parameter "
synobootup.log:Jan 24 11:23:08 LerryStation systemd[1]: nginx.service: control process exited, code=exited status=1
synobootup.log:Jan 24 11:23:08 LerryStation synopkgctl[26787]: nginx_conf_register.cpp:755 nginx test fail, nginx: [emerg] too long parameter "nginx: configuration file /var/tmp/nginx/test/plugin_config/nginx.conf test failed
synobootup.log:Jan 24 11:23:15 LerryStation synopkgctl[27186]: systemd_reload.cpp:17 synosystemd: [nginx] reloading ...
synobootup.log:Jan 24 11:23:15 LerryStation synopkgctl[27186]: systemd_reload.cpp:21 synosystemd: [nginx] reloaded.
synobootup.log:Jan 24 11:23:15 LerryStation synow3tool[28026]: nginx_deploy.cpp:906 Nginx config has problem, change to abnormal mode
synobootup.log:Jan 24 11:23:16 LerryStation synow3tool[28026]: nginx: [emerg] too long parameter "
synobootup.log:Jan 24 11:23:16 LerryStation systemd[1]: nginx.service: control process exited, code=exited status=1
synobootup.log:          1.936s nginx.service
synobootup.log:  nginx.service                                                                                                        loaded    active   running Nginx
synobootup.log:Jan 24 12:50:50 LerryStation synopkgctl[18761]: systemd_reload.cpp:17 synosystemd: [nginx] reloading ...
synobootup.log:Jan 24 12:50:50 LerryStation synopkgctl[18761]: systemd_reload.cpp:21 synosystemd: [nginx] reloaded.
synobootup.log:Jan 24 12:50:50 LerryStation synow3tool[20586]: nginx_deploy.cpp:906 Nginx config has problem, change to abnormal mode
synobootup.log:Jan 24 12:50:51 LerryStation systemd[1]: nginx.service: control process exited, code=exited status=1
synobootup.log:Jan 24 12:50:51 LerryStation synow3tool[20586]: nginx: [emerg] too long parameter "
synobootup.log:Jan 24 12:51:09 LerryStation synopkgctl[19258]: systemd_reload.cpp:17 synosystemd: [nginx] reloading ...
synobootup.log:Jan 24 12:51:09 LerryStation synopkgctl[19258]: systemd_reload.cpp:21 synosystemd: [nginx] reloaded.
synobootup.log:Jan 24 12:51:09 LerryStation synow3tool[21650]: nginx_deploy.cpp:906 Nginx config has problem, change to abnormal mode
synobootup.log:Jan 24 12:51:10 LerryStation synow3tool[21650]: nginx: [emerg] too long parameter "
synobootup.log:Jan 24 12:51:10 LerryStation systemd[1]: nginx.service: control process exited, code=exited status=1

synobootup.log

Jan 24 14:01:50 LerryStation cloud-control[20408]: (20408:47648) [ERROR] file-lock.cpp(58): wrong fd
Jan 24 14:01:50 LerryStation cloud-control[20408]: (20408:47648) [ERROR] file-lock.cpp(58): wrong fd
Jan 24 14:01:50 LerryStation cloud-control[20408]: (20408:47648) [ERROR] file-lock.cpp(58): wrong fd
Jan 24 14:01:50 LerryStation cloud-control[20408]: (20408:47648) [ERROR] file-lock.cpp(58): wrong fd
Jan 24 14:01:50 LerryStation cloud-control[20408]: (20408:47648) [ERROR] file-lock.cpp(58): wrong fd
```

-- | -- | -- | -- | --

--

BeatSkip, 2021-01-24:
> Yes, it's ok on DSM 6.2. I don't why ,but it just happened
>
> Errors in Log Center
> it appeared when system is broken.
>
> `错误 | 系统 | 2021/01/24 12:50:01 | SYSTEM | Task Scheduler Output results path is [...]

okay, that means it's introduced in my package. can you provide me with the Debug.dat file. As the nginx error is the resulting problem but not the source. the debug.dat should provide me possibly with the source of the corruption (hopefully)

lerry, 2021-01-24:
[r8152.log](https://github.com/bb-qq/r8152/files/5862568/r8152.log)

lerry, 2021-01-24:
generating of  Debug.dat is very slow

BeatSkip, 2021-01-24:
@lerry wait, are you doing anything with or inside docker?
i see docker come up in the log file a lot

lerry, 2021-01-24:
@BeatSkip I use Docker a lot, just common usage, redis, mysql, and so on

BeatSkip, 2021-01-24:
@lerry yeah, but are you using this with docker in any way on the machine, or is it "just installed on the same system" as docker souldn't come up in the r8152 log if you just install it normally via the package manager. Otherwise i would say docker is the issue here. preventing the spk file from properly installing the kernel module.

BeatSkip, 2021-01-24:
my system doesn't support docker, so don't know how the docker stuff on synology works, I just know it from the command line

lerry, 2021-01-24:
I install docker from package manager, they just installed on the same system

lerry, 2021-01-24:
I can't give you all logs for privacy, maybe you can tell me which one is useful for debugging

BeatSkip, 2021-01-24:
@lerry that's understandable, i'm generating my own debug.dat to see wich files might be interesting to look at

BeatSkip, 2021-01-24:
debug\dsm\var\log\synosystemd.log
debug\dsm\var\log\synoservice.log
debug\dsm\var\log\syslog.log
debug\dsm\var\log\dmesg (this one is very interesting)
debug\dsm\var\log\messages (this one too)
debug\dsm\var\log\kern.log

lerry, 2021-01-24:
[logs.zip](https://github.com/bb-qq/r8152/files/5862634/logs.zip)

lerry, 2021-01-24:
[messages1.zip](https://github.com/bb-qq/r8152/files/5862655/messages1.zip)

BeatSkip, 2021-01-24:
@lerry is see r8152 errors going back to beginning of this month, do you have any residual files/stuff or anything still on you system from those attempts that might interfere with this package? if there's an older non-functioning r8152.ko file somewhere that my package can't overwrite or anything that could interfere. as it seems the kernel crashed quite a few times in the earlier attempts.
i see this coming up a lot:

`2021-01-24T10:11:59+08:00 LerryStation kernel: [  177.269680] r8152: module verification failed: signature and/or required key missing - tainting kernel`

but that doesn't pop-up in my logs, that means either my system doesn't care about the signatures and yours does, or you system persists the older .ko file and keeps trying that instead of mine. either way, the kernel module can't load

BeatSkip, 2021-01-24:
@lerry can you try to extract the r8152.ko file from the spk, put it on the nas somewhere and load the kernel manually with insmod? then you can check if it's the actual module that won't load or something else about the package that messes with things

lerry, 2021-01-25:
I tried again just now, I run `sudo insmod r8152.ko` and plug in the adapter, the DSM system rebooted itself immediately.
Luckily the system is good after reboot

BeatSkip, 2021-01-25:
Okay, can you gather the same logs as earlier? I'll look at them tomorrow and check if it's indeed the module signing problem.

lerry, 2021-01-25:
> Okay, can you gather the same logs as earlier? I'll look at them tomorrow and check if it's indeed the module signing problem.

[logs.zip](https://github.com/bb-qq/r8152/files/5864017/logs.zip)

LuxKeiwoker, 2021-01-25:
ok just received the Club3D Adapter for my DS418j. I can confirm that your package is working (kind of) @BeatSkip. After installation I see the adapter, I can configure it and connect to DSM with the PC attached to it. But I can't transfer big files - as soon as it starts transferring data, the connection collapses.
Reducin the Link-Speed to 1 GBit/s makes usable, but defeats the purpose. I think I've read about this issue already somewhere else and it might be related to MTU size? (I had Auto = 1500)

BeatSkip, 2021-01-25:
I had a problem that I needed to reboot windows on my pc (oddly enough) for everything to work correctly. And had to use a USB hub between the synology and the USB ethernet adapter. @LuxKeiwoker could you try both of those and see if that improves, just curious if those are limited to my setup.

BeatSkip, 2021-01-25:
As for the USB hub, I suspect something odd is going on with the synology driver for the USB 3.0 card in the nas

BeatSkip, 2021-01-25:
[image]
this i the iPerf report i get, anyone else tested iPerf3?

LuxKeiwoker, 2021-01-25:
Hmm I haven't restarted the PC after connecting, I'll try it later. Unfortunately I don't have a USB 3.0 hub laying around.

LuxKeiwoker, 2021-01-25:
> [image]
> 
> this i the iPerf report i get, anyone else tested iPerf3?
> 
> EDIT: new driver on windows and MTU to 9000 bumped it from 1.43Gbit to 1.69Gbit

I'm getting a 

`unable to connect to server: Connection refused `

Do I need to login or something?

BeatSkip, 2021-01-25:
@LuxKeiwoker You need to open the port on the nas in the configuration manager, and maybe even run it as sudo

LuxKeiwoker, 2021-01-25:
So a restart of the PC (or the NAS for that matter) didn't help and after a couple of attempts to copy files from the NAS it seems the whole network stack has crashed or something because the NAS is unresponsive now over both NICs. I can ping it, but I can't open the webinterface, or ssh into it. The only thing I can do at this point is to switch it off with the physical button and restart it. 

Any idea in which logs I should start digging to identify the issue?

LuxKeiwoker, 2021-01-25:
Ok I started digging bit into the files. Dmesg was giving me mostly nothing useful. 

When attempting to copy over a 300 MB video file this is what happens

> [Mon Jan 25 19:51:09 2021] r8152 5-1:1.0 eth1: carrier on
> [Mon Jan 25 19:52:49 2021] NETDEV WATCHDOG: eth1 (r8152): transmit queue 0 timed out
> [Mon Jan 25 19:52:49 2021] r8152 5-1:1.0 eth1: Tx timeout
> [Mon Jan 25 19:52:49 2021] r8152 5-1:1.0 eth1: Tx status -2
> [Mon Jan 25 19:52:49 2021] r8152 5-1:1.0 eth1: Tx status -2
> [Mon Jan 25 19:52:49 2021] r8152 5-1:1.0 eth1: Tx status -2
> [Mon Jan 25 19:52:49 2021] r8152 5-1:1.0 eth1: Tx status -2
> [Mon Jan 25 19:52:51 2021] r8152 5-1:1.0 eth1: get_registers -108
> [Mon Jan 25 19:52:51 2021] r8152 5-1:1.0 eth1: get_registers -71
> [Mon Jan 25 19:52:51 2021] r8152 5-1:1.0 eth1: get_registers -71
> [Mon Jan 25 19:52:51 2021] r8152 5-1:1.0 eth1: get_registers -71
> [Mon Jan 25 19:52:51 2021] r8152 5-1:1.0 eth1: get_registers -71
> [Mon Jan 25 19:52:51 2021] r8152 5-1:1.0 eth1: get_registers -71
> [Mon Jan 25 19:52:51 2021] r8152 5-1:1.0 eth1: get_registers -71
> [Mon Jan 25 19:52:51 2021] r8152 5-1:1.0 eth1: get_registers -71
> [Mon Jan 25 19:52:51 2021] r8152 5-1:1.0 eth1: get_registers -71
> [Mon Jan 25 19:52:51 2021] r8152 5-1:1.0 eth1: get_registers -71
> [Mon Jan 25 19:52:51 2021] r8152 5-1:1.0 eth1: get_registers -71
> [Mon Jan 25 19:52:51 2021] r8152 5-1:1.0 eth1: get_registers -71
> [Mon Jan 25 19:52:51 2021] r8152 5-1:1.0 eth1: get_registers -71
> [Mon Jan 25 19:52:51 2021] r8152 5-1:1.0 eth1: get_registers -71
> [Mon Jan 25 19:52:51 2021] r8152 5-1:1.0 eth1: get_registers -71
> [Mon Jan 25 19:52:51 2021] r8152 5-1:1.0 eth1: get_registers -71

The whole adapter becomes unresponsive afterwards.

the r8152.log mentions a missing file at each startup

> 2021/01/25 18:30:29	install r8152 2.14.0-11 Begin preinst
> 2021/01/25 18:30:29	install r8152 2.14.0-11 End preinst ret=[0]
> 2021/01/25 18:30:29	install r8152 2.14.0-11 Begin /bin/mkdir -p /volume1/@appstore/r8152
> 2021/01/25 18:30:29	install r8152 2.14.0-11 End /bin/mkdir -p /volume1/@appstore/r8152 ret=[0]
> 2021/01/25 18:30:29	install r8152 2.14.0-11 Begin /bin/rm -rf /volume1/@appstore/r8152
> 2021/01/25 18:30:29	install r8152 2.14.0-11 End /bin/rm -rf /volume1/@appstore/r8152 ret=[0]
> 2021/01/25 18:30:29	install r8152 2.14.0-11 Begin /bin/mv -f /volume1/@tmp/pkginstall/extract.O4wvOG/package /volume1/@appstore/r8152
> 2021/01/25 18:30:29	install r8152 2.14.0-11 End /bin/mv -f /volume1/@tmp/pkginstall/extract.O4wvOG/package /volume1/@appstore/r8152 ret=[0]
> 2021/01/25 18:30:29	install r8152 2.14.0-11 Begin /bin/rm -rf /var/packages/r8152
> 2021/01/25 18:30:29	install r8152 2.14.0-11 End /bin/rm -rf /var/packages/r8152 ret=[0]
> 2021/01/25 18:30:29	install r8152 2.14.0-11 Begin /bin/mkdir -p /var/packages/r8152
> 2021/01/25 18:30:29	install r8152 2.14.0-11 End /bin/mkdir -p /var/packages/r8152 ret=[0]
> 2021/01/25 18:30:29	install r8152 2.14.0-11 Begin /bin/touch /var/packages/r8152/installing
> 2021/01/25 18:30:29	install r8152 2.14.0-11 End /bin/touch /var/packages/r8152/installing ret=[0]
> 2021/01/25 18:30:29	install r8152 2.14.0-11 Begin /bin/mv -f /volume1/@tmp/pkginstall/extract.O4wvOG/INFO /var/packages/r8152/INFO
> 2021/01/25 18:30:29	install r8152 2.14.0-11 End /bin/mv -f /volume1/@tmp/pkginstall/extract.O4wvOG/INFO /var/packages/r8152/INFO ret=[0]
> 2021/01/25 18:30:29	install r8152 2.14.0-11 Begin /bin/rm -rf /var/packages/r8152/scripts
> 2021/01/25 18:30:29	install r8152 2.14.0-11 End /bin/rm -rf /var/packages/r8152/scripts ret=[0]
> 2021/01/25 18:30:29	install r8152 2.14.0-11 Begin /bin/mv -f /volume1/@tmp/pkginstall/extract.O4wvOG/scripts /var/packages/r8152/scripts
> 2021/01/25 18:30:29	install r8152 2.14.0-11 End /bin/mv -f /volume1/@tmp/pkginstall/extract.O4wvOG/scripts /var/packages/r8152/scripts ret=[0]
> 2021/01/25 18:30:29	install r8152 2.14.0-11 Begin /bin/rm -rf /var/packages/r8152/conf
> 2021/01/25 18:30:29	install r8152 2.14.0-11 End /bin/rm -rf /var/packages/r8152/conf ret=[0]
> 2021/01/25 18:30:29	install r8152 2.14.0-11 Begin /bin/mv -f /volume1/@tmp/pkginstall/extract.O4wvOG/conf /var/packages/r8152/conf
> 2021/01/25 18:30:29	install r8152 2.14.0-11 End /bin/mv -f /volume1/@tmp/pkginstall/extract.O4wvOG/conf /var/packages/r8152/conf ret=[0]
> 2021/01/25 18:30:32	install r8152 2.14.0-11 Begin postinst
> 2021/01/25 18:30:32	install r8152 2.14.0-11 End postinst ret=[0]
> 2021/01/25 18:30:32	install r8152 2.14.0-11 Begin start-stop-status start
> **ls: cannot access '/sys/class/net/sit0/device/driver': No such file or directory**
> 2021/01/25 18:30:37	install r8152 2.14.0-11 End start-stop-status start ret=[0]
> 2021/01/25 18:31:32	(system) trigger r8152 2.14.0-11 Begin start-stop-status stop
> ls: cannot access '/sys/class/net/sit0/device/driver': No such file or directory
> 2021/01/25 18:31:33	(system) trigger r8152 2.14.0-11 End start-stop-status stop ret=[0]
> 2021/01/25 18:35:43	(system) trigger r8152 2.14.0-11 Begin start-stop-status start
> **ls: cannot access '/sys/class/net/sit0/device/driver': No such file or directory**
> 2021/01/25 18:35:50	(system) trigger r8152 2.14.0-11 End start-stop-status start ret=[0]
> 2021/01/25 19:13:06	(system) trigger r8152 2.14.0-11 Begin start-stop-status stop
> 2021/01/25 19:22:01	(system) trigger r8152 2.14.0-11 Begin start-stop-status start
> **ls: cannot access '/sys/class/net/sit0/device/driver': No such file or directory**
> 2021/01/25 19:22:07	(system) trigger r8152 2.14.0-11 End start-stop-status start ret=[0]

I created a debug.dat but not sure where to look for further information

BeatSkip, 2021-01-25:
i too get the:
`[Mon Jan 25 19:52:51 2021] r8152 5-1:1.0 eth1: get_registers -71`
although keeps working fine and dandy.
maybe you can try checking if anything is up with the kernel:
`sudo dmesg | grep kernel`
to see if there are any useful errors. 

`ls: cannot access '/sys/class/net/sit0/device/driver': No such file or directory`
should not be a problem, that's just the startup script of the package looping through all network devices.
i'll try to build a version with 2.12 or 2.13 later to check if that maybe gives less issues for you.
first now i have to continue studying for an examn....

LuxKeiwoker, 2021-01-25:
good luck then :)

BlackBird11, 2021-01-25:
Installed it on my DS918+ without problems, waiting for the dongle (CableCreation USB 3.0 to 2.5 Gigabit LAN Ethernet Cable Adapter) to arrive and will test it with a Netgear GS110EMX Switch. Thank you for your implementation!

BeatSkip, 2021-01-26:
@LuxKeiwoker the kernel taint shouldn't necessarily be an issue, it's what happens after that matters. My kernel is even tainted by a synology module. In the case of @lerry, right after the kernel taint the module would directly be unloaded. wich made stuff faulty. but with my system it wouldn't unload and just continue like nothing hapenned. Getting a DS415+ motherboard soon. i'll see what that one says.

[81 more replies]

closed 2022-01-22 by bb-qq