bb-qq/r8152 #89: Have you tried compiling 8125(b) drivers?

txcanyon, 2021-01-26:
I really appreciate your work with the USB realtek drivers. Have you tried getting any of the pcie 2.5G  (r8125b) drivers to work?

minlang112, 2021-02-08:
> I really appreciate your work with the USB realtek drivers. Have you tried getting any of the pcie 2.5G (r8125b) drivers to work?

if you want to use RTL8125 pcie NIC on XPEnology，you must add the driver into boot img，and so far as i know RTL8125B is not supported yet

bb-qq, 2021-02-20:
I have not tried to build that driver.

If you are using products with a PCIe slot, I highly recommend using a card with an Aquantia chip like [XG-C100C](https://amzn.to/3k3SEdD). It works with the built-in driver and is superior in terms of performance and stability.

If you want to save money, you may be better off using a USB adapter.

bb-qq, 2021-07-11:
I close this issue because I don't have any plans to support PCIe-based products.

closed 2021-07-11 by bb-qq