bb-qq/r8152 #91: DS1621+ not working

HerbertLim, 2021-01-26:
### Description of the problem
No new LAN port is added after the installation of `r8152-v1000-2.14.0-2.spk`.
In case of DS920+, I was able to see a new LAN port after installation of `r8152-geminilake-2.14.0-2.spk`

I tried to run the device driver many times in `Package Center - Installed`

### Description of your products
> - the product name of your NAS model: Synology DS1621+
>     - and output of ``uname -a`` command
>     - Linux Herbert 4.4.59+ #25426 SMP Mon Dec 14 18:52:44 CST 2020 x86_64 GNU/Linux synology_v1000_1621+
> - the product name of the ethernet dongle: 
>   - CLUB 3D CAC-1420 USB 3.2 Type A to RJ45 2.5 Gigabit LAN Ethernet Cable Adapter
>.  - UGREEN USB C to 2.5G Ethernet Adapter Type C to 2.5 Gigabit Network Thunderbolt 3 RJ45 LAN 2.5Gbps Converter

### Description of your environment
MacBook Pro (15-inch, 2018)
macOS 10.15.7
No Hub, connected directly from macbook to NAS
Cat.5e

### Output of ``dmesg`` command

````
[    0.000000] Dentry cache hash table entries: 524288 (order: 10, 4194304 bytes)
[    0.000000] Inode-cache hash table entries: 262144 (order: 9, 2097152 bytes)
[    0.000000] Memory: 3900280K/4131980K available (5591K kernel code, 486K rwdata, 1868K rodata, 860K init, 692K bss, 231700K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=8, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000] 	Build-time adjustment of leaf fanout to 64.
[    0.000000] 	RCU restricting CPUs from NR_CPUS=16 to nr_cpu_ids=8.
[    0.000000] RCU: Adjusting geometry for rcu_fanout_leaf=64, nr_cpu_ids=8
[    0.000000] NR_IRQS:4352 nr_irqs:1032 16
[    0.000000] Console: colour dummy device 80x25
[    0.000000] clocksource: hpet: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 133484873504 ns
[    0.000000] hpet clockevent registered
[    0.000000] tsc: Fast TSC calibration using PIT
[    0.000000] tsc: Detected 2199.755 MHz processor
[    0.000004] Calibrating delay loop (skipped), value calculated using timer frequency.. 4399.51 BogoMIPS (lpj=2199755)
[    0.010574] pid_max: default: 32768 minimum: 301
[    0.015169] ACPI: Core revision 20150930
[    0.025997] ACPI: 8 ACPI AML tables successfully acquired and loaded
[    0.037582] Security Framework initialized
[    0.041516] AppArmor: AppArmor initialized
[    0.045593] Mount-cache hash table entries: 8192 (order: 4, 65536 bytes)
[    0.052259] Mountpoint-cache hash table entries: 8192 (order: 4, 65536 bytes)
[    0.059550] Initializing cgroup subsys io
[    0.063397] Initializing cgroup subsys memory
[    0.067732] Initializing cgroup subsys devices
[    0.072148] Initializing cgroup subsys freezer
[    0.076588] CPU: Physical Processor ID: 0
[    0.080556] CPU: Processor Core ID: 0
[    0.192624] mce: CPU supports 23 MCE banks
[    0.196581] LVT offset 1 assigned for vector 0xf9
[    0.201298] LVT offset 2 assigned for vector 0xf4
[    0.205919] Last level iTLB entries: 4KB 1024, 2MB 1024, 4MB 512
[    0.211897] Last level dTLB entries: 4KB 1536, 2MB 1536, 4MB 768, 1GB 0
[    0.218488] Speculative Store Bypass: Vulnerable
[    0.223397] Freeing SMP alternatives memory: 28K (ffffffff81952000 - ffffffff81959000)
[    0.813539] ..TIMER: vector=0x30 apic1=0 pin1=2 apic2=-1 pin2=-1
[    0.930737] smpboot: CPU0: AMD Ryzen Embedded V1500B (family: 0x17, model: 0x11, stepping: 0x0)
[    0.939265] Performance Events: Fam17h core perfctr, AMD PMU driver.
[    0.945578] ... version:                0
[    0.949564] ... bit width:              48
[    0.953638] ... generic registers:      6
[    0.957624] ... value mask:             0000ffffffffffff
[    0.962912] ... max period:             00007fffffffffff
[    0.968199] ... fixed-purpose events:   0
[    0.972186] ... event mask:             000000000000003f
[    0.979494] NMI watchdog: enabled on all CPUs, permanently consumes one hw-PMU counter.
[    0.987367] x86: Booting SMP configuration:
[... 1760 lines cut ...]
eth3      Link encap:Ethernet  HWaddr 00:11:32:DF:FA:12  
          inet addr:169.254.72.149  Bcast:169.254.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:76 base 0x6000  

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:236682 errors:0 dropped:0 overruns:0 frame:0
          TX packets:236682 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:28856998 (27.5 MiB)  TX bytes:28856998 (27.5 MiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
````

bb-qq, 2021-02-20:
It seems to be an issue on the DSM toolkit side. I applied a workaround to my build environment and rebuild the package.
Please try this.
https://github.com/bb-qq/r8152/releases/download/2.14.0-2a/r8152-v1000-2.14.0-2a.spk

first1117, 2021-02-21:
> It seems to be an issue on the DSM toolkit side. I applied a workaround to my build environment and rebuild the package.
> Please try this.
> https://github.com/bb-qq/r8152/releases/download/2.14.0-2a/r8152-v1000-2.14.0-2a.spk

Hi,thanks for u job.

It's not working.

my device:
DS 1621+
DSM 6.2.3-25426 Update 3
UGREEN USB C to 2.5G Ethernet Adapter

bb-qq, 2021-02-23:
Cloud you paste the output of dmesg?
I would like to confirm whether this error is gone.
```
r8152: version magic '4.4.59+ SMP preempt mod_unload ' should be '4.4.59+ SMP mod_unload '
```

first1117, 2021-02-25:
Thank you, brother.
I hope you can succeed
Otherwise I'll have to give up 2.5GB and spend a lot of money to upgrade to 10GB

`[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 4.4.59+ (root@build5) (gcc version 4.9.3 20150311 (prerelease) (crosstool-NG 1.20.0) ) #25426 SMP Mon Dec 14 18:52:44 CST 2020
[    0.000000] Command line: root=/dev/md0 netif_num=4 SMBusHddDynamicPower=1 syno_hw_version=DS1621+ vender_format_version=2 synoboot2 sn=20A0S7RRYVGMM macs=001132dfe863,001132dfe864,001132dfe865,001132dfe866 syno_ttyS0=mmio,0xfedc9000,115200,48000000 syno_ttyS1=mmio,0xfedca000,115200
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000] tseg: 00cf800000
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: Supporting XSAVE feature 0x01: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x02: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x04: 'AVX registers'
[    0.000000] x86/fpu: Enabled xstate features 0x7, context size is 832 bytes, using 'standard' format.
[    0.000000] e820: BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x0000000000086fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000087000-0x0000000000087fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000088000-0x000000000009ffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000000a0000-0x00000000000bffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x0000000009afffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000009b00000-0x0000000009dfffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000009e00000-0x0000000009efffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000009f00000-0x0000000009f0afff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000009f0b000-0x00000000b7138fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000b7139000-0x00000000b7b38fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000b7b39000-0x00000000cdd4efff] usable
[    0.000000] BIOS-e820: [mem 0x00000000cdd4f000-0x00000000ce11efff] type 20
[    0.000000] BIOS-e820: [mem 0x00000000ce11f000-0x00000000cee8efff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000cee8f000-0x00000000cf77efff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000cf77f000-0x00000000cf7fefff] ACPI data
[    0.000000] BIOS-e820: [mem 0x00000000cf7ff000-0x00000000cf7fffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000cf800000-0x00000000cfffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000f8000000-0x00000000fbffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fdc00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec10000-0x00000000fec10fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed80000-0x00000000fed80fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff800000-0x00000000fff4ffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000082f33ffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] e820: update [mem 0xb38b1000-0xb38b1af6] usable ==> usable
[    0.000000] extended physical RAM map:
[    0.000000] reserve setup_data: [mem 0x0000000000000000-0x0000000000086fff] usable
[    0.000000] reserve setup_data: [mem 0x0000000000087000-0x0000000000087fff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000000088000-0x000000000009ffff] usable
[    0.000000] reserve setup_data: [mem 0x00000000000a0000-0x00000000000bffff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000000100000-0x0000000009afffff] usable
[    0.000000] reserve setup_data: [mem 0x0000000009b00000-0x0000000009dfffff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000009e00000-0x0000000009efffff] usable
[    0.000000] reserve setup_data: [mem 0x0000000009f00000-0x0000000009f0afff] ACPI NVS
[    0.000000] reserve setup_data: [mem 0x0000000009f0b000-0x00000000b38b0fff] usable
[    0.000000] reserve setup_data: [mem 0x00000000b38b1000-0x00000000b38b1af6] usable
[    0.000000] reserve setup_data: [mem 0x00000000b38b1af7-0x00000000b7138fff] usable
[    0.000000] reserve setup_data: [mem 0x00000000b7139000-0x00000000b7b38fff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000b7b39000-0x00000000cdd4efff] usable
[    0.000000] reserve setup_data: [mem 0x00000000cdd4f000-0x00000000ce11efff] type 20
[    0.000000] reserve setup_data: [mem 0x00000000ce11f000-0x00000000cee8efff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000cee8f000-0x00000000cf77efff] ACPI NVS
[    0.000000] reserve setup_data: [mem 0x00000000cf77f000-0x00000000cf7fefff] ACPI data
[    0.000000] reserve setup_data: [mem 0x00000000cf7ff000-0x00000000cf7fffff] usable
[    0.000000] reserve setup_data: [mem 0x00000000cf800000-0x00000000cfffffff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000f8000000-0x00000000fbffffff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000fdc00000-0x00000000fec00fff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000fec10000-0x00000000fec10fff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000fed80000-0x00000000fed80fff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[... 1352 lines cut ...]
[... cut ...]
[642578.023347] r8152: Unknown symbol _raw_spin_unlock (err 0)
[642671.776918] r8152: Unknown symbol _raw_spin_unlock (err 0)
[642706.595019] r8152: Unknown symbol _raw_spin_unlock (err 0)
[642942.009502] r8152: Unknown symbol _raw_spin_unlock (err 0)
[935161.890977] r8152: Unknown symbol _raw_spin_unlock (err 0)
[935795.297366] init: inetd main process (8984) killed by TERM signal
[935865.814223] r8152: Unknown symbol _raw_spin_unlock (err 0)
[935876.574135] r8152: Unknown symbol _raw_spin_unlock (err 0)`

bb-qq, 2021-02-26:
Hmm... it looks like using the old driver...
Could you type "grep -a vermagic= /var/packages/r8152/target/r8152/r8152.ko" ?

first1117, 2021-02-27:
shawn@shawnNAS:~$ grep -a vermagic= /var/packages/r8152/target/r8152/r8152.ko
??
  r???
      r??
??       r
  r???
      r??
tx_packetsrx_packetstx_errorsrx_errorsrx_missedalign_errorstx_single_collisionstx_multi_collisionsrx_unicastrx_broadcastrx_multicasttx_abortedtx_underrun:::::::::::::version=v2.14.0 (2020/09/24)license=GPLdescription=Realtek RTL8152/RTL8153 Based USB Ethernet Adaptersauthor=Realtek nic sw <nic_swsd@realtek.com>srcversion=1234A846AB17265E93614E6alias=usb:v13B1p0041d*dc*dsc*dp*ic02isc0Dip00in*alias=usb:v13B1p0041d*dc*dsc*dp*ic02isc06ip00in*alias=usb:v13B1p0041d*dc*dsc*dp*icFFisc*ip*in*alias=usb:v20F4pE02Bd*dc*dsc*dp*ic02isc0Dip00in*alias=usb:v20F4pE02Bd*dc*dsc*dp*ic02isc06ip00in*alias=usb:v20F4pE02Bd*dc*dsc*dp*icFFisc*ip*in*alias=usb:v0955p09FFd*dc*dsc*dp*ic02isc0Dip00in*alias=usb:v0955p09FFd*dc*dsc*dp*ic02isc06ip00in*alias=usb:v0955p09FFd*dc*dsc*dp*icFFisc*ip*in*alias=usb:v2357p0601d*dc*dsc*dp*ic02isc0Dip00in*alias=usb:v2357p0601d*dc*dsc*dp*ic02isc06ip00in*alias=usb:v2357p0601d*dc*dsc*dp*icFFisc*ip*in*alias=usb:v17EFpA387d*dc*dsc*dp*ic02isc0Dip00in*alias=usb:v17EFpA387d*dc*dsc*dp*ic02isc06ip00in*alias=usb:v17EFpA387d*dc*dsc*dp*icFFisc*ip*in*alias=usb:v17EFpA359d*dc*dsc*dp*ic02isc0Dip00in*alias=usb:v17EFpA359d*dc*dsc*dp*ic02isc06ip00in*alias=usb:v17EFpA359d*dc*dsc*dp*icFFisc*ip*in*alias=usb:v17EFp8153d*dc*dsc*dp*ic02isc0Dip00in*alias=usb:v17EFp8153d*dc*dsc*dp*ic02isc06ip00in*alias=usb:v17EFp8153d*dc*dsc*dp*icFFisc*ip*in*alias=usb:v17EFp721Ed*dc*dsc*dp*ic02isc0Dip00in*alias=usb:v17EFp721Ed*dc*dsc*dp*ic02isc06ip00in*alias=usb:v17EFp721Ed*dc*dsc*dp*icFFisc*ip*in*alias=usb:v17EFp7214d*dc*dsc*dp*ic02isc0Dip00in*alias=usb:v17EFp7214d*dc*dsc*dp*ic02isc06ip00in*alias=usb:v17EFp7214d*dc*dsc*dp*icFFisc*ip*in*alias=usb:v17EFp720Cd*dc*dsc*dp*ic02isc0Dip00in*alias=usb:v17EFp720Cd*dc*dsc*dp*ic02isc06ip00in*alias=usb:v17EFp720Cd*dc*dsc*dp*icFFisc*ip*in*alias=usb:v17EFp720Bd*dc*dsc*dp*ic02isc0Dip00in*alias=usb:v17EFp720Bd*dc*dsc*dp*ic02isc06ip00in*alias=usb:v17EFp720Bd*dc*dsc*dp*icFFisc*ip*in*alias=usb:v17EFp720Ad*dc*dsc*dp*ic02isc0Dip00in*alias=usb:v17EFp720Ad*dc*dsc*dp*ic02isc06ip00in*alias=usb:v17EFp720Ad*dc*dsc*dp*icFFisc*ip*in*alias=usb:v17EFp7205d*dc*dsc*dp*ic02isc0Dip00in*alias=usb:v17EFp7205d*dc*dsc*dp*ic02isc06ip00in*alias=usb:v17EFp7205d*dc*dsc*dp*icFFisc*ip*in*alias=usb:v17EFp3098d*dc*dsc*dp*ic02isc0Dip00in*alias=usb:v17EFp3098d*dc*dsc*dp*ic02isc06ip00in*alias=usb:v17EFp3098d*dc*dsc*dp*icFFisc*ip*in*alias=usb:v17EFp3082d*dc*dsc*dp*ic02isc0Dip00in*alias=usb:v17EFp3082d*dc*dsc*dp*ic02isc06ip00in*alias=usb:v17EFp3082d*dc*dsc*dp*icFFisc*ip*in*alias=usb:v17EFp3069d*dc*dsc*dp*ic02isc0Dip00in*alias=usb:v17EFp3069d*dc*dsc*dp*ic02isc06ip00in*alias=usb:v17EFp3069d*dc*dsc*dp*icFFisc*ip*in*alias=usb:v17EFp3062d*dc*dsc*dp*ic02isc0Dip00in*alias=usb:v17EFp3062d*dc*dsc*dp*ic02isc06ip00in*alias=usb:v17EFp3062d*dc*dsc*dp*icFFisc*ip*in*alias=usb:v17EFp3057d*dc*dsc*dp*ic02isc0Dip00in*alias=usb:v17EFp3057d*dc*dsc*dp*ic02isc06ip00in*alias=usb:v17EFp3057d*dc*dsc*dp*icFFisc*ip*in*alias=usb:v17EFp3054d*dc*dsc*dp*ic02isc0Dip00in*alias=usb:v17EFp3054d*dc*dsc*dp*ic02isc06ip00in*alias=usb:v17EFp3054d*dc*dsc*dp*icFFisc*ip*in*alias=usb:v17EFp3052d*dc*dsc*dp*ic02isc0Dip00in*alias=usb:v17EFp3052d*dc*dsc*dp*ic02isc06ip00in*alias=usb:v17EFp3052d*dc*dsc*dp*icFFisc*ip*in*alias=usb:v17EFp304Fd*dc*dsc*dp*ic02isc0Dip00in*alias=usb:v17EFp304Fd*dc*dsc*dp*ic02isc06ip00in*alias=usb:v17EFp304Fd*dc*dsc*dp*icFFisc*ip*in*alias=usb:v04E8pA101d*dc*dsc*dp*ic02isc0Dip00in*alias=usb:v04E8pA101d*dc*dsc*dp*ic02isc06ip00in*alias=usb:v04E8pA101d*dc*dsc*dp*icFFisc*ip*in*alias=usb:v045Ep07C6d*dc*dsc*dp*ic02isc0Dip00in*alias=usb:v045Ep07C6d*dc*dsc*dp*ic02isc06ip00in*alias=usb:v045Ep07C6d*dc*dsc*dp*icFFisc*ip*in*alias=usb:v045Ep07ABd*dc*dsc*dp*ic02isc0Dip00in*alias=usb:v045Ep07ABd*dc*dsc*dp*ic02isc06ip00in*alias=usb:v045Ep07ABd*dc*dsc*dp*icFFisc*ip*in*alias=usb:v0BDAp8156d*dc*dsc*dp*ic02isc0Dip00in*alias=usb:v0BDAp8156d*dc*dsc*dp*ic02isc06ip00in*alias=usb:v0BDAp8156d*dc*dsc*dp*icFFisc*ip*in*alias=usb:v0BDAp8155d*dc*dsc*dp*ic02isc0Dip00in*alias=usb:v0BDAp8155d*dc*dsc*dp*ic02isc06ip00in*alias=usb:v0BDAp8155d*dc*dsc*dp*icFFisc*ip*in*alias=usb:v0BDAp8153d*dc*dsc*dp*ic02isc0Dip00in*alias=usb:v0BDAp8153d*dc*dsc*dp*ic02isc06ip00in*alias=usb:v0BDAp8153d*dc*dsc*dp*icFFisc*ip*in*alias=usb:v0BDAp8152d*dc*dsc*dp*ic02isc0Dip00in*alias=usb:v0BDAp8152d*dc*dsc*dp*ic02isc06ip00in*alias=usb:v0BDAp8152d*dc*dsc*dp*icFFisc*ip*in*alias=usb:v0BDAp8050d*dc*dsc*dp*ic02isc0Dip00in*alias=usb:v0BDAp8050d*dc*dsc*dp*ic02isc06ip00in*alias=usb:v0BDAp8050d*dc*dsc*dp*icFFisc*ip*in*depends=usbcorev?rmagic=4.4.59+ SMP mod_unload ?-??
 ?
  ?

bb-qq, 2021-03-05:
It seems vermagic is the correct value...: 4.4.59+ SMP mod_unload
If you run insmod manually, will you still get the error?

```
/sbin/insmod /var/packages/r8152/target/r8152/r8152.ko
```

first1117, 2021-03-11:
Manual installation also failed. 【invalid module format】

With a UP editor translation driver

https://www.vediotalk.com/archives/34078
http://veelove.i234.me:5000/d/f/iOAzw5aLObZZNe9tdiJl2rXEENsyIgKc

I think I'll wait and see if more people have this problem. Thank you for your help.

joejohnson1701, 2021-03-12:
> Manual installation also failed. 【invalid module format】
> 
> With a UP editor translation driver
> 
> https://www.vediotalk.com/archives/34078
> http://veelove.i234.me:5000/d/f/iOAzw5aLObZZNe9tdiJl2rXEENsyIgKc
> 
> I think I'll wait and see if more people have this problem. Thank you for your help.

I’m ds1821, in the same situation as you, have you found a solution?

first1117, 2021-03-13:
> I’m ds1821, in the same situation as you, have you found a solution?

I haven't found a solution yet

If there is, I will tell you here

CookieSir, 2021-03-13:
> vediotalk.com/archives/34078

> I haven't found a solution yet
>
> If there is, I will tell you here

OK  think you

CookieSir, 2021-03-13:
> It seems vermagic is the correct value...: 4.4.59+ SMP mod_unload
> If you run insmod manually, will you still get the error?
> 
> ```
> /sbin/insmod /var/packages/r8152/target/r8152/r8152.ko
> ```

I’m ds1821, the same as his situation, I hope you can update the driver files that can be used by the V1000 framework

CookieSir, 2021-03-24:
> OK think you

The driver has been updated to 2.14.0-2b, 1821 can start the driver normally, but I have purchased X550 T2, you can try the new driver file

first1117, 2021-03-29:
> The driver has been updated to 2.14.0-2b, 1821 can start the driver normally, but I have purchased X550 T2, you can try the new driver file

Thank you for your informational friend

I think I should go all the way to the X550 T2 as well

bb-qq, 2021-05-16:
I'm sorry for forgetting to inform the new release that fixes this issue. I would like to close this issue since the driver be loaded correctly. Please create a new issue if that still has problems.

closed 2021-05-16 by bb-qq